Koenkk/zigbee2mqtt #18798: [CC2531]Router

pdrobek, 2023-08-31:
### What happened?

Recently i upgraded my zigbee network by replace CC2531 coordinator with CC2652P based one. Now im trying to use my old CC2531 device as router but after reflash it wont appear in my network (only fast red blinking, and sometimes green). Also no information in zigbee2mqtt log also.

what i'm doing wrong?

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

1.30.4 zstack-3x

### Adapter

Sonoff zigbee 3.0

### Debug log

info  2023-08-31 11:47:08: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"daedc54","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20230507,"transportrev":2},"type":"zStack3x0"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0x00124b0014d99fad","panID":6754},"permit_join":false,"version":"1.30.4"}'
debug 2023-08-31 11:47:40: Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'true'
info  2023-08-31 11:47:40: Zigbee: allowing new devices to join.
info  2023-08-31 11:47:40: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"daedc54","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20230507,"transportrev":2},"type":"zStack3x0"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0x00124b0014d99fad","panID":6754},"permit_join":true,"version":"1.30.4"}'
debug 2023-08-31 11:49:39: Received Zigbee message from 'Router', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
info  2023-08-31 11:49:39: MQTT publish: topic 'zigbee2mqtt/Router', payload '{"led":false,"linkquality":65}'

pdrobek, 2023-08-31:
i found also

debug 2023-08-31 12:08:20: Received Zigbee message from 'Coordinator', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0, ignoring since it is from coordinator
debug 2023-08-31 12:09:20: Received Zigbee message from 'Coordinator', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0, ignoring since it is from coordinator

devastar, 2023-09-30:
I'm having the same problem with a CC2530. It was being used as a coordinator. I then flashed router FW but it still gets seen as a coordinator

francisp2, 2023-09-30:
If you upgraded to a newer coordinator, it will have copied the IEEE-address from your old coordinator, thus preventing the old coordinator to work as a router after reflashing. You have to change the IEEE-address while flashing.

devastar, 2023-09-30:
How do you change it? I even tried deleting dicker images and reinstalled

pdrobek, 2023-09-30:
i can change only secondary IEEE via Flash Programmer but zigbee2mqtt did not found any changes in my zigbee network

pdrobek, 2023-09-30:
debug 2023-09-30 22:09:07: Device 'Coordinator' announced itself
info  2023-09-30 22:09:07: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Coordinator","ieee_address":"0x00124b0014d99fad"},"type":"device_announce"}'
info  2023-09-30 22:09:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Coordinator"},"type":"device_announced"}'

francisp2, 2023-10-01:
> Device 'Coordinator' announced itself
Seems the change was not successful

devastar, 2023-10-01:
For some reason its retaining the device name, type and IEEE

Is there a way to completely erase the chip?

pdrobek, 2023-10-01:
I guess not, primary IEEE always is the same, secondary i can update without any problem.

devastar, 2023-10-01:
How do I do that?

I have tried selecting secondary and changing the IEEE but after flashing
z2m still sees old IEEE

francisp2, 2023-10-02:
I found this:
 the “Retain IEEE address when reprogramming the chip” checkbox  is not checked (should be checked)

https://e2e.ti.com/support/wireless-connectivity/other-wireless-group/other-wireless/f/other-wireless-technologies-forum/15751/question-about-the-ieee-address-of-cc2530

mrneutron42, 2023-10-26:
I've got the same thing going on but with 2 new CC2531 USB sticks I bought to be routers.
I've got a [CC2531 USB stick](https://www.amazon.com/gp/product/B081WRGL31/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) with 20211115 version of CC2531ZNP-Prod.hex and it's working for Zigbee devices.
I got another [CC2531 USB stick](https://www.amazon.com/gp/product/B081WRGL31/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) and loaded it with the 2020_09_29 version of router-cc2531-std.hex.
The CC2531 USB board with router firmware in it blinks the red LED once per second when powered up - indicating it's in pairing mode. I put the CC2531 USB stick into pairing mode from inside Home Assistant and the messages show it's waiting for devices to pair. The CC2531 USB board with router firmware never pairs! Home Assistant settings shows that the Zigbee network is using channel 15.
So, I loaded the CC2531 USB stick with the 2020_09_29 version of router-cc2531-diag-usb.hex.
I put the CC2531 USB board with router firmware into a USB port on my Windwos 10 machine and a COM5 port showed up in the Windows Device Manager....good.
When I open PUTTY to view the serial port info, I see this over and over.

PANID:F5B,Ch:15,Join:0

PANID:F5B,Ch:15,Join:0

PANID:F5B,Ch:15,Join:0

Here are my Home Assistant component versions:
Home Assistant 2023.9.3
Supervisor 2023.10.0
Operating System 11.0
Frontend 20230911.0 - latest

I checked and all of my CC2531 USB sticks have different IEEE addresses, so address conflict can't be the problem.

pdrobek, 2023-10-26:
ups i forgot mention that i resolved this issue :-)

Mayby it can be useful for somebody somehow
1) change zigbee2mqtt location from proxmox vm to rpi3 (i got conflicts with usb devices to mine crypto soon after i started zigbee2mqtt and cgminer - coordinator and those block erupters are on the same chip so cgminer tries to mine hashes on coordinator 😜)
2) change zigbee channel to 20 from default 11
3) step 2 forced me to create new network
4) in production network i had to add more powerful repeaters (ikea tradfre)
5) final result --> happy little zigbee home network

[image]

closed 2023-10-26 by pdrobek