Koenkk/zigbee2mqtt #18799: Paulmann 93999 not recognized

JonasB2497, 2023-08-31:
### What happened?

Hi, I have a Paulmann 93999 Plug Shine Zigbee controller, which should be recognized by zigbee2mqtt as it is listed under supported devices. But after pairing it shows up as unsupported. Unpairing, factory resetting and re-pairing did not fix this issue.

I have found this issue which might be relevant as well: #3134 

It looks like it's zigbeeModel is "Dimmable" and not "H036-0001" as written in the device configuration file.
I have got it working now by adding an external converter which contains the same device definition but with the changed model as such:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Dimmable'],
    model: '93999',
    vendor: 'Paulmann',
    description: 'Plug Shine Zigbee controller',
    extend: extend.light_onoff_brightness(),
};

module.exports = definition;
```

The firmware build date of the Paulmann device is 20190515 and the firmware version is 1400-0001.

I currently can't update the adapter firmware version because it came preflashed and I don't own the appropriate flashing device. But if it could help in fixing the issue I will spend some time and solder wires to flash it with a raspberry pi.

### Zigbee2MQTT version

1.32.2 commit 1ec1e57

### Adapter firmware version

20190608

### Adapter

CC2531

Koenkk, 2023-09-01:
Can you provide the data/database.db entry of this device?

JonasB2497, 2023-09-01:
Yes sure. Here is the corresponding entry:
```
{"id":2,"type":"Router","ieeeAddr":"0x00158d0008c4cb16","nwkAddr":59869,"manufId":4632,"manufName":"Paulmann Licht GmbH","powerSource":"Mains (single phase)","modelId":"Dimmable","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,4,3,6,8,5,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Dimmable","manufacturerName":"Paulmann Licht GmbH","powerSource":1,"zclVersion":3,"appVersion":20,"stackVersion":1,"hwVersion":16,"dateCode":"20190515","swBuildId":"1400-0001"}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":20,"stackVersion":1,"hwVersion":16,"dateCode":"20190515","swBuildId":"1400-0001","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1693509416537,"defaultSendRequestWhen":"immediate"}
```

Koenkk, 2023-09-01:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-09-01 by Koenkk

JonasB2497, 2023-09-02:
Great. Thank you very much!