Koenkk/zigbee2mqtt #18804: White LED of RGBW driver 5110.40 is always on

Stefan-3, 2023-08-31:
### What happened?

Main issue
Controlling the RGB LEDs with LED driver 5110.40 (iluminize.js) is working fine but it's not possible to control the white LEDs of the RGBW LED stripe. The white LEDs are always on with max. brightness. Modifying iluminize.js is not enough (adding multiendpoint results in error log message "not supported by converter").

"Expanded" issue
The LED driver 511.000 from sunricher can be configured with 2 mechanical dip switches to match to the connected LED stripe type (DIM, CCT, RGBW, RGB+CCT). Currently it's quite a "chaos" regarding this device:

[image]

So, the converter for LED driver 511.000 is implemented at 4 different locations. I suggest cleaning this up:
a) all 4 configurations in file sunricher.js (and within this file one configuration after the other)
b) harmonized model name in converter file (e.g. modelId). This means HK-ZD-DIM-A instead of SRP-ZG9105-CC, HK-ZD-RGB-A instead of 5110.40, ...

These are the corresponding entries in the shepard file for all 4 dip switch settings:
[shepard answers.txt](https://github.com/Koenkk/zigbee-herdsman-converters/files/11859993/shepard.answers.txt)

Note: The RGBW driver 5110.40 is currently the "one and only" recommended device at the wholesaler isolicht.com:
https://www.isolicht.com/led-beleuchtung/installationsmaterial/dimmer-schalter-schutz/zigbee-3-0-dimmaktor-fur-led-12-24v-4x1-5a-36-48v-4x0-75a

### What did you expect to happen?

Endpoint to control white LED.

### How to reproduce it (minimal and precise)

Set dip switches of LED driver to RGBW dimmer --> endpoint to control white LED is missing

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20211210

### Adapter

SONOFF ZIGBEE 3.0 USB-STICK

Stefan-3, 2023-10-01:
The problem is still present and the device is used quite often especially in Germany. So sooner or later many other users will have the same problem.

Stefan-3, 2023-11-01:
still waiting

Stefan-3, 2024-05-03:
still waiting

immanuelsch, 2025-01-07:
I do use the SR-ZG1029-5C and Conbee III to control a RGBW strip from HERA. I tried ZHA, deConz and zigbee2mqtt. Always the same: W-Channel and RGB-Channel even do not combine to the right color temperature.

But it should:
https://www.radium.de/sites/default/files/2022-01/TI4_LMZA4000_DIMMER-ZIGBEE-5CH-480W12-24V_DE.pdf

closed 2025-03-17 by github-actions[bot]