Koenkk/zigbee2mqtt #18806: TS0203 Door Sensor Tamper not working

dinde, 2023-08-31:
### What happened?

Tested on:
_TZ3000_decxrtwa
_TZ3000_oxslv1c9

I remove the cover and device should change the tamper state.
State never changes. 
Tamper is always false no matter if contact true or false.

### What did you expect to happen?

Tamper should change its state to true when cover is removed.

### How to reproduce it (minimal and precise)

Pair in Z2M
Remove sensor cover
Remove battery (even)

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20230507

### Adapter

Sonoff ZBDongle-P

### Debug log

Debug 2023-09-01 01:31:34Received Zigbee message from 'door-sensor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
Info 2023-09-01 01:31:34MQTT publish: topic 'zigbee2mqtt/door-sensor', payload '{"battery":100,"battery_low":false,"contact":false,"last_seen":"2023-08-31T23:31:39.771Z","linkquality":153,"tamper":false,"voltage":3000}'
Debug 2023-09-01 01:31:40Received Zigbee message from 'door-sensor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID 0
Info 2023-09-01 01:31:40MQTT publish: topic 'zigbee2mqtt/door-sensor', payload '{"battery":100,"battery_low":false,"contact":true,"last_seen":"2023-08-31T23:31:45.891Z","linkquality":153,"tamper":false,"voltage":3000}'

tdslot, 2023-09-01:
Same to me, it was worked but now not working. Ir pairs get sensor register, but when I open contact didn't receive data. And no tamper data receive. In beginning when open contact it blink red led, now not.

tdslot, 2023-10-03:
Up

Koenkk, 2023-10-03:
does the device have a tamper switch?

can you provide the debug log when opening the cover?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Xyaren, 2023-10-06:
I'm pretty sure my devices identifying as the TS0203 do not have a tamper function.
I've got 5 from 2 different orders.

Xyaren, 2023-11-06:
Still relevant I guess

jbouv55151, 2024-03-28:
I have the same problem with the TS0203.

Aldekein, 2024-04-28:
Still relevant. 

TS0203 by _TZ3000_7d8yme6f door sensor imports 2 sensors into HA.
First is magnet-activated and works well.
Second is tamper activated. I see the red LED in sensor blinking when I toggle it. But HA always presents it as off/closed.

closed 2024-11-25 by github-actions[bot]