Koenkk/zigbee2mqtt #18809: Debug log not showing special characters on new device.

liquidacid, 2023-09-01:
### What happened?

I was adding a new Ikea device which appears to have a VERY weird model ID which makes it very hard to add a new device.
This was my first addition for new devices. The log of Zigbee2MQTT reported:

TRADFRI bulb GU10 WW 345lm8

So i'd add this device to a new JS file named LED2104R3.js but it was still unrecognized with 
zigbeeModel: ['TRADFRI bulb GU10 WW 345lm8'],

After some messing around i found database.db showing:
"modelId":"\u001aTRADFRI bulb GU10 WW 345lm8"

This random square in the ModelID did take some time to find. I could actually add this to the documentation of supporting a new device but maybe it would be better if this modelid is fully shown in the debug log.

### Zigbee2MQTT version

1.32.2-1

### Adapter firmware version

20230507

### Adapter

Sonoff Zigbee_3.0_USB_Dongle_Plus

lokanidao9991, 2023-09-05:
I recently encountered the same issue and found a viable solution that could potentially be integrated into future releases of the zigbee-herdsman-converters device list.

To resolve the issue, like you, I incorporated the Unicode character `U+001A` at the beginning of the "TRADFRI" word in the `zigbeeModel` line, as demonstrated below:

```javascript
zigbeeModel: ['\u001ATRADFRI bulb GU10 WW 345lm8'],
```

Subsequently, I created a new device definition by following the guidelines available on the [Zigbee2MQTT support for new devices documentation](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html). Here is how my updated file appears:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['\u001ATRADFRI bulb GU10 WW 345lm8'],
    model: 'LED2005R5',
    vendor: 'Ikea',
    description: 'MasterConnect LEDtube',
    extend: extend.light_onoff_brightness_colortemp(),
};

module.exports = definition;
```

I believe this workaround might be a valuable addition to a future release of the zigbee-herdsman-converters device list to prevent similar issues.

closed 2023-10-13 by github-actions[bot]

IvSan, 2023-12-06:
Ok, I was able to solve the problem adding external converter LED2104R3.js
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TRADFRI bulb GU10 WW 345lm'],
    model: 'LED2104R3',
    vendor: 'IKEA',
    description: 'TRADFRI LED bulb GU10 WW 345 lumen, dimmable',
    extend: extend.light_onoff_brightness(),
}

module.exports = definition;
```