Koenkk/zigbee2mqtt #18810: [Wrong device]: Miboxer FUT035Z+ recognized as Tuya TS0502B Mercator IkuÜ SMD9300 _TZB210_lmqquxus

filewalker1, 2023-09-01:
### Link

https://miboxer.com/de/produkt/2-in-1-led-controller-zigbee-3-0-2-4g/

### Model

FUT035Z+

### Description

CCT LED Controller

### Vendor

Miboxer

### Picture (link)

https://miboxer.com/wp-content/uploads/a1-51-100x100.jpg

### Database entry

-

allthecogs, 2023-09-25:
Also, when there is a push type wall switch connected, the state does not update within Z2M, however, if you press the refresh icon on the Exposes > State entry, then the correct state is updated. 
State actions have been set to 'True' and Availability has been reduced to '1', but still requires manually refreshing in Z2M.
Can this be fixed to update when an external switch is used?

RdnRzm, 2023-09-26:
I have the same problem

vladopp, 2023-10-10:
I have the issue with the controller. Running latest versions of Z2MQTT and HomeAssistant.
Any way I can help with resolution of the problem?

yorickvP, 2023-11-05:
See also: https://github.com/Koenkk/zigbee2mqtt/issues/16768

closed 2023-11-11 by filewalker1

yorickvP, 2023-11-13:
The + version has some different features though, it can be switched to a mode where it controls a single led strip.