Koenkk/zigbee2mqtt #18811: failed with status '(0xc7: NWK_TABLE_FULL)'

AndreSeinhorst, 2023-09-01:
### What happened?

Zigbee Network not stable, once in a while delay in response and dropping devices. type and brand of dropping devices is vary (not a specific brand)

### What did you expect to happen?

Devices staying in network and no delay in response (switching light on, can take up to 20seconds)

### How to reproduce it (minimal and precise)

the failure happens to be intermediate

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20230507

### Adapter

Sonoff zigbee CC2652P 3.x

### Debug log

error 2023-08-30 23:30:35: Publish 'set' 'state' to 'tuinlamp_schuur_links' failed: 'Error: Command 0xa4c13856c7fede8e/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 53067 - 1 - 18 - 6 - 11 after 10000ms)'
error 2023-08-31 16:07:59: Publish 'set' 'state' to 'slaapkamerjoey_wcd_buro2' failed: 'Error: Command 0x588e81fffed38c6d/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> ZDO - extRouteDisc - {"dstAddr":33729,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)'))'
error 2023-08-31 22:07:39: Failed to execute LQI for 'boiler_wcd'
error 2023-08-31 22:08:28: Failed to execute LQI for 'tuin_sierverlichting'
error 2023-08-31 22:09:18: Failed to execute LQI for 'slaapkamerjoey_wcd_buro2'
error 2023-08-31 22:29:49: Publish 'set' 'state' to 'gang_voordeur_switch' failed: 'Error: Command 0x60a423fffe621504/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> ZDO - extRouteDisc - {"dstAddr":22672,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)'))'
error 2023-08-31 22:30:00: Publish 'set' 'state' to 'gang_voordeur_switch' failed: 'Error: Command 0x60a423fffe621504/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> ZDO - extRouteDisc - {"dstAddr":22672,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)'))'
error 2023-08-31 22:35:02: No converter available for 'flash' (2)
error 2023-09-01 02:07:09: Publish 'set' 'state' to 'gang_voordeur_switch' failed: 'Error: Command 0x60a423fffe621504/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> ZDO - extRouteDisc - {"dstAddr":22672,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)'))'
error 2023-09-01 07:42:19: Failed to read state of 'tuinlamp_schuur_links' after reconnect (Read 0xa4c13856c7fede8e/1 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> ZDO - extRouteDisc - {"dstAddr":53067,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)')))

adekloet, 2023-09-01:
Is a know problem on that firmware. Have a read through https://github.com/Koenkk/Z-Stack-firmware/issues/439#issuecomment-1703119477

Koenkk, 2023-09-02:
Let's continue this in https://github.com/Koenkk/Z-Stack-firmware/discussions/474

closed 2023-09-02 by Koenkk