Koenkk/zigbee2mqtt #18812: Fail to start after updated to 1.32.1 with error "TypeError: Cannot read properties of undefined (reading 'sync')"

juantxorena, 2023-09-01:
### What happened?

After updated to 1.32.1, z2m fail to start with the following log:

C:\Users\Administrator\node_modules\zigbee2mqtt\lib\util\logger.ts:145
directories.forEach((dir) => {
^
TypeError: Cannot read properties of undefined (reading 'sync')
at C:\Users\Administrator\node_modules\zigbee2mqtt\lib\util\logger.ts:147:20
at Array.forEach ()
at cleanup (C:\Users\Administrator\node_modules\zigbee2mqtt\lib\util\logger.ts:145:21)
at Object.logOutput (C:\Users\Administrator\node_modules\zigbee2mqtt\lib\util\logger.ts:160:9)
at Controller.start (C:\Users\Administrator\node_modules\zigbee2mqtt\lib\controller.ts:93:16)
at start (C:\Users\Administrator\node_modules\zigbee2mqtt\index.js:107:22)�

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

unknown

### Adapter

CC2652P

juantxorena, 2023-09-01:
I cannot reopen the bug #18275, but it still happens.

william-sy, 2023-09-01:
You should be able to try 1.33 in a bit

juantxorena, 2023-09-06:
Sorry for the delay, I could finally try it. The problem still happens (I'm in linux, btw, using nodesource 18 as the javascript distro).

The problem is obviously the upgraded dependency from rifraf, that was commited here in the logger.ts file: b3c8866.

I don't know why it happens to so few people, I'll gladly help to fix it with the little time I have.

More details:

- Running on a lxc container on proxmox, debian 12 based (with mosquitto, openhab, and some other things in the same container).
- user and group of all the files of the zigbee2mqtt is "zigbee2mqtt". The systemd service that launches it has the correct user.
- As I said, nodesource 18 for debian (tbh I don't know why I running 18 instead of 16, but I've just tried 16 and I get the same problem).

juantxorena, 2023-09-06:
Because the error appeared with that change and because the error message points to the exact line where that rimraf package is used

I know that most of the people it works, but for the purposes of this bug, in my case (and in the other bug reporter's case), that particular change seems to be the cause.

Koenkk, 2023-09-06:
Did you do a `npm ci` in the z2m folder after updating?

juantxorena, 2023-09-06:
Yes, of course. I also tried removing node_modules dir before running npm ci just in case, with the same result. 

Maybe tomorrow I have time to spin up a new container and start from scratch, to see what's what.

Koenkk, 2023-09-06:
And also a `npm build`?

juantxorena, 2023-09-07:
> And also a `npm run build`?

This actually worked, thanks. Maybe the documentation should be updated [here](https://www.zigbee2mqtt.io/guide/installation/01_linux.html#installing) and [here](https://www.zigbee2mqtt.io/guide/installation/01_linux.html#for-later-update-zigbee2mqtt-to-the-latest-version)? I can made the pull request if needed.

Koenkk, 2023-09-07:
yes please make a PR!

closed 2023-09-08 by Koenkk