Koenkk/zigbee2mqtt #18821: [Wrong device]: IKEA LED1923R5 identified as IKEA LED1925G6

Joakim-T, 2023-09-01:
### Link

https://www.zigbee2mqtt.io/devices/LED1925G6.html#ikea-led1925g6

### Model

https://www.zigbee2mqtt.io/devices/LED1923R5.html#ikea-led1923r5

### Description

Same as IKEA LED1923R5

### Vendor

Same as IKEA LED1923R5

### Picture (link)

Same as IKEA LED1923R5

### Database entry

Cant find on HA OS

### Notes

Just bought 3 IKEA LED1923R5 lights and paired them. 2 out of 3 pair correctly as IKEA LED1923R5. One pairs as IKEA LED1925G6.
Have tried removing, resetting and re-adding multiple times.
Seems to work fine, just identifies as wrong model.

No idea where to find data/database.db in HA OS.

Interview of device that pairs as the **incorrect** model...

`info  2023-09-01 16:56:28: Device '0x90395efffe537578' joined
info  2023-09-01 16:56:28: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x90395efffe537578","ieee_address":"0x90395efffe537578"},"type":"device_joined"}'
info  2023-09-01 16:56:28: MQTT publish: topic 'homeassistant/light/0x90395efffe537578/light/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"brightness":true,"brightness_scale":254,"color_mode":true,"command_topic":"zigbee2mqtt/0x90395efffe537578/set","device":{"identifiers":["zigbee2mqtt_0x90395efffe537578"],"manufacturer":"IKEA","model":"TRADFRI LED bulb E14 470 lumen, opal, dimmable, white spectrum, color spectrum (LED1925G6)","name":"0x90395efffe537578","sw_version":"1.0.021"},"effect":true,"effect_list":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"max_mireds":454,"min_mireds":250,"name":"0x90395efffe537578","schema":"json","state_topic":"zigbee2mqtt/0x90395efffe537578","supported_color_modes":["xy","color_temp"],"unique_id":"0x90395efffe537578_light_zigbee2mqtt"}'
info  2023-09-01 16:56:28: MQTT publish: topic 'homeassistant/select/0x90395efffe537578/effect/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x90395efffe537578/set/effect","device":{"identifiers":["zigbee2mqtt_0x90395efffe537578"],"manufacturer":"IKEA","model":"TRADFRI LED bulb E14 470 lumen, opal, dimmable, white spectrum, color spectrum (LED1925G6)","name":"0x90395efffe537578","sw_version":"1.0.021"},"enabled_by_default":false,"icon":"mdi:palette","name":"0x90395efffe537578 effect","options":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"unique_id":"0x90395efffe537578_effect_zigbee2mqtt"}'
info  2023-09-01 16:56:28: MQTT publish: topic 'homeassistant/sensor/0x90395efffe537578/power_on_behavior/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x90395efffe537578"],"manufacturer":"IKEA","model":"TRADFRI LED bulb E14 470 lumen, opal, dimmable, white spectrum, color spectrum (LED1925G6)","name":"0x90395efffe537578","sw_version":"1.0.021"},"enabled_by_default":false,"entity_category":"config","icon":"mdi:power-settings","name":"0x90395efffe537578 power on behavior","state_topic":"zigbee2mqtt/0x90395efffe537578","unique_id":"0x90395efffe537578_power_on_behavior_zigbee2mqtt","value_template":"{{ value_json.power_on_behavior }}"}'
info  2023-09-01 16:56:28: MQTT publish: topic 'homeassistant/select/0x90395efffe537578/power_on_behavior/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0x90395efffe537578/set/power_on_behavior","device":{"identifiers":["zigbee2mqtt_0x90395efffe537578"],"manufacturer":"IKEA","model":"TRADFRI LED bulb E14 470 lumen, opal, dimmable, white spectrum, color spectrum (LED1925G6)","name":"0x90395efffe537578","sw_version":"1.0.021"},"entity_category":"config","icon":"mdi:power-settings","name":"0x90395efffe537578 power on behavior","options":["off","on","toggle","previous"],"state_topic":"zigbee2mqtt/0x90395efffe537578","unique_id":"0x90395efffe537578_power_on_behavior_zigbee2mqtt","value_template":"{{ value_json.power_on_behavior }}"}'
info  2023-09-01 16:56:28: MQTT publish: topic 'homeassistant/sensor/0x90395efffe537578/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x90395efffe537578"],"manufacturer":"IKEA","model":"TRADFRI LED bulb E14 470 lumen, opal, dimmable, white spectrum, color spectrum (LED1925G6)","name":"0x90395efffe537578","sw_version":"1.0.021"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0x90395efffe537578 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x90395efffe537578","unique_id":"0x90395efffe537578_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info  2023-09-01 16:56:28: MQTT publish: topic 'homeassistant/sensor/0x90395efffe537578/update_state/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x90395efffe537578"],"manufacturer":"IKEA","model":"TRADFRI LED bulb E14 470 lumen, opal, dimmable, white spectrum, color spectrum (LED1925G6)","name":"0x90395efffe537578","sw_version":"1.0.021"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:update","name":"0x90395efffe537578 update state","state_topic":"zigbee2mqtt/0x90395efffe537578","unique_id":"0x90395efffe537578_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
info  2023-09-01 16:56:28: MQTT publish: topic 'homeassistant/binary_sensor/0x90395efffe537578/update_available/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x90395efffe537578"],"manufacturer":"IKEA","model":"TRADFRI LED bulb E14 470 lumen, opal, dimmable, white spectrum, color spectrum (LED1925G6)","name":"0x90395efffe537578","sw_version":"1.0.021"},"device_class":"update","enabled_by_default":false,"entity_category":"diagnostic","name":"0x90395efffe537578 update available","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x90395efffe537578","unique_id":"0x90395efffe537578_update_available_zigbee2mqtt","value_template":"{{ value_json['update']['state'] == \"available\" }}"}'
info  2023-09-01 16:56:28: MQTT publish: topic 'homeassistant/update/0x90395efffe537578/update/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/bridge/request/device/ota_update/update","device":{"identifiers":["zigbee2mqtt_0x90395efffe537578"],"manufacturer":"IKEA","model":"TRADFRI LED bulb E14 470 lumen, opal, dimmable, white spectrum, color spectrum (LED1925G6)","name":"0x90395efffe537578","sw_version":"1.0.021"},"device_class":"firmware","entity_picture":"https://github.com/Koenkk/zigbee2mqtt/raw/master/images/logo.png","latest_version_template":"{{ value_json['update']['latest_version'] }}","latest_version_topic":"zigbee2mqtt/0x90395efffe537578","name":"0x90395efffe537578","payload_install":"{\"id\": \"0x90395efffe537578\"}","state_topic":"zigbee2mqtt/0x90395efffe537578","unique_id":"0x90395efffe537578_update_zigbee2mqtt","value_template":"{{ value_json['update']['installed_version'] }}"}'
info  2023-09-01 16:56:28: Starting interview of '0x90395efffe537578'
info  2023-09-01 16:56:28: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x90395efffe537578","ieee_address":"0x90395efffe537578","status":"started"},"type":"device_interview"}'
info  2023-09-01 16:56:38: MQTT publish: topic 'zigbee2mqtt/Knapp Vardagsrum Tak', payload '{"action":null,"consumption":0.17,"current":0.02,"device_temperature":28,"energy":0.17,"flip_indicator_light":null,"led_disabled_night":false,"linkquality":204,"operation_mode_left":"decoupled","operation_mode_right":"decoupled","power":2.79,"power_outage_count":5,"power_outage_memory":true,"state":"ON","state_left":"ON","state_right":"ON","voltage":235.69}'
info  2023-09-01 16:56:38: Successfully interviewed '0x90395efffe537578', device has successfully been paired
info  2023-09-01 16:56:38: Device '0x90395efffe537578' is supported, identified as: IKEA TRADFRI LED bulb E14 470 lumen, opal, dimmable, white spectrum, color spectrum (LED1925G6)
[... cut ...]
info  2023-09-01 16:53:09: MQTT publish: topic 'zigbee2mqtt/0x4c5bb3fffef80df4', payload '{"color_mode":"xy","linkquality":204,"power_on_behavior":null,"state":null,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info  2023-09-01 16:53:09: MQTT publish: topic 'zigbee2mqtt/0x4c5bb3fffef80df4', payload '{"color_mode":"xy","linkquality":204,"power_on_behavior":null,"state":null,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info  2023-09-01 16:53:09: Successfully configured '0x4c5bb3fffef80df4'`

Koenkk, 2023-09-01:
These devices seem to use the same modelID, I've changed the model to `LED1923R5/LED1925G6`

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-09-01 by Koenkk

DennisJohnsen, 2023-10-03:
Not sure if this is related, but now my LED1923R5 is not showing up with a picture in Zigbee2mqtt frontend, but since i comment here, is because https://www.zigbee2mqtt.io/devices/LED1923R5.html#ikea-led1923r5 is now showing up as 404. So not sure what changed.

mrskycriper, 2024-03-29:
@Joakim-T Hello, can you please share Zigbee models of those different lamps? I'm doing a bit of a cleanup and it seems that `LED1925G6` is an E12/E14/E17 model. While `LED1923R5` is associated with GU10 bulbs. In fact, LED1925G6 exist in a database separately.

@Koenkk is there any particular reason, as to why the model name was changed? The Zigbee models seem to be unaffected, so it should not have changed the model detection here, just renamed existing GU10 bulbs.

This whole situatuon is looking really confusing to me.

Koenkk, 2024-03-31:
@mrskycriper if they have different Zigbee models, we can separate them again.

mrskycriper, 2024-03-31:
> @mrskycriper if they have different Zigbee models, we can separate them again.

The other model is already present in the database. I'm really confused as to why it was written twice.

Koenkk, 2024-03-31:
It's strange, either the fw @Joakim-T is bugged or it's a one time mess up from IKEA