Koenkk/zigbee2mqtt #18828: Iris 3460-L Action not happening

KyleStilkey, 2023-09-01:
### What happened?

Clicking or holding the button registers no actions. I have the device synced without issues with the latest release and fixed that was pushed out today. Now I can register my buttons and it pulls the data from the device like Temp and Battery level.

What I do notice is that clicking or holding the button doesn't register any action for the devices, the button state will stay at NONE no matter if I am holding or clicking the buttons.

I'm unsure if there is something more I can do on my end to provide more data to help. I am also running this through Home Assistant and unaware if that version might be causing a conflict or not.

Also a side note, I have brand new batteries in these buttons and both are registering as 27% and might also be pulling data incorrectly.

### What did you expect to happen?

An action and or hold action should function.

### How to reproduce it (minimal and precise)

Click or hold button and a action should display.

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff zigbee 3.0 Dongle E

### Debug log

I am not seeing any log issues with this, no errors or action reports happening when pushing the buttons on these devices.

KyleStilkey, 2023-10-04:
Still an issue.

KyleStilkey, 2023-11-04:
Still an issue

annbob, 2023-12-20:
Still an issue

annbob, 2023-12-21:
Thank you to @shackrat, [Iris 3460-L Unsupported Attribute Errors #15971](https://github.com/Koenkk/zigbee2mqtt/issues/15971#issuecomment-1372777469), for a fix to the Iris 3460-L Button problem.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
  {
      zigbeeModel: ['3460-L'],
      model: '3460-L',
      vendor: 'Iris',
      description: 'Smart button',
      fromZigbee: [fz.command_on, fz.command_off, fz.battery, fz.temperature],
      toZigbee: [],
      exposes: [e.battery(), e.temperature(), e.action(['on', 'off'])],
      meta: {battery: {voltageToPercentage: '3V_2100'}},
      configure: async (device, coordinatorEndpoint, logger) => {
          const endpoint = device.getEndpoint(1);
          await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genPowerCfg', 'msTemperatureMeasurement']);
          await reporting.batteryVoltage(endpoint);
          await reporting.temperature(endpoint);
      },
  },
];
```

closed 2024-06-19 by KyleStilkey