Koenkk/zigbee2mqtt #18830: Error after update to 1.33.0-1

DChe47, 2023-09-01:
### What happened?

Today after update to 1.33.0-1 I see the many errors from device TuYa [TS0601_smart_CO_air_box] _TZE200_7bztmfm1

`2023-09-01 19:01:49.197 ERROR (MainThread) [homeassistant.util.logging] Exception in state_message_received when handling msg on 'zigbee2mqtt/Датчик CO': '{"carbon_monoxide":"OFF","co":0,"last_seen":"2023-09-01T16:01:49.122Z","linkquality":109}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 44, in wrapper
    msg_callback(msg)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/binary_sensor.py", line 243, in state_message_received
    self._config[CONF_NAME],
    ~~~~~~~~~~~~^^^^^^^^^^^
KeyError: 'name'
`

### How to reproduce it (minimal and precise)

I have this error happen several times a second. Probably every time the sensor gives data

### Zigbee2MQTT version

1.33.0-1

### Adapter firmware version

20220302

### Adapter

zStack3x0

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12501238/log.txt)

mpuff, 2023-09-01:
i am getting error now like this:

`
Warning 2023-09-01 23:30:13Failed to ping 'Osram_E27_Multicolor_GF_Bath' (attempt 1/2, Read 0x7cb03eaa00ae2deb/3 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Warning 2023-09-01 23:30:34Failed to ping 'Osram_E27_Multicolor_GF_Bath' (attempt 2/2, Read 0x7cb03eaa00ae2deb/3 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Warning 2023-09-01 23:30:58Failed to ping 'Osram_E27_Multicolor_Outdoor_5' (attempt 1/2, Read 0x7cb03eaa00aa26f4/3 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 310 - 3 - 102 - 0 - 1 after 10000ms))
Warning 2023-09-01 23:31:28Failed to ping 'Ikea_Pilskott_Livingroom' (attempt 1/2, Read 0x385cfbfffe638bdd/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Warning 2023-09-01 23:31:57Failed to ping 'Osram_GU10_FF_Bath_Spot_1' (attempt 1/1, Read 0xf0d1b8000012ca8c/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
Warning 2023-09-01 23:32:22Failed to ping 'Osram_GU10_FF_Bath_Spot_2' (attempt 1/1, Read 0xf0d1b8000012ade1/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 49791 - 1 - 113 - 0 - 1 after 10000ms))
`

Koenkk, 2023-09-02:
Are you running at least Home Assistant 2023.8.0?

rchiileea, 2023-09-02:
I rollled back for some reason my ikea blinds stopped responding after update

s0ftice, 2023-09-02:
Ikea devices lost their control entities with the 1.33 update.

DChe47, 2023-09-02:
My Home Assistant version:
Home Assistant 2023.8.4
Supervisor 2023.08.1
Пользовательский интерфейс 20230802.1 - latest

Bill-Dung, 2023-09-03:
Same error here, but only for Tuya thermostats TS0601 (_TZE200_z1tyspqw)

rchiileea, 2023-09-03:
> Ikea devices lost their control entities with the 1.33 update.

not for, me my blinds are fine in the 1.33 update, upgrading to 1.33.1 causes them not to respond to any up, down or open close . they litterally stop working on the main feature (battery is still reported though) but all other controls are failed.

DChe47, 2023-09-03:
> Are you running at least Home Assistant 2023.8.0?

I use:
Home Assistant 2023.8.4
Supervisor 2023.08.1
User Interface 20230802.1 - latest

DChe47, 2023-09-03:
Warnings appeared simultaneously with the errors:
`Logger: homeassistant.helpers.template
Source: helpers/template.py:735
First occurred: 08:53:25 (9753 occurrences)
Last logged: 13:12:34

Template variable warning: 'dict object' has no attribute 'carbon_monoxide' when rendering '{{ value_json.carbon_monoxide }}'
Template variable warning: 'dict object' has no attribute 'current_heating_setpoint' when rendering '{{ value_json.current_heating_setpoint }}'`

SVH-Powel, 2023-09-04:
> Ikea devices lost their control entities with the 1.33 update.

Yes, same here. I had to roll back to 1.32.2-1

s0ftice, 2023-09-04:
> Yes, same here. I had to roll back to 1.32.2-1

I got it first fixed by disabling `Persist cache state` and `Send cached state on startup` under Z2M → Settings → Advanced, followed by a HA restart. However, this morning the issue was back. I rebooted the entire VM (and re-enabled these two settings) which seems to have worked for now - all IKEA entities are available and useable in HA again for now...

stigvi, 2023-09-04:
Forget the message I wrote earlier today (SVH-Powel: Yes, same here. I had to roll back to 1.32.2-1)

I am not sure what the problem is, but it happened with 1.32.2-1, too. I updated Mosquitto broker in HA so maybe that is causing problems and not zigbee2mqtt

rchiileea, 2023-09-05:
> Forget the message I wrote earlier today (SVH-Powel: Yes, same here. I had to roll back to 1.32.2-1)
> 
> I am not sure what the problem is, but it happened with 1.32.2-1, too. I updated Mosquitto broker in HA so maybe that is causing problems and not zigbee2mqtt

mine is fine with mosquitto, but I did a partial restore from a full backup of zigbee2mqtt as the backup of just zigbee2mqtt still had issues.

bcdonadio, 2023-09-05:
I can confirm that after multiple attempts to update to v1.33.0 from v1.32.1, I can't connect to my Mosquitto v2.0.17 broker (with TLSv1.3, OpenSSL 3.0.9-2, MQTTv5), but everything works alright after a backup restore. Mosquitto was restarted multiple times, other clients are working just fine, the certificate isn't expired and all involved clocks are synchronized within 50ms or less. All involved systems are fully up to date Fedora 38 OSes on x86_64.

I will try to get a stack trace soon.

closed 2023-10-13 by github-actions[bot]

Nico04, 2024-09-06:
I have a similar issue for months now, stuck at version 1.32.2-1.
@bcdonadio did you find a solution on how to update ?