Koenkk/zigbee2mqtt #18832: TS0202 Tamper entity gone even present on state

dinde, 2023-09-01:
### What happened?

Suddenly all tamper entities on PIR disappeared.

Tested on:
_TZ3000_msl6wxk9
_TZ3000_6ygjfyll
Expose tab
[image]

State tab
[image]

### What did you expect to happen?

Tamper should be on expose and report entities on HA

### How to reproduce it (minimal and precise)

Just upgrade MQTT Z2M and HA to latest.
Tamper will disappear on PIR.

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20230507

### Adapter

Sonoff ZBDongle-P

Koenkk, 2023-09-02:
This is due to https://github.com/Koenkk/zigbee-herdsman-converters/pull/6062, I will revert it for the TS0202 @Chartreusito

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-09-02 by Koenkk

Chartreusito, 2023-09-02:
@dinde what were your use case for the tamper entity ? I've got one of both _TZ3000_msl6wxk9 and  _TZ3000_6ygjfyll and neither have any tamper switch to trigger this entity, which stays false whatever i do.

ClearlyDazed, 2023-09-02:
The tamper entity has also disappeared for TuYa TS0202 manufacturer _TZ3000_otvn3lne.
MQTT does report its state:
{
    "battery": 100,
    "battery_low": false,
    "last_seen": "2023-09-02T14:41:12-05:00",
    "linkquality": 91,
    "occupancy": false,
    "tamper": false,
    "voltage": 2200
}

I haven't opened the PIR case to verify that a tamper switch exists, but why eliminate it since it is reported.

dinde, 2023-09-02:
@Chartreusito : Tamper is useful when you use PIR with alarm system. It ensures nobody modified the location of the sensor neither jammed it by opening it.
PIR tamper works with the 2 devices I have (at least used to work)

@Koenkk Thanks for handling this issue fast !

geekman, 2023-09-03:
To be fair, I have the `_TZ3000_6ygjfyll` device and it has **no physical tamper switch**. So even though the "tamper" property exists in the code, it will never change status because there's nothing in the device that will ever trigger it. Unless it's triggered by something like changing the battery.

You can find [teardown photos of it here](https://smarthomescene.com/reviews/cheapest-zigbee-motion-sensor-for-your-smart-home/), and there's only one switch, which is the reset/pair button.

[image]

Chartreusito, 2023-09-03:
@geekman that was my point, i know those devices expose a tamper entity, but in my humble opinion, it's a fake entity as it isn't related to anything physical nor its status changes. I opened both of them with the same result as you.
@dinde I know what tamper is, and with my understanding of it, it didn't have any real usefulness on those devices. I was wondering if you managed to use it in another way.
I still thinks i was really doing the right thing with this PR, by removing an entity which is just not working for those devices.
@Koenkk i was under the impression that you regularly removed some non working entities for some devices, what's your opinion on this case ?

ClearlyDazed, 2023-09-03:
Extreme care is required to eliminate phantom entities from devices. In this case my device TuYa TS0202 manufacturer _TZ3000_otvn3lne is totally different physically and internally yet it was included in the change to eliminate the tamper entity. I have no plans on confirming the tamper switch exists and being TuYa, it would be a daunting task to confirm all the devices. Please be careful and change entities to those specific devices that have been inspected only.

dinde, 2023-09-03:
@ClearlyDazed IIRC my tamper entities were working on these devices. 
Not sure if it's with a specific switch or the battery disconnected.
Since the back cover is stuck on wall and you need to unscrew to remove the sensor, I believe the tamper sensor works with the battery being unplugged.
I'll wait the next stable release to test as I don't want to mess going to dev branch then reverting.

Chartreusito, 2023-09-04:
@ClearlyDazed To be fair, my [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6062) only included _TZ3040_bb6xaihh, _TZ3000_msl6wxk9 and _TZ3000_msl6wxk9 which i own and opened to confirm there wasn't any hidden tamper switch. When i tested them, opening them physically disconnects the battery as it's holded by the cover.