Koenkk/zigbee2mqtt #18839: Schneider Electric 2GANG/SWITCH/2 MEG5126-0300 / MEG5162

linuxrrze, 2023-09-02:
### What happened?

I just got excited when reading the release notes of version 1.33.0 mentioning

[#18595](https://github.com/Koenkk/zigbee2mqtt/issues/18595) MEG5126-0300 Schneider Electric Merten MEG5165 PlusLink relais insert with Merten Wiser System M push button (2fold) ([@tech2mar](https://github.com/tech2mar))

I was hoping to get my 2-gang Schneider electric switch (MEG5126-0300) connected to a 2-way relay (MEG5162) working - finally.

However its behavior is still the same as before the update:

I can turn on/off both switch (lights) using zigbee2mqtt, however only one of them does have an effect on one of the relays.

The mentioned path seems to affect only the model  '2GANG/SWITCH/1'.
So I guess '2GANG/SWITCH/2' is relevant for my setup, right?

Is there anything I can do to help get this setup working?

### What did you expect to happen?

I expected both relays to turn on/off.

### How to reproduce it (minimal and precise)

Connect 2-way switch to 2-way relay.
Turn on/off both relays via zb2mqtt.

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20230507

### Adapter

Slaesh

linuxrrze, 2023-09-02:
Here are the links to the products (sorry, German only):
https://merten.de/produkt/meg5126-0300-connected-taster-modul-2fach-system-m.html
https://merten.de/produkt/meg5162-0000-relais-schalt-einsatz-2fach.html

closed 2023-10-10 by github-actions[bot]

Highflyer100, 2024-02-17:
Hey, is this still open?
Working on the same problem on my 2GANG/DIMMER/2  MEG 5126-0300   with double relay   and still dealing with
- No converter available for 'state_3' ("ON")
- No converter available for 'state_4' ("ON")
Any help?

Nailik, 2024-03-15:
@Highflyer100 I just opened a pr to support MEG5126-0300 with MEG5172-0000.

@linuxrrze you probably need to create a custom converter and try to fix your issue there and open a PR afterwards.