Koenkk/zigbee2mqtt #18840: BRT-100 useless after OTA update

carlos-algms, 2023-09-02:
### What happened?

I was suggested to update my device from version `67` to `192`.
After the update succeeded the device stopped working via the web UI. 

I tried to force remove it to add it again, but holding the `-` (minus) button doesn't make the antenna icon blink, (pairing mode), but instead, it keeps reducing the target temperature. 
Also, the LEDs don't turn off, just dim. 

A second device had a similar issue, the antenna icon remains still after holding `-`, but it doesn't register, and the screen turns off normally. 

### What did you expect to happen?

the update to happen, and the device to work normally. 

### How to reproduce it (minimal and precise)

just update the device. 

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20221226

### Adapter

Sonoff Zigbee dongle plus

carlos-algms, 2023-10-09:
Bump to keep this alive. 
It's a pretty expensive issue to happen as it kills the device.

Koenkk, 2023-10-09:
Removed OTA support for this device, thanks for reporting.

closed 2023-10-09 by Koenkk

gyver, 2023-10-28:
As it seems to be a recurrent problem when the manufacturer change revisions, may I suggest updating the https://www.zigbee2mqtt.io/devices/BRT-100-TRV.html page ? A warning that firmware updates have been known to be risky would go a long way to protect users with perhaps more details (the problem might occur both with new devices and old devices with old versions of Z2M if I understand correctly) ?

To give more context...

Before purchasing any Zigbee device I check the device pages for compatibility and any quirks. I did it for my recent purchases of BRT-100-TRV, didn't see any warning and started updating their firmware recently.
This killed 2 out of my 5 (2 were probably OK because they were an older version and I didn't try to update the last one). I thought I had a lemon with the first one to fail as updating a firmware is normally something safe to do (not sure about Zigbee specifically but in other domains usually devices refuse to flash incompatible firmwares). I only looked for people with similar problems after bricking the second device and here am I...

Impact123, 2023-10-29:
> may I suggest updating the https://www.zigbee2mqtt.io/devices/BRT-100-TRV.html page

Every docs' page has a edit button at the bottom.
[image]

You could create a PR with a warning linking to this issue.