Koenkk/zigbee2mqtt #18842: Missing devices already paired

jd1900, 2023-09-02:
### What happened?

After several months using Z2M, three devices dissapeared from Z2M. They were all [067776](https://www.zigbee2mqtt.io/devices/067776.html#legrand-067776), but only a few of them dissapeared, some other of the same model didn't.

They still appear on devices.yaml and groups.yaml, but they don't appear on database.db anymore and they don't show on the UI or Home Assistant anymore.

I had them bound to some groups, and when interacting with those groups, they still react, but I can't control them individually any more.

Is there any way to force zigbee2mqtt to add them back or query them back to add them without having to re-pair them?
They are not very accessible (physically).

What may have done them dissapearing?

### What did you expect to happen?

Don't loose already joined devices.

### How to reproduce it (minimal and precise)

No idea.

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20230507

### Adapter

zzh!

Danacus, 2023-09-03:
Same issue here. Update to zigbee2mqtt version 1.33 appears to be the culprit. Downgrading solves the issue for now.

Danacus, 2023-09-03:
@jd1900 zigbee2mqtt 1.33 only works with home assistant 2023.08 or later. Either downgrade zigbee2mqtt or upgrade home assistant and the issue should be resolved.

jd1900, 2023-09-03:
> @jd1900 zigbee2mqtt 1.33 only works with home assistant 2023.08 or later. Either downgrade zigbee2mqtt or upgrade home assistant and the issue should be resolved.

In my case, it has nothing to do with it or with 1.33.
It happened before that, and my Home Assistant was updated to 2023.08 before 1.33.

Danacus, 2023-09-03:
Ah, I'm sorry, my bad! Than this issue is unrelated. Hopefully you'll be able to resolve your issue as well.

Ra72xx, 2023-10-09:
I also have the problem that suddenly some devices are missing from the device list without any warning. This happened several times already after using z2m for about two weeks now?!

alexh3o, 2024-04-08:
Still a problem. I also have devices silently disappearing and I need to repair them. Happens less than once a month. They are not in groups. I'll investigate more next time because I didn't look at devices and database files at the time. Last time was 13th of march on Aqara WSDCGQ11LM.

JCinside13, 2024-09-12:
I have the same problem I last without warning, 3 devices ( 2 SONOFF SNZB-01 and 1 Aqara MCCGQ11LM).

closed 2025-05-15 by github-actions[bot]