Koenkk/zigbee2mqtt #18846: [Wrong device]: Gledopto GL-B-004P

dankarization, 2023-09-03:
### Link

https://aliexpress.ru/item/1005003189229651.html?aff_fcid=838cd87b9730419a949bb351a9225daf-1635037251054-04238-_9gTCE1&tt=CPS_NORMAL&aff_fsk=_9gTCE1&aff_platform=shareComponent-detail&sk=_9gTCE1&aff_trace_key=838cd87b9730419a949bb351a9225daf-1635037251054-04238-_9gTCE1&terminal_id=c60aa1c2bd3d4f80907b0cc2716fb935&gatewayAdapt=glo2rus&sku_id=12000024577823460

### Model

GL-B-004P

### Description

Lightbulb

### Vendor

Gledopto

### Picture (link)

-

### Database entry

-

### Notes

This device is currently supported in Z2M with all info being right, but it indicates in light settings, that RGB colors can be changed, which is not true. This light only supports CCT

oat1, 2023-09-04:
The same problem with https://www.zigbee2mqtt.io/devices/GL-B-002P.html#gledopto-gl-b-002p
It only supports CCT

Koenkk, 2023-09-04:
Removed for both GL-B-002P and GL-B-004P .

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-09-04 by Koenkk