Koenkk/zigbee2mqtt #18847: [New device support]: Mercator SSRM-ZB - rotary dimmer

easyoz, 2023-09-03:
### Link

https://zigbee.blakadder.com/Mercator_SSWRM-ZB.html

### Database entry

 {"id":16,"type":"Router","ieeeAddr":"0x003c84fffe09a756","nwkAddr":20797,"manufId":4098,"manufName":"_TZE200_a0syesf5","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_a0syesf5","powerSource":1,"zclVersion":3,"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":83,"stackVersion":0,"hwVersi

### Comments

Tried using the templates but still showing unknown

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_a0syesf5']),
    model: 'TS0601_Dimmer',
    description: 'Dimmer Switch',
    vendor: 'Mercator',
    ota: ota.zigbeeOTA,
    extend: tuya.extend.switch({
        electricalMeasurements: true, powerOutageMemory: true, indicatorMode: true, childLock: true, endpoints: ['l1', 'l2']}),
    whiteLabel: [
        tuya.whitelabel('Mercator', 'SSRM-ZB', 'Mercator Dimmer Switch', ['_TZE200_a0syesf5']),
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await reporting.rmsVoltage(endpoint, {change: 5});
        await reporting.rmsCurrent(endpoint, {change: 50});
        await reporting.activePower(endpoint, {change: 10});
        await reporting.currentSummDelivered(endpoint);
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        device.save();
    },
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    meta: {multiEndpoint: true},
};

module.exports = definition;
```

cloudbr34k84, 2023-09-06:
i would be interested in this getting developed. I have another zigbee device im trying to make a converter for, but no luck

Koenkk, 2023-09-06:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/a481e1f034099f5bfa7f48f0e16d7db3

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

easyoz, 2023-09-08:
Thanks you so much, have got file and will tomorrow

jpg0, 2023-09-08:
I just installed one and these and tested the linked external converter; it now works fine for me.

Koenkk, 2023-09-09:
@jpg0 thanks, added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-09-09 by Koenkk

bjammin, 2023-11-22:
I just got one of these to test, and although it's working, I am getting warnings like this spammed in the log whenever the brightness is changed (either with physical dial, slider on "exposes" tab,  or in HA.

Zigbee2MQTT:warn  2023-11-22 13:32:38: zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #2 from 0x84ba20fffe97ca04 with raw data '{"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,55]}}': type='commandDataResponse', datatype='value', value='55', known DP# usage: ["heatingSetpoint","coverPosition","eardaDimmerLevel","moesHold","moesSheatingSetpoint","silvercrestChangeMode","tuyaSabCO2","tuyaSahkMP25","tuyaSabCO","moes105DimmerLevel1","trsSensitivity","trsfSensitivity","tvMode","nousHumidity","tthHumidity","tIlluminanceLux","evanellMode","AM02PercentControl","x5hMode","connecteMode","tshpscSensitivity","alectoSmokeValue","zsMode"]

Also, the min brightness and max brightness on the "exposes" tan do not seem to save or get applied in any way.

Apart from this, it's working OK.

I am using Zigbee2Mqtt 1.33.2-1 with no external converters.

Any chance of fixing either of those issues?