Koenkk/zigbee2mqtt #18848: BTicino N4003C OTA not working

cybergatto, 2023-09-03:
### What happened?

Hello,
   I've got a couple of btcino N4003C on my network.
I see that now OTA is supported, but it seems not to be working:

```
zigbee2mqtt      | Zigbee2MQTT:info  2023-09-03 15:17:07: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"ledScalaSotto"},"error":"Failed to check if update available for 'ledScalaSotto' (Device didn't respond to OTA request)","status":"error","transaction":"jx3qe-10"}'
zigbee2mqtt      | Zigbee2MQTT:error 2023-09-03 15:17:07: Failed to check if update available for 'ledScalaSotto' (Device didn't respond to OTA request)
zigbee2mqtt      | Zigbee2MQTT:debug 2023-09-03 15:17:07: Error: Device didn't respond to OTA request
zigbee2mqtt      |     at requestOTA (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:202:15)
zigbee2mqtt      |     at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:259:25)
zigbee2mqtt      |     at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:203:45)
```

It also seems to me that those device are acting as a router so badly that frequently cascade devices are dropping commands.

### What did you expect to happen?

firmware to be checked for updates

### How to reproduce it (minimal and precise)

check for new OTA in OTA page on any N4003C device

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20230507

### Adapter

Sonoff ZBDongle-P

### Debug log

```
zigbee2mqtt      | Zigbee2MQTT:debug 2023-09-03 15:24:23: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"luciCameraLore","transaction":"jx3qe-11"}'
zigbee2mqtt      | Zigbee2MQTT:info  2023-09-03 15:24:23: Checking if update available for 'luciCameraLore'
zigbee2mqtt      | Zigbee2MQTT:debug 2023-09-03 15:24:23: Check if update available for '0x000474000117db6a' ( Light switch with neutral)
zigbee2mqtt      | Zigbee2MQTT:debug 2023-09-03 15:24:23: Using endpoint '1'
zigbee2mqtt      | Zigbee2MQTT:debug 2023-09-03 15:24:36: OTA update at 30.82%, remaining 4367.901019731649 seconds
zigbee2mqtt      | Zigbee2MQTT:info  2023-09-03 15:24:36: Update of 'buttonCameraLore' at 30.82%, ≈ 73 minutes remaining
zigbee2mqtt      | Zigbee2MQTT:info  2023-09-03 15:24:36: MQTT publish: topic 'zigbee2mqtt/buttonCameraLore', payload '{"action":null,"battery":87,"linkquality":0,"update":{"installed_version":587699761,"latest_version":604241926,"progress":30.82,"remaining":4368,"state":"updating"},"update_available":null}'
zigbee2mqtt      | Zigbee2MQTT:debug 2023-09-03 15:25:07: OTA update at 31.33%, remaining 4333.06401863354 seconds
zigbee2mqtt      | Zigbee2MQTT:info  2023-09-03 15:25:07: Update of 'buttonCameraLore' at 31.33%, ≈ 72 minutes remaining
zigbee2mqtt      | Zigbee2MQTT:info  2023-09-03 15:25:07: MQTT publish: topic 'zigbee2mqtt/buttonCameraLore', payload '{"action":null,"battery":87,"linkquality":0,"update":{"installed_version":587699761,"latest_version":604241926,"progress":31.33,"remaining":4333,"state":"updating"},"update_available":null}'
zigbee2mqtt      | Zigbee2MQTT:debug 2023-09-03 15:25:15: Received Zigbee message from 'luceComodinoLore', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 11 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:warn  2023-09-03 15:25:15: Failed to ping 'luceComodinoLore' (attempt 1/2, Read 0x0017880103d08b66/11 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
zigbee2mqtt      | Zigbee2MQTT:debug 2023-09-03 15:25:20: Received Zigbee message from 'luceComodinoLore', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 11 with groupID 0
zigbee2mqtt      | Zigbee2MQTT:info  2023-09-03 15:25:23: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"luciCameraLore"},"error":"Failed to check if update available for 'luciCameraLore' (Device didn't respond to OTA request)","status":"error","transaction":"jx3qe-11"}'
zigbee2mqtt      | Zigbee2MQTT:error 2023-09-03 15:25:23: Failed to check if update available for 'luciCameraLore' (Device didn't respond to OTA request)
zigbee2mqtt      | Zigbee2MQTT:debug 2023-09-03 15:25:23: Error: Device didn't respond to OTA request
zigbee2mqtt      |     at requestOTA (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:202:15)
zigbee2mqtt      |     at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:259:25)
zigbee2mqtt      |     at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:203:45)

```

Luca1996O, 2023-09-04:
Same issue here. I've got about 40 devices BTicino, and none of them can be updated, even if there are updates available for all of them.

For some of them update crashes during installation, for others they don't respond to OTA request almost immediately.

dragorex71, 2023-09-19:
I have almost the same problem and the update fail with this report:
`Device didn't respond to OTA request` 
Almost because sometimes the update goes well. I have 30 device and now, after hundreds of trials, only 7 have to update. 
But, some updated devices now have this firmware:

```
{
    "linkquality": 85,
    "state": "OFF",
    "update": {
        "installed_version": 3556351,
        "latest_version": 3555843,
        "state": "idle"
    },
    "update_available": false
}
```

others:

```
{
    "linkquality": 167,
    "state": "OFF",
    "update": {
        "installed_version": 54594304,
        "latest_version": 3555843,
        "state": "idle"
    },
    "update_available": false
}
```
so two different updated version: most `3556351` and a few `54594304` for the same BTicino N4003C

I also have a Legrand 412015, reported as upgradable but i have always this report

`Update failed with reason: 'invalid image'`

The same (upgradable but fail due to an `invalid image`) for the BTicino 3584C/Legrand 064888.

brunialti, 2023-10-06:
Same as #19167

giodaprile, 2023-10-15:
Hello,
same for me. Device N4003C. It appears upgradable from OTA menu with firmware version 002c. The update process starts correcly, lasts about half an hour but at the end the version remains the same.
By the way I'd like to update the firmware because Z2M does not receive the right status of the device when I turn on and off the light on whatever the deviators\inverter I switch.

IvanGamba037, 2023-12-23:
Same issue but also with device N4411C... sometimes they doesn't respond to ota request and sometimes i get invalid image

stewepylon, 2024-01-02:
It seems the issue is still there with the latest versions of HA and Z2M.
Can't upgrade some [K4003C/L4003C/N4003C/NT4003C] devices with starting firmware 001c (displayed as 1851907).
Other devices with later firmware were able to be upgraded after some trials.

Debug 2024-01-02 23:32:10Update of 'Veranda Cucina Luce' failed (Error: Timeout: device did not request any image blocks)
Info 2024-01-02 23:32:10MQTT publish: topic 'zigbee2mqtt/Veranda Cucina Luce', payload '{"action":null,"led_if_on":"ON","led_in_dark":"ON","linkquality":42,"state":"OFF","update":{"installed_version":1851907,"latest_version":2245123,"state":"available"},"update_available":null}'
Info 2024-01-02 23:32:10MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"0x000474000117a5a6"},"error":"Update of 'Veranda Cucina Luce' failed (Timeout: device did not request any image blocks)","status":"error"}'
Error 2024-01-02 23:32:10Update of 'Verana Cucina Luce' failed (Timeout: device did not request any image blocks)

mircolino, 2024-01-03:
Same problem.

When updating K4003C/L4003C/N4003C/NT4003C firmware I always the the same error:

`Error 2024-01-03 10:15:11Update of 'Track Lights First Floor Stairs' failed (Timeout: device did not request any image blocks)`

- sometimes right at the beginning ( the upgrade process doesn't even start)
- sometimes right at the end

every once in a while the update process completes successfully

max2win, 2024-02-17:
I had the same issue here too: the progress bar reaches 100% and then the upgrade procedure fails to complete.
A note in the device page here https://www.zigbee2mqtt.io/devices/K4003C_L4003C_N4003C_NT4003C.html suggests to push the button during pairing to avoid disconnect. That worked in 5 out of 5 case for me. I suppose that keeps the device on and prevents some shorter-than-needed timeout. I wanted to try if that trick works with the OTA too and indeed I made it. Here are my steps: 
1) wait patiently till the progress bar reaches 100%
2) push the switch button one time every 30s until your device turns off and reboots (you can hear no action from the inside electromechanical relay when you push the button during reboot).
3) done. 

Now I see my 0x001c N4003C upgraded to 0x0022. Hope this method is confirmed by everyone.

Z2M 1.35.1 in a Docker container, running on Raspberry PI 3B, 64-bit OS. Coordinator: Sonoff ZBDongle-P

shoom3301, 2024-06-05:
Same problem with K4003C, K4411C

[image]

donatigno, 2024-11-09:
Same issue! 

Someone have solve it and can share solution?

Thanks in advanced!

LeonardoGandini, 2024-11-18:
Same here! Has been this way for a year now, unable to update them at all

mircolino, 2024-11-18:
I remember having the same issue about one year ago.
I'm not 100% sure but I think it could be the older firmware that prevents the update.
I decided to buy a [Bticino S36GWDC gateway](https://www.amazon.it/dp/B0B12PLPZD), unpair all my N4003C and N4411C, pair them temporarily to the Bticino gateway, wait a few hours until they got upgraded and pair them back to zigbee2mqtt.
Since then I no longer have problems. Recently I upgraded all of them with zigbee2mqtt to the latest firmware (0041 & 0025) without a single error.

primeroz, 2024-11-18:
Starting with some recent version of z2mqtt I was able to update all my BTicino switches and dimmers

I do have to hit the update button a few times but it eventually worked for me

closed 2025-01-25 by github-actions[bot]