Koenkk/zigbee2mqtt #18849: Sonoff Door Sensor False Positives

BlueWings172, 2023-09-03:
### What happened?

I was using my Sonoff SNZB-04 door sensor with ZHA for months and it was working fine. I switched the zigbee2mqtt and everything seemed fine at the beginning but then I started to noticed that the sensor reports multiple close events when the door is already closed and multiple open events when the door is open to Home Assistant

I don't understand why this happens if no one touched the door. How can I stop my door sensor and HA from hallucinating?

I would appreciate any help.

Thanks

[image]

[image]

[image]

[image]

### Zigbee2MQTT version

1.32.2 commit: 1ec1e57

### Adapter firmware version

20211115

### Adapter

CC2531

BlueWings172, 2023-10-04:
Does anyone have an idea how to solve this??
My sensor became completely useless and reports false events all the time.

BlueWings172, 2023-11-04:
I would appreciate any help on this. Still having the same issue.

BlueWings172, 2024-03-18:
@KayGt3 
Unfortunately, I haven't.
It appears that zigbee2mqtt reports the battery level and then Home Assistant interprets this as a door action. so if the door is closed, it keeps reporting that door is closed every hour or so again and again and then when the door is opened, it keeps reporting the door is open multiple times.

I don't know how to disable the battery reports in z2m or how to get HA to distinguish between these battery reports and an actual door opening or closing. 

I tried to search online but couldn't find a solution. If you find one, please let me know.

Thanks

KayGt3, 2024-03-19:
This is the behaviour of mine:

[image]

I will let you know if i will find something!

closed 2024-10-17 by github-actions[bot]