Koenkk/zigbee2mqtt #18851: Setting the state of groups containing Inovelli Blue switches result in "rampRateOffToOnRemote" error

rccoleman, 2023-09-03:
### What happened?

I have a bunch of groups that include Inovelli Blue switches and I get the following error every time something tries to set the state via MQTT:

`Publish 'set' 'state' to 'DNH Group' failed: 'TypeError: Cannot read properties of undefined (reading 'rampRateOffToOnRemote')'`

The Inovelli switches do have a parameter called `rampRateOffToOnRemote`, but perhaps it's not included in a group containing them?  It doesn't seem to matter if there are only Inovelli switches in the group, or if there are other devices in it as well.  The command to the group still appears to be successful, I just get errors in the Zigbee2MQTT log whenever a group receives a command.  I haven't changed the value of that parameter, so it's set to whatever the default is.

### What did you expect to happen?

I expect the command to be executed without an error.

### How to reproduce it (minimal and precise)

Create a group with one or more Inovelli Blue switches in it and attempt to set the state of that group through MQTT (via HA, for instance).  Observe that you get those errors in the Zigbee2MQTT log.

### Zigbee2MQTT version

1.33.0-dev commit: 3285850

### Adapter firmware version

zStack3x0 20221226

### Adapter

Tubeszb USB adapter

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12508150/log.txt)

The error is here:

[image]

rccoleman, 2023-10-15:
Still an issue.

rccoleman, 2023-12-12:
Finally coming back to this, the message is clearly coming from here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/0c21893dd438aeb2b5fb26b78ee1d5b4f6bc2684/src/devices/inovelli.ts#L1072

And specifically this bit of code:

```
                        const defaultTransitionTime = await entity.read(
                            'manuSpecificInovelli',
                            ['rampRateOffToOnRemote'],
                        );
                        return {
                            state: {state: 'ON'},
                            readAfterWriteTime: transition.specified ?
                                transition.time * 100 :
                                // @ts-expect-error
                                defaultTransitionTime.rampRateOffToOnRemote * 100,
                        };
```

It happens when I turn on a Zigbee group that contains some Inovelli Blue switches, and it happens when Z2M tries to execute the code above on the *group* and not on the individual devices.  Unfortunately, the group doesn't have the Inovelli-specific `rampRateOffToOnRemote` parameter, and the last statement throws an exception.  Here's `entity` from above, and the error that immediately follows:

```
Group {
  _events: [Object: null prototype] {},
  _eventsCount: 0,
  _maxListeners: undefined,
  databaseID: 73,
  groupID: 8,
  _members: Set(2) {
    Endpoint {
      _events: [Object: null prototype] {},
      _eventsCount: 0,
      _maxListeners: undefined,
      ID: 1,
      profileID: 260,
      deviceID: 257,
      inputClusters: [Array],
      outputClusters: [Array],
      deviceNetworkAddress: 45500,
      deviceIeeeAddress: '0xb43a31fffe33be13',
      clusters: [Object],
      _binds: [Array],
      _configuredReportings: [Array],
      meta: {},
      pendingRequests: Set(0) {},
      sendInProgress: false,
      [Symbol(kCapture)]: true
    },
    Endpoint {
      _events: [Object: null prototype] {},
      _eventsCount: 0,
      _maxListeners: undefined,
      ID: 1,
      profileID: 260,
      deviceID: 257,
      inputClusters: [Array],
      outputClusters: [Array],
      deviceNetworkAddress: 58871,
      deviceIeeeAddress: '0xb43a31fffe33ce8f',
      clusters: [Object],
      _binds: [Array],
      _configuredReportings: [Array],
      meta: {},
      pendingRequests: Set(0) {},
      sendInProgress: false,
      [Symbol(kCapture)]: true
    }
  },
  meta: {},
  [Symbol(kCapture)]: true
}
Zigbee2MQTT:error 2023-12-11 15:40:13: Publish 'set' 'state' to 'DNH Group' failed: 'TypeError: Cannot read properties of undefined (reading 'rampRateOffToOnRemote')'
```

I'm not sure what the expected behavior here should be, but the Zigbee group could contain any devices, and it doesn't seem right to look for manufacturer-specific properties on it.

Part of what confuses me is that the Inovelli-specific converter is being called *at all* for these groups.  Again, the fact that they happen to contain Inovelli devices seems independent of telling the group to turn on or off, unless it's supposed to be a superset of all its members and their specific properties.

It's also weird that this is triggered when the following is true:

`globalStore.getValue(entity, 'turnedOffWithTransition') === true`

And I'm just telling the `light` entity in Home Assistant to turn on and off:

```
zigbee2mqttp/DNH Group/set
10 most recently transmitted message(s)
Transmitted 1:58:13 PM
QoS: 0
Payload: state: 'ON'
Transmitted 2:00:45 PM
QoS: 0
Payload: state: 'OFF'
Transmitted 2:16:34 PM
QoS: 0
Payload: state: 'ON'
Transmitted 2:20:07 PM
QoS: 0
Payload: state: 'OFF'
Transmitted 3:32:15 PM
QoS: 0
Payload: state: 'ON'
Transmitted 3:34:47 PM
QoS: 0
Payload: state: 'OFF'
Transmitted 3:36:46 PM
QoS: 0
Payload: state: 'ON'
```
I have set the `DimmingSpeedUpRemote` parameter to 0 (no transition, immediate), which essentially dictates all the other dimming/ramp rate parameters, so maybe somehow that's causing Z2M to think that the light was turned off with a transition?

efaden, 2023-12-18:
Inovelli forum post I started related to this: https://community.inovelli.com/t/zigbee2mqtt-groups-giving-ramprateofftoonremote-error/15713/2

rccoleman, 2024-04-20:
The code that generated this error was removed in https://github.com/Koenkk/zigbee-herdsman-converters/pull/7115, and I'm no longer seeing it.  I'm going to close this now.

closed 2024-04-20 by rccoleman