Koenkk/zigbee2mqtt #18853: Error while starting zigbee-herdsman

richiepp, 2023-09-04:
### What happened?

I was successfully running this stick on a Raspberry Pi but I'm moving it to a Windows Machine.  I can't seem to start the Zigbee service

Error while starting zigbee-herdsman

### What did you expect to happen?

Zigbee2Mqtt to start and I'd be able to access the frontend.

### How to reproduce it (minimal and precise)

Try to start the service with

npm start

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

Unknown

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus (EFR32MG21)

### Debug log

Windows 11 64bit

Config file
`
# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

frontend: true
  
# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee
  # MQTT server URL
  server: mqtt:192.168.1.154:1883

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: \\.\COM8`
`
npm start

> zigbee2mqtt@1.32.2 start
> node index.js

Zigbee2MQTT:info  2023-09-03 20:12:47: Logging to console and directory: 'C:\Users\Rich\Downloads\zigbee2mqtt-1.32.2\zigbee2mqtt-1.32.2\data\log\2023-09-03.20-12-47' filename: log.txt
Zigbee2MQTT:info  2023-09-03 20:12:47: Starting Zigbee2MQTT version 1.32.2 (commit #unknown)
Zigbee2MQTT:info  2023-09-03 20:12:47: Starting zigbee-herdsman (0.17.2)
Zigbee2MQTT:error 2023-09-03 20:13:07: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-09-03 20:13:07: Failed to start zigbee
Zigbee2MQTT:error 2023-09-03 20:13:07: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-09-03 20:13:07: Exiting...
Zigbee2MQTT:error 2023-09-03 20:13:07: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (C:\Users\Rich\Downloads\zigbee2mqtt-1.32.2\zigbee2mqtt-1.32.2\node_modules\zigbee-herdsman\src\adapter\z-stack\adapter\zStackAdapter.ts:103:27)
    at Controller.start (C:\Users\Rich\Downloads\zigbee2mqtt-1.32.2\zigbee2mqtt-1.32.2\node_modules\zigbee-herdsman\src\controller\controller.ts:132:29)
    at Zigbee.start (C:\Users\Rich\Downloads\zigbee2mqtt-1.32.2\zigbee2mqtt-1.32.2\lib\zigbee.ts:59:27)
    at Controller.start (C:\Users\Rich\Downloads\zigbee2mqtt-1.32.2\zigbee2mqtt-1.32.2\lib\controller.ts:101:27)
    at start (C:\Users\Rich\Downloads\zigbee2mqtt-1.32.2\zigbee2mqtt-1.32.2\index.js:107:5)`

this is debug info from MQTT Explorer
'state = offline
info = {"commit":"1ec1e57","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":11,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"last_seen":"disable","legacy_api":true,…
devices = [{"definition":null,"disabled":false,"endpoints":{"1":{"bindings":[],"clusters":{"input":["genBasic","genIdentify","genOnOff","genTime","genOta","26","lightingColorCtrl"],"output":["genBasic","genIdentify","genGroups","genScenes","genOnOff","genLevelCtrl","genPollCtrl","lightingColorCtrl","msIlluminanceMeasurement","msTemperatureMeasurement","msRelativeHumidity","msOccupancySensing","ssIasZone","h…
groups = []
extensions = []
config = {"commit":"1ec1e57","coordinator":{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"},"log_level":"info","network":{"channel":11,"extendedPanID":221,"panID":6754},"permit_join":true,"version":"1.32.2"}
'
How can I get this up and running so I can access the frontend? Thanks!
formatting this post is a nightmare...

shrx, 2023-09-04:
Same issue here, started happening tonight without me updating anything.

shrx, 2023-09-04:
I managed to make it work by using backports of systemd and udev as this user suggested: https://github.com/Koenkk/zigbee2mqtt/issues/17535#issuecomment-1532241700

closed 2023-10-12 by github-actions[bot]