Koenkk/zigbee2mqtt #18856: [New device support]:

manbading, 2023-09-04:
### Link

https://www.aqara.com/controller_overview

### Database entry

{"id":13,"type":"Router","ieeeAddr":"0x54ef44100092e5e2","nwkAddr":14162,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.switch.acn047","epList":[1,2,21],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[2820,1794,5,4,3,18,0,6,64704],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.switch.acn047","appVersion":27}},"genOnOff":{"attributes":{"245":117440575,"onOff":0}},"aqaraOpple":{"attributes":{"10":1,"223":{"type":"Buffer","data":[22,33,0,0,23,33,0,0,12,33,0,0,24,33,0,0]},"229":{"type":"Buffer","data":[6,0,6,0,0,36,73,68,93,37,8,252,100,37,68,228,103,37,8,175,112,5,188,150,152,37,93]},"235":200,"247":{"type":"Buffer","data":[3,40,37,5,33,18,0,9,33,0,4,10,33,0,0,12,32,10,13,35,27,0,0,0,17,35,1,0,0,0,100,16,0,101,16,0,149,57,0,0,0,0,150,57,205,132,12,69,152,57,0,0,0,0,151,57,0,0,0,0,154,32,0]},"512":1,"515":0,"649":0,"720":0,"1303":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b747c7","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[5,4,3,18,6,64704],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"245":53957120,"onOff":1}},"aqaraOpple":{"attributes":{"512":1}}},"binds":[],"configuredReportings":[],"meta":{}},"21":{"profId":260,"epId":21,"devId":0,"inClusterList":[12],"outClusterList":[],"clusters":{"genAnalogInput":{"attributes":{"presentValue":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":27,"stackVersion":27,"hwVersion":1,"dateCode":"20230606","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1466202208},"lastSeen":1693799779280,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to control the lamp with external_converter, but only can turn it ON and OFF.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.switch.acn047'],
    model: 'LLKZMK12LM',
    vendor: 'Xiaomi',
    description: 'Aqara wireless relay controller T2',
    fromZigbee: [fz.xiaomi_basic, fz.xiaomi_power, fz.ignore_multistate_report, fz.on_off, fz.temperature],
    toZigbee: [tz.on_off, tz.LLKZMK11LM_interlock, tz.xiaomi_power, tz.xiaomi_switch_power_outage_memory],
    exposes: [e.switch()],
    // The configure method below is needed to make the device reports on/off state changes
    // when the device is controlled manually through the button on it.
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;
```

kei81131, 2023-09-11:
Hi, I made a convertor, that can control L1/ L2 separately and show energy status
However, fail to get switch_type/ flip_indicator_light/ power_outage_memory (so I commented those codes)

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.switch.acn047'], 
    model: 'LLKZMK12LM', 
    vendor: 'Xiaomi', 
    description: 'Aqara Dual Relay Module T2', 
    fromZigbee: [
        fz.on_off,
        fz.aqara_opple,
        fz.xiaomi_power
    ],
    toZigbee: [
        tz.on_off
        // tz.xiaomi_switch_operation_mode_opple,
        // tz.xiaomi_flip_indicator_light,
        // tz.xiaomi_switch_power_outage_memory
    ], 
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        return { 'l1': 1, 'l2': 2 };
    },
    exposes: [
        e.switch().withEndpoint('l1'),
        e.switch().withEndpoint('l2'),
        e.power(),
        e.current(),
        e.energy(),
        e.voltage(),
        e.device_temperature()
        // e.switch_type(),
        // e.flip_indicator_light(),
        // e.power_outage_memory()
    ],
};

module.exports = definition;

Koenkk, 2023-09-11:
- Does power outage memory work with: https://gist.github.com/Koenkk/4d8b2b11998465c7e8228d18e4a9e8e3
- Are you sure the other features are supported? LLKZMK11LM also doesn't support these.

kei81131, 2023-09-12:
> * Does power outage memory work with: https://gist.github.com/Koenkk/4d8b2b11998465c7e8228d18e4a9e8e3
> * Are you sure the other features are supported? LLKZMK11LM also doesn't support these.

Thanks for your advice. 
I've changed convertor, sadly it shows error

Those features are supported in Aqara app.
BTW, my main purpose is to get energy status, other features are not necessary to me

Again, thank you Koenkk and zigbee2mqtt is greatful

[image]
Error 2023-09-12 09:51:57Publish 'set' 'power_outage_memory' to 'Toilet double socket' failed: 'ReferenceError: manufacturerOptions is not defined'

Koenkk, 2023-09-13:
Made a mistake, updated https://gist.github.com/Koenkk/4d8b2b11998465c7e8228d18e4a9e8e3

kei81131, 2023-09-14:
> Made a mistake, updated https://gist.github.com/Koenkk/4d8b2b11998465c7e8228d18e4a9e8e3

Thanks Koenkk 
After changing converter, power_outage_momory returns another error 
Here's the error shown, thank you

[image]

Error 2023-09-14 09:35:36Publish 'set' 'power_outage_memory' to 'Toilet double socket' failed: 'Error: Write 0x54ef44100092f9b2/1 genBasic({"65520":{"value":[170,128,5,209,71,7,1,16,1],"type":65}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

Koenkk, 2023-09-14:
Seems this device works different than e.g. the LLKZMK11LM, to add support for these function sniffing with the Xiaomi gateway when setting these options is needed.

https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

kei81131, 2023-09-15:
> Seems this device works different than e.g. the LLKZMK11LM, to add support for these function sniffing with the Xiaomi gateway when setting these options is needed.
> 
> https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

Thanks
But it seems too advanced for me
study, study, study...…

Koenkk, 2023-09-16:
No problem, I've added initial support for this device without the power outage memory.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

timdemand, 2023-10-05:
Hi, thanks for making the initial support available,
is it possible to add device specific options for the interlock mode?
This is very important for roller shutter operation, as there must only be one phase active at any moment.
I dont have an Aqara Hub unfortunatly so I can't sniff traffic for that option.

AitNev, 2023-10-05:
Sadly I neither have an Aqara Hub and thankfully it works on Z2M

Even if the interlock feature is not found. Is is possible to do the interlock functionality within Z2M?
All we need is to read the states of the switch and safely only turn one switch at a time finally publish an interface that only allows to push up or push down the roller. Right now I don't know how to do it but it might do the trick for the time being.  

Another way is to address this by hardware

HardGay, 2023-10-06:
I have add this device to my network, but i think the energy meter is not kWH but is Wh, can you correct the définition ?

HardGay, 2023-10-08:
Hi, add more the energy correct to kWH, can you check the data source of energy because in the history chart, the curve seems linear .. I use this device to follow my solar product and the curve must be more chaotic 😁

HardGay, 2023-10-08:
[image]

HardGay, 2023-10-08:
À good historic chart 👍
[image]

nicocha75, 2023-10-09:
Hello, some function are note present in module configuration
This module can be configured with pulse commande or interlock to be use with shutter. This parameter can be seen with xiomi hub (that i don't have).

jb1228, 2023-10-25:
I just bought this device since I saw it was supported. 

I was able to hook it up to control my garage door (although I have to turn the switch on then off again). However, the sensors (S1/S2) don't seem to be available. I use them for my open/closed contact sensors. 

Is this something that can be added or is there a config I can use to expose them?

Thanks. 😊

xirius, 2023-11-20:
Has anyone found out how to set Interlock mode ?

DAVIZINH0, 2023-11-26:
> I have add this device to my network, but i think the energy meter is not kWH but is Wh, can you correct the définition ?

i think the same. the energy meter its imposible to measure in kwh. a simply lamp have more consumption than a lot of devices in my home, jejeje

can i change this in zigbee2mqtt or need a new version?

latel, 2023-12-01:
> Has anyone found out how to set Interlock mode ?

yeah, I want this interlock mode

ShaunTosler, 2023-12-23:
> i think the same. the energy meter its imposible to measure in kwh. a simply lamp have more consumption than a lot of devices in my home, jejeje
>
> can i change this in zigbee2mqtt or need a new [...]

I also concur that this reading is in kW and is most defiantly not kWh - everything else is working as expected apart from it thinks that my Gas boiler has used 97kWh of electricity in an hour. This is compliantly imposable.
That's nearly 6 times the energy suppliers daily meter reading of the whole house.
When you do the math's of an average of 0.4A (at 240v) over the same time is 95Wh confirming this.
To pull 95kWh in the same time period, I would have to be drawing 395A. I dont think my neighbors or my DNO would like me if that was the case on my domestic supply 😆 

Is it possible to divide this by 1000 in the config to make it 95 wH?

Matssa56, 2024-01-07:
Hi everyone,
Not sure if I need to open another bug ticket but I repaired every zigbee device I have at home and I keep getting an error concerning this device. Here is the log:

```
Zigbee2MQTT:error 2024-01-07 21:25:33: Failed to configure 'Relais PaC', attempt 3 (Error: Read 0x54ef441000990ea9/2 aqaraOpple([10], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:463:22)
    at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:62:9)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:72:9)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:140:25)
    at Configure.configure (/app/lib/extension/configure.ts:120:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:167:17))
```

The device itself appears to work fine when changing L1 or L2. I'm currently configured as below:

[image]

Is it maybe linked to the new Zigbee2Mqtt update where it tries to automatically get functions for unsupported devices?

lsc597, 2024-01-27:
> I just bought this device since I saw it was supported.
>
> I was able to hook it up to control my garage door (although I have to turn the switch on then off again). However, the sensors (S1/S2) don't [...]

Hello, this is also a problem I am facing. Want to use S1 as a manual button for garage opening/closing and S2 for monitoring the state of the door with a magnetic contact sensor. For this ideally I should be able to configure S2 as a "wireless switch" (that's how Aqara name this feature in their app). However I am not sure that you can do this even in the Aqara app currently, but only to set both S1/S2 switches as wireless switches or not.

Anyway, a great forst step would be to have the switches state separated and visible from the relays.

rvalitov, 2024-01-27:
> Want to use S1 as a manual button for garage opening/closing and S2 for monitoring the state of the door with a magnetic contact sensor

To use an external button you should connect it to S1/S2 and COM. I use a rocker switch for that. In Zigbee2MQTT there are 3 options for external buttons:

1. `toggle` that I suppose is for a button swtich
2. `momentum` that I suppose is for a rocker switch
3. `none`

For my rocker switch I tried both: `toggle` and `momentum` do work OK. Perhaps it's just the state how they are being reported in HA, but I had no issues with opening/closing the garage door with the external button.

If you want to add a contact sensor, indeed there's no way to select operational mode for S1 and S2 separately. But I suppose that `toggle` mode will work just fine for this case. You just need to set the option `operation_mode` to `decoupled` and control manually the actions when S1 and S2 contacts are triggered... As for a connection scheme I think the following should work:
1. For external switch - connect wires to S1 and COM
2. For contact sensor - connect wires to S2 and COM

> hey guys, I saw the interlock is implemented already in https://github.com/Koenkk/zigbee-herdsman-converters/pull/6554, but works not alright.

In my tests the interlock mode works fine.

lsc597, 2024-01-27:
> If you want to add a contact sensor, indeed there's no way to select operational mode for S1 and S2 separately. But I suppose that `toggle` mode will work just fine for this case. You just need to [...]

That was exactly my plan to make the connections and wiring. But, I have the relay on my test bench right no (only power supplied) and I cannot see anywhere in z2m the state of the switches, only the outputs. And of course that's because the switches are binded to the outputs. 

I wanted to suggest an improvement because I saw that Aqara supports the option of having "wireless switches" in their native app (connected via aqara hub), so if you check that and the switches become wireless (not binded to outputs) this means you are able to have different states for the switches and the relays. And ideally, for me, would be to be able to have only one of them as a wireless switch - basically the contact sensor (S2) not to control the output anymore, to be decoupled from the output. And the button (S1) to work as a wired switch directly to the output.

Just wanted to give an idea of improvement, maybe it's possible to be done.

Appreciate all the work that is done already and I am a very happy 6-months old user of zigbee2mqtt!

rvalitov, 2024-01-27:
>  I cannot see anywhere in z2m the state of the switches

I think you can catch it in [`action` event](https://www.zigbee2mqtt.io/devices/LLKZMK12LM.html#action-enum) which is triggered when the external button is pressed. However, I'm not sure that it's possible to distinguish here between what source triggered the event - S1 or S2. This should be tested...

rvalitov, 2024-01-28:
Ok, here's what I got. The Aqara T2 really reports information from sensors S1 and S2 via zigbee2mqtt - it sends `action` with value either `single_l1` or `single_l2` for S1 and S2 sensor respectively. Example:
[image]
[image]

These actions are sent no matter of the operation mode (`decoupled` or `control_relay`). And the trigger only works for single tap - double tap or hold are not detected by the relay. However, I have a problem of receiving the state action in HA automation. My example of automation is as follow:

```yaml
---
alias: "Garage Door Switch"
description: >-
  Controls the garage door switch.

id: "switch.garage.door"
mode: queued
max_exceeded: silent
trace:
  stored_traces: 5

trigger:
  - platform: state
    entity_id: sensor.garage_door_relay_action
action:
  - service: notify.channel
    data:
      message: "Garage relay state change\r\nFrom {{trigger.from_state}} to {{trigger.to_state}}"
``` 

This automation never runs. And no issues in logs. Can anyone test that the `action` trigger works for T2?

lsc597, 2024-01-28:
Great info you provided here, @rvalitov. Yesterday I finished installing my Aqara T2 on the garage door and I will definitely try this next week!

rvalitov, 2024-01-30:
I also have some progress in working with the contact sensor. I will share my results after finishing all the tests.

ekurgan, 2024-02-15:
Have you been able to get the status of the S1 sensor ?
On my side Action is always null btw

Matssa56, 2024-02-18:
I want to use this relay with my gate motor and I'm wondering what signal must go in S1/S2 and COM.

My motor has a control unit that sends the voltage to the motor, I was thinking of connecting the two outputs to S1 or S2 in order to see which pin has 0V and 24v (or maybe 220V, need to test with a multimeter). With this, I'll know if it is closing or opening.

Raphael38210, 2024-02-19:
Bonjour, heureux utilisateur du matériel Aqara je me confronte à un problème que je n'arrive pas résoudre:
Je m'explique: J'utilise un relais T2 Aqara en contact sec pour domotiser mon portail, il est brancher sur l'entrée serrure a cle. Lorsque je rajoute ce module sur mon hub Aqara et l'application Aqara je le paramètre en contact sec, bouton momentané. Aucun problème si j'appuie sur le Switch ça ouvre si j'appuie a nouveau ca arrête et si j'appuie encore une fois ca referme. Il fait le boulot que je lui demande.
En revanche dans l'optique de supprimer mon hub Aqara j'ai basculer ce module sur zigbee2mqtt avec une clé sonoff zigbee 3.0-P. l'inclusion se passe bien j'ai bien tout les réglages qui apparaissent mais par contre une fois configurer correctement lorsque j'appuie une fois sur le bouton ça ouvre. Et si j'appuie une nouvelle fois ça remet le bouton a l'état initial mais ça fait rien sur le portail, il faut appuyer encore une fois pour stopper la fermeture...idem on appuie derrière ça fait rien et faut encore appuyer pour lancer la fermeture, j'en reviens donc a un problème de réglage mais j'ai essayé tout les modes en dry rien a faire. 

[image]

Voici un screen de mes réglages.

Bien cordialement et en vous remerciant.

rvalitov, 2024-03-08:
@Raphael38210 Bonjour, si tu ne parles par anglais, utilises Google Translate stp.

I think to fix your problem you should set the operation mode to `decoupled` and then control the logic via scripts. In my case it works perfectly.

lsc597, 2024-04-07:
I've also noticed on my side that action is taking values "single_l1" or "single_l2" when the S1 or S2 inputs are triggered, but I didn't tried to catch those events. What I noticed is that the time is very short, I think the relay is sending only one message of that type and then puts back action to 'null'.

My current configuration now is that I have a contact sensor connected to S2 input and I am monitoring the door state by watching the output for the S2 input (since my configuration is "control_relay" not "decoupled"). So depending on the the state of the output I can take a decision if the doors is closed or not. However, I noticed a strange behaviour - sometimes the output relay is changing its state one more time after triggering, and this results in an inversed behaviour. In other words, I had some cases when, after triggering the door to close, it was showing open in HA, even tough it was closed in reality. That is because the state of the relay was open instead of closed. I think this is related with the working mode of the relays - PULSE. So for pulse whenever the relay is triggered, it would stay ON for the configured time (let's say 300ms) and then it will go back OFF, but in Z2M I still see the relay as ON. If I trigger it again from state ON to state OFF, it does the same thing: open for 300ms and then closed (I can hear it switching). 

My question basically is if this is normal behaviour or is a bug in Z2M? If the relay mode is set to pulse, shouldn't it go back to OFF also in Z2M interface after triggering it?

And also the other thing I have in mind is: is it possible to be able to set the "mode" and "switch type" for each relay an switch independently? Because in my configuration I would need S1 to be momentary (it's a push button which is opening/closing the garage door) and S2 to be toggle (it's the door contact sensor which acts like a toggle switch rather than momentary) and for the outputs L1 should work in PULSE mode and L2 should work in power mode (ON/OFF) so that it can be controlled directly by the contact sensor.

HarvsG, 2024-04-25:
Is  there any evidence that this device does, or could support binding with a bulb (for use with decoupled mode)

TheRealWaldo, 2024-05-01:
I'm having some similar challenges with this device, but my use case is different.

On pairing/configuring, there are a few errors about types not matching (strings where integers are expected, undefined where a 1, 2, or 3 is expected, etc.) from the converter, but it does seem to work, at least partially.

I want to use S1/S2 in dry contact mode, decoupled from both relays.  I need to know when contact is made, and when it ends, and so far I am unable to see any state change that would be applicable.

In addition, it is reporting a firmware update (27 to 30), but the update simply will not apply.

```
Error 2024-04-30 18:17:45Exception while calling fromZigbee converter: Expected one of: 1, 2, 3, got: 'undefined'}
Error 2024-04-30 18:17:45Exception while calling fromZigbee converter: Expected one of: 0, 1, 2, 3, got: 'undefined'}
Error 2024-04-30 18:17:45Exception while calling fromZigbee converter: Expected one of: 0, 1, 3, got: 'undefined'}
Error 2024-04-30 18:17:45Exception while calling fromZigbee converter: Value is not a number, got undefined (undefined)}
```

TheRealWaldo, 2024-05-01:
Just stumbled across some time print on an image that says something along the lines of `please note, the device needs to be connected to DC power to use as in wireless switch mode`.  This was in context of using S1 and S2 as contact sensors.

I'm definitely operating in AC.

BeFygo, 2024-07-22:
> Is there any evidence that this device does, or could support binding with a bulb (for use with decoupled mode)

I managed to bind the LLKZMK12LM to a group with LWG004 spots.
Even in decouple mode the relay switches to the same state as lightbulbs.
And only a action change of "" when pressing the connected switch.

Unbinding is impossible because of the errors mentioned by @TheRealWaldo 

Even in toggle mode there is a noticeable delay between using the switch and the switching of the relay.

pswel1, 2024-10-01:
Hi all, very detailed thread!

had some tips for lumi.switch.acn047 (the T2 relay) on HA (particularly zigbee2mqqt for sure, possibly ZHA)

To get the relay to operate successfully as a dual sensor (ex- garage door open/closed magnetic switches), it has to use AC voltage to power the T2 unit.  Then it needs to be in "decoupled" mode in "Exposes", and additionally in "toggle" mode for the switch.

This combination now emits an mqqt code "single_l1" (or l2) when the pin is bridged to COM

But, most garage door magnetic sensors don't work with A/C.  So now you need a dual 24V octocoupler relay board.  Tap into the garage door transformer or 24V to power a compatible board, or use wall wart from mains to drive the contact sensor's loops and use them to control the relay on the AC side of COM/S1/S2

annoying but finally "does it all" for garage door from a single device.

rvalitov, 2024-10-02:
Ok, below is my story how I configured Aqara T2 to work in garage. I've been using it almost for one year, so I guess I can share my experience now. The Aqara T2 module with some limitations is able to:

- remotely open and close my garage door
- open and close garage door using standard physical switch
- get information if door is open/closed using standard wired magnetic sensor.

My setup scheme is below:
[image]

**Scheme and comments**
1. Aqara T2 is connected to AC power line, 220V in my case, using terminal points `L` and `N`. Now it's available via Z2M.
2. Terminal points `L1` and `LOUT` are connected to the garage door opener - your motor/control unit, check your device manual. In manual it's written how to connect a switch that controls the garage door. For most garage motors the logic is simple: when 2 wires are connected for short period of time it gives the signal to the motor to start or stop. After each stop the motor changes the direction: it either opens or closes the door. It's very simple and you don't need to respect the polarity of wires: just connect them to the terminal points in any order. You can first test that your motor works correctly by connecting the wires by hand, and if that works, then connect the wires to Aqara T2.
3. A light switch is connected to Aqara T2 using terminal points `S1` and `COM`. It does not matter the polarity of the wires, just connect them. It's important to use a light switch which is rocker-type. Visually they look the same. But the logic is that for rocker-type switch there's no "ON"/"OFF" position - every time you push it, the button returns back to normal position, it works like a mouse click. By the way this type of switch can be used directly with the motor, because it allows to send the required "pulse" signal. In my case I use IP54 switch that is perfect for garage environment.
4. Standard wired magnetic sensors are connected using terminal points `S1` and `COM`. Polarity does not matter. There are lots of magnetic sensors of different forms and shapes. The only difference between them is the logic how they operate: they can be normally opened or normally closed. As a result they provide signal either when they become closed or opened.

**The Aqara T2 configuration.**

1. Work mode is `pulse`. Well, it's obvious that you need this mode to send a short pulse signal to the garage motor.
2. Switch type is `momentary`. Someone said here that we must use `toggle`. I made tests and no other modes than `momentary` do work and provide desired behavior. 
3. Interlock is `OFF`, we don't need this in our scenario.
4. Pulse length is `200 ms` - that's default, I didn't change that. It works good.
5. L1 operation mode: `control_relay` or `decoupled`. `L1` in Z2M corresponds to the connected switch. I use `control_relay`. In this case when switch is pressed, the Aqara T2 does everything automatically. However, if you would like to add some advanced logic, for example, if the switch should be ignored sometimes, then you should use `decoupled` mode and write your own script in HA that will manage all this.
6. L2 operation mode: `decoupled`. `L2` corresponds to the magnetic sensors. The mode must be `decoupled`, because you just need to read this data and do not want to pass any command to the motor.

Screenshot:
[image]
[image]
[image]

**How to get data when something is triggered in HA?**
Checking Z2M logs gives us information that when the switch or the magnetic sensors send a signal to Aqara T2 - we receive `action` with payload `single_l1` (switch click) or `single_l2` (magnetic sensor):
[image]

You can write a binary sensor like this to catch the events:
```yaml
# Binary sensor to detect if the garage door is closed, returns true if closed, false otherwise
- trigger:
    # Magnetic sensor says it's open
    - platform: state
      entity_id: sensor.garage_door_relay_action
      to: 'single_l2'
      id: 'opened'
    # Switch button is triggered, so the door is moving, i.e. the garage door is open or partially open or is stopped in open state
    - platform: state
      entity_id: sensor.garage_door_relay_action
      to: 'single_l1'
      id: 'opened'
  # Think and add some other condition for the "closed" state! I couldn't think of something useful without making this example too complicated.
#   - platform: state
#      entity_id: something
#      to: 'something'
#      id: 'opened'
  binary_sensor:
    name: "Garage door sensor"
    unique_id: garage_door_sensor
    state: >-
      {% if trigger.id == 'closed' %}
        true
      {% elif trigger.id == 'opened' %}
        false
      {% else %}
        unknown
      {% endif %}
    device_class: 'garage_door'
```

The logic above is very simple and not accurate. The main problem is the magnetic sensors. They are very simple and trigger only one event! I have a standard magnetic sensor that sends the event `single_l2` only when the door sensor goes to "open" position. It sends nothing when the sensor goes to "closed" position. Perhaps, you could buy another type of magnetic sensor that will send the signal when it goes to "closed" position - this seems to be more useful. But still does not look secure. 

I was thinking that perhaps using an approach with `from` could do the trick:
```yaml
 # What if we catch the situation when the magnetic sensor stops sending the "open" action?
    - platform: state
      entity_id: sensor.garage_door_relay_action
      from: 'single_l2'
      id: 'closed'
[... cut ...]
3. I think that the standard wired magnetic sensors are useless, because they send only one state in a single event. As a result this is not secure and informational. As proof of concept that magnetic sensors can be connected to Aqara T2 and Z2M - yes, that's possible, but it does not work as I would like or expect. I removed this magnetic sensor and use a normal Zigbee door sensor that sends proper events with both states explicitly defined and triggered: `opened` and `closed`. Perhaps, if the Aqara T2 integration in Z2M can be improved, the standard wired sensors could become helpful.

rvalitov, 2024-10-02:
@pswel1 thank you for reading and finding the typo about `S2`! I updated the text.

> Instead I'm using two sensors, so I (should) receive a "single_l1" for closed and a "single_l2" for open. 

Sure, this will work if you have different sensors (not the same model/version), i.e. one that triggers the "open" state, another one that triggers the "closed" state.

> However, my primary question is - the COM/S1/S2 seem to be passing AC current (matches voltage across L/N). This wasn't compatible with my magnetic sensors. The mag sensors do work in DC mode, but then the S1/S2 do not recognize a connection to COM when the T2 unit is powered by DC. Are your magnetic sensors working despite AC?

Good question. Indeed, most magnetic sensor's technical requirements say that they work with 10-30V DC. Some googling shows that there are also AC-compatible magnetic sensors, but it seems rather a rare case. I used a cheap magnetic sensor that costs 2-3$ maximum. I doubt that it is AC-compatible. It has very thin wires, so I think it also proves that it is intended for low voltage and current. However, in my tests it worked. I was testing the sensor not for a long time, perhaps one hour maximum, just to confirm that it works and that there's no way to handle my situation just with one sensor. After that I removed it. During this test the sensor always worked, there were no "false-positive" events, or "missed" events, and the wires, nor the sensor became hot - I checked the temperature. I think the wires should be checked with multi-meter to confirm that it's AC. I'm pretty sure the parameters at S1 and S2 should match the incoming power line: if it's AC then it's AC, if it's DC then it's DC, because T2 does not have any built-in AC/DC transformer. In my tests the sensor only worked in `momentary` switch type, `toggle` or `none` did not pass the events from the sensor.

TheRealWaldo, 2024-10-02:
This is going way off-topic, folks. Can we please start a new thread?

pswel1, 2024-10-04:
Just to conclude - my mag sensors do not work on AC, except when energizing initially.  So I was able to switch to DC using a octoisolator dual relay.  Now single_L1 and single_L2 fire correctly!

[image]

I already had the sensors mounted, and wanted to minimize zigbee traffic for future debugging, so this was better for me.

figuring out this functionality was a detour.  Earlier I had misinterpreted that the missing data (the errors shown via Zigbee2Mqtt ,noted above) were the issue.  Using debugging mode, I managed to trace the errors to cluster "manuSpecificLumi" with the message data "65522".  Hope that's useful.  It would be nice to get rid of the errors, but I'm pleased to finally get the status switches working!

swannet, 2024-10-07:
I don't know if its a user error or not but I give it a shot for some help. I have a problem with my roller shutter motor and the LLKZMK12LM. I have a momentary contact switch on the wall to control the roller shutter, which means I need to hold the switch to operate the motor.

I've connected my roller shutter motor to the LLKZMK12LM as follows:
- L1 is connected to the Up wire of the motor.
- L2 is connected to the Down wire of the motor.

For the switch connections:
- S1 to the switch for Up.
- S2 to the switch for Down.
- COM to the common terminal of the wallswitch.

Configuration:
Switch type is set to momentary.
Mode is set to power.
Operation Modes: Both are set to control relay.

I can control the shutters via Zigbee2MQTT (Z2M) just fine, so that part works well. However, the problem arises when using the physical switch. The motor doesn’t start moving while I hold the switch; instead, it starts moving when I release it and continues until I press the switch again. It seems like the relay is in 'toggle' mode, even though I've set it to 'momentary'.

I also occasionally encounter the following error: "z2m: Exception while calling fromZigbee converter: Expected one of: 0, 1, 3, got: 'undefined'".

State:
```
{
    "consumption": 0.0010999999046325684,
    "current": 0.02,
    "device_temperature": 28,
    "energy": 0,
    "interlock": "ON",
    "led_disabled_night": false,
    "linkquality": 180,
    "mode": "power",
    "operation_mode_l1": "control_relay",
    "operation_mode_l2": "control_relay",
    "power": 0.15,
    "power_on_behavior": "off",
    "power_outage_count": 9,
    "pulse_length": 200,
    "state": "OFF",
    "state_l1": "OFF",
    "state_l2": "OFF",
    "switch_type": "momentary",
    "update": {
        "installed_version": 30,
        "latest_version": 30,
        "state": "idle"
    },
    "voltage": 228.86,
    "action": null,
    "update_available": null
}
```

Anyone who can give me some tips how to troubleshoot / fix this?

Belox86, 2024-12-23:
Hi everyone,
i'm having a minor issue with the aqara T2 used for Dry Contact pulse.
My use case is fairly simple, i have an automatic swing gate with its own control board, i just want to make it smart by adding the T2 which should momentary close the dry contact that initiate the opening-pause-close routine of the board.
No input, only a pulse close-open of a dry contact connected to L1 output (L2 is unused).
i configured the switch as follow (copying its state):
```
{
    "consumption": 0,
    "current": 0,
    "device_temperature": 30,
    "energy": 0,
    "interlock": "OFF",
    "led_disabled_night": false,
    "linkquality": 168,
    "mode": "pulse",
    "operation_mode_l1": "control_relay",
    "operation_mode_l2": "control_relay",
    "power": 0,
    "power_on_behavior": "off",
    "power_outage_count": 4,
    "pulse_length": 500,
    "state": "OFF",
    "state_l1": "OFF",
    "state_l2": "OFF",
    "switch_type": "none",
    "voltage": 216.77,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
    }
}
```

The aqara d it's job but unfortunatly does not report back the state when switch-off after the pulse interval. What it happen is that the switch report a false state.
For example, starting from an OFF state, i press activate the L1 switch on HA (or z2m inferface); the state of the switch stay ON even though the rele is switched off after the pulse interval.
Now with the switch on ON state (but the L1 contact open) if i press again, the state become OFF but the contact is actually closed and reopen according to the pulse configuration.
Is it possible to made the aqara report back the state of the switch after it is automatically turnd off by the pulse interval?
Thank you.
Michele

Belox86, 2024-12-25:
i'm using the T2 for a very simple case:
the T2 should momentary close (and then reopen) a dry contact to start the opening of a gate which has it's own board; no physical input.
this is how i configured the T2:
```
{
    "consumption": 0,
    "current": 0,
    "device_temperature": 29,
    "energy": 0,
    "interlock": "OFF",
    "led_disabled_night": false,
    "linkquality": 208,
    "mode": "pulse",
    "operation_mode_l1": "control_relay",
    "operation_mode_l2": "control_relay",
    "power": 0,
    "power_on_behavior": "off",
    "power_outage_count": 6,
    "pulse_length": 200,
    "state": "OFF",
    "state_l1": "OFF",
    "state_l2": "OFF",
    "switch_type": "none",
    "voltage": 216.45,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
    }
}
```
The relay its self it's behaving as expected, it open and close the circuit, but unfortunately it never report back the proper state after the pulse length. 
So both in z2m and in Home Assistant i see the state of the switch as ON while it has been automatically turn off by the pulse mode.
Is there a way to fix it?
Thank you.

rvalitov, 2024-12-26:
@Belox86 no way to fix that. It's intended way of work. If you run in pulse mode then you don't need to rely on the state. Just ignore it. Why do you need that?

Belox86, 2024-12-26:
> @Belox86 no way to fix that. It's intended way of work. If you run in pulse mode then you don't need to rely on the state. Just ignore it. Why do you need that?

Actually, it's not that I really need it, but there are several reasons why I think it should behave this way:
- Many other pulse switches I have work like this.
- I like the idea of the state reflecting the actual state of the relay.
- If I add the switch to the HA dashboard, it would behave like a proper push button, returning to its original state rather than staying in a meaningless ON or OFF state. I understand there are ways to work around this, but they often require creating unnecessary helpers or automations, which feels redundant.

rvalitov, 2025-01-03:
> I like the idea of the state reflecting the actual state of the relay.

I doubt that it reflects the true situation. Because the pulse length is 200ms or 400ms, and I doubt that any Zigbee device can report such state in real time. In my tests the minimum interval was 500ms. Perhaps it depends on how many Zigbee devices you have and how heavy traffic on the network you have. So what I want to say that you may get the state but not the actual, i.e. it's impossible to catch the situation when the switch is really sending the pulse: you may get this event but not in real time....

> I understand there are ways to work around this, but they often require creating unnecessary helpers or automations, which feels redundant

Yes, I see... I think you should create a dedicated issue about that. This issue is about general device support.

SanderDevCode, 2025-03-05:
Still relevant (I still think the switch type is switched on this device)

closed 2025-05-13 by github-actions[bot]

SanderDevCode, 2025-08-03:
I'd like to reopen this issue to continue the investigation.

After the latest firmware update from Aqara, changing the `switch_mode` doesn't have any effect anymore — the device always behaves as a toggle, even if `momentary` is selected. Additionally, the `interlock` feature seems to have stopped working (it was working on the previous firmware version).

I'd like to provide diagnostic data and potentially contribute a PR, but I understand this requires some hardware.

I'm planning to buy an **nRF52840-dongle** to sniff Zigbee traffic and observe what's actually happening on the wire. Alongside that, I want to use an **official hub** and app to toggle the settings so I can capture the real Zigbee commands sent by the ecosystem.

@Koenkk you previously mentioned using a Xiaomi gateway — do you have a recommendation on what hub to get?  
Would an **Aqara Hub E1** be sufficient, or would you recommend something more featured like the **Aqara M2**?

Also, a lot of hubs require selecting “Mainland China” as the region in the app — does this have any impact on the Zigbee protocol itself (e.g., encryption keys, command formats), or is it just a cloud-related limitation that won’t affect sniffing local Zigbee traffic?

Thanks in advance!

Koenkk, 2025-08-04:
> do you have a recommendation on what hub to get?

Any hub that is compatible with this device should do it

> does this have any impact on the Zigbee protocol itself (e.g., encryption keys, command formats), or is it just a cloud-related limitation that won’t affect sniffing local Zigbee traffic?

I expect it won't affect the Zigbee communication