Koenkk/zigbee2mqtt #18872: [New device support]: _TZE200_rxntag7i TS0601 / Tuya - Avatto TRV16

Paalap, 2023-09-05:
### Link

https://www.aliexpress.com/item/1005005850177778.html

### Database entry

{"id":51,"type":"EndDevice","ieeeAddr":"0xa4c138d4a8fef677","nwkAddr":10287,"manufId":4417,"manufName":"_TZE200_rxntag7i","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_rxntag7i","powerSource":3,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1693927818317,"defaultSendRequestWhen":"immediate"}

### Comments

A "new" Tuya TRV.

Paalap, 2023-09-27:
Now it is supported as ME168

closed 2023-09-27 by Paalap

derdon999, 2023-11-01:
> Now it is supported as ME168

Hi, I know this issue is closed, but I really don´t know how to add this TVR16 to my Conbee II instance on Raspi with deconz. The devices are not linking/connecting and therefiore I don´t see them in phoscon, REST api or openHAB. Is it possible @Paalap you give me a hint or guide me?

ConBee II
Gateway Version 2.24.1
Firmware 26780700
deconz-2.24.1-qt5.deb
openHAB 3.4.5

Thanks a lot!

Paalap, 2023-11-01:
@derdon999 Talking about deconz you should open an issue in the according repository - the integration approaches are quite different

gvisol, 2023-12-26:
HI everyone,

Few words to note that _TZE200_rxntag7i TS0601 / Tuya - Avatto TRV16 does not fit to ME168 TRV model.

In the meantime, I will try to provide more specific information. To the experts at ZB2MQTT: feel free to ask me for more information.

Thanks in advance

agnus777, 2024-02-01:
> HI everyone,
> 
> Few words to note that _TZE200_rxntag7i TS0601 / Tuya - Avatto TRV16 does not fit to ME168 TRV model.
> 
> In the meantime, I will try to provide more specific information. To the experts at ZB2MQTT: feel free to ask me for more information.
> 
> Thanks in advance

mee too, I tried several different trv quirks, without lucky.

It works nice under Tuya, if need some info from that platform I can try to help in that way.

TheDzey, 2024-02-23:
Hi, having problem to add Tesla Smart Thermostatic Valve TV500 TSL-TRV500-TV05ZG, Z2M says that is unsupportet. Do you know if will be sone? or how i should add it
Thank you,
Jarda