Koenkk/zigbee2mqtt #18874: Device commands keep timing out

jackmurray, 2023-09-05:
### What happened?

I have a somewhat problematic device on my network. It's a TuYa TS011F_plug_1 smart plug, and very often it doesn't respond to commands.

In the network map I can see it's got link strengths as follows:

Coordinator: 10
Router 1: 0
Router 2: 114

Both routers (lightbulbs in this case) themselves work perfectly.

In the z2m logs I get errors like this:

Publish 'set' 'state' to 'outside_plug_1' failed: 'Error: Command 0xa4c1383ab3a8e116/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 13070 - 1 - 254 - 6 - 11 after 10000ms)'

I've tried removing it from the network and re-joining it with only the router with good link strength allowing joins, but that doesn't seem to have made a difference.

Some days it works fine, and others (like today) it will take 10+ attempts to toggle the device's state before a command gets through to it.

Is there something I can do to force z2m to send commands via the specific router that has higher strength, or should the coordinator be doing that automatically? Is there anything else I can try here to get this device to respond properly?

### Zigbee2MQTT version

1.32.2 commit: 1ec1e57

### Adapter firmware version

20210708

### Adapter

electrolama zzh

tecstuff, 2023-10-02:
Potentially related issue for me as well. Sending commands works when device is pluged in for a few minutes. Then it starts timing out. On/Off event triggered by the hardware button of the device is reported back to Z2M.

Did you check if taking the device out of the power supply for few seconds brings it back to working state?

I updated the device firmware to version 192 with no effect.

jackmurray, 2023-10-02:
>I updated the device firmware to version 192 with no effect.

Same here.

>Did you check if taking the device out of the power supply for few seconds brings it back to working state?

I did, and it didn't make a difference.

Interestingly the messages from the device --> coordinator (updating voltage, current etc) seem to be received fine, it's only the messages going to the device that are affected.

tecstuff, 2023-10-04:
I've done some more testing on this and:
* updated my Sonoff 3.0 Dongle-E Firmware 
* updated Zigbee2MQTT to latest available versions
* Changed one 2.4GHz WiFi network to not interfere on same channel as Zigbee

At least in my case I think it can be assumed that it's related to a weak signal strenght. Only if I bring the Tuya plug really close it will allow for bi-directional communication. Else, only the Coordinator is able to receive data from it but cannot send. 
The pairing I have done at the power plug destination. But shortly after pairing, connection for signals from Coordinator to Tuya device was broken again.

Device strenght ~5m away from Sonoff stick and behind a fridge: **0/98**
Device strenght ~1m away from Sonoff stick: **69/155**

With signal strength of 69 in close distance to the Coordinator, I consider the device somehow broken.

Have ordered a more expensive Aqara. But, I may try with a second Tuya (from a different sales company to avoid same batch).

jackmurray, 2023-11-04:
Not stale

kizio1000, 2023-12-30:
Same here, have been trying to track issue for couple of days and I found so far that problem has following circumstances:

At start of z2m there is error in log:

`Assertion failed: Command (setConfigurationValue) returned unexpected state: [object Object]`

After some time that system is working there are problems with connection to mosquito:

`Zigbee2MQTT:error 2023-12-30 03:32:49: Not connected to MQTT server!
Zigbee2MQTT:error 2023-12-30 03:32:49: Cannot send message: topic: 'zigbee2mqtt/pompa CWU', payload: '{"current":0.3,"energy":10.79,"linkquality":188,"power":58,"power_outage_memory":"restore","state":"ON","switch_type":"state","voltage":241}
Zigbee2MQTT:error 2023-12-30 03:32:49: MQTT error: connect ECONNREFUSED 172.30.33.0:1883
Zigbee2MQTT:error 2023-12-30 03:32:50: MQTT error: connect ECONNREFUSED 172.30.33.0:1883
Zigbee2MQTT:error 2023-12-30 03:32:51: MQTT error: connect ECONNREFUSED 172.30.33.0:1883
Zigbee2MQTT:error 2023-12-30 03:32:52: MQTT error: connect ECONNREFUSED 172.30.33.0:1883
Zigbee2MQTT:error 2023-12-30 03:32:52: Not connected to MQTT server!`

During work of HA there are errors in supervisor log:

`23-12-30 10:01:35 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.4:8099 ssl:default [Connect call failed ('172.30.33.4', 8099)]
23-12-30 10:01:37 INFO (MainThread) [supervisor.auth] Auth request from 'core_mosquitto' for 'homeassistant'
23-12-30 10:01:38 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.33.4:8099 ssl:default [Connect call failed ('172.30.33.4', 8099)]
23-12-30 10:01:38 INFO (MainThread) [supervisor.auth] Successful login for 'homeassistant'
23-12-30 10:02:34 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token`

As a process of bugtracking I have tried with earlier version of mosquito (6.3.1) not much different and with edge branch of z2m, same problem or even worst. I have reinstalled whole system (and loaded backup after). I have updated stick ( ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2 ) fw from 7.2.2 to 7.3.1.0 build 176 and no difference. 
Restarting z2m solves problem for a while. I also noticed that some devices (routers) are going offline and there are issues with pairing them. 

My system was working w/o problems for months. I have around 60 devices in total ( Battery: 28 / Mains (single phase): 27 ) and it's hard to point exact moment when problems appeared (idk if it was one of z2m updates or mosquito) as it takes while until you see that some exotic switches that you switch once couple days do not respond. at this point I am pretty much frustrated as  it seams that I tried everything as on top of above I have also added extension cable for coordinator (which I never used/needed before) and checked again channel corelation among wifi-zigbee. Still network is unstable as never before.

endevo-mschulz, 2024-02-22:
Since last week I have exactly the same issue. As kizzio1000 mentioned: "Still network is unstable as never before."

kizio1000, 2024-02-22:
I got enough of this fight and and migrated to Sonoff P as it looks like issues are mostly related to Sonoff E. And poof all my problems are gone. Sure, repairing 59 devices took couple hours but now (after week of testing) my network is rock solid again (in spite of significantly lower LQI than on sonoff E). I would really recommend this solution to anyone struggling time after time with ezsp :)

endevo-mschulz, 2024-02-22:
@kizio1000 Thanks for sharing your experience with the Sonoff P. I will try that solution, because its making me sick... :D

closed 2024-09-20 by github-actions[bot]