Koenkk/zigbee2mqtt #18881: Several Aqara Devices can not join or left the network

smartmatic, 2023-09-06:
### What happened?

I've migrated back from ZHA to zigbee2mqqt and re-added all my devices to zigbee2mqtt. I use a Sonoff Dongle Plus V2 with the latest release of Z2M 1.33.0-1. The Sonoff Dongel is runninh with the latest firmware 7.3.1.0 build 176

Some devices like a temperature and humidty sensor and door windows sensor have never joined the network. I am not able to add them. Other devices like the Aqara Mini Wireless Switch WXKG11LM left the network and now i am not able to add the device again! For that specific device i have changed the battery to be sure that the battery is full. Then i tried to add the switch in the room where the switch is used without success. After that i was close to the coordinator but the device is will not be interviewed.

### What did you expect to happen?

To be able to add Aqara devices which stay in the network

### How to reproduce it (minimal and precise)

1. Deleted the Offline device
2. Replaced the battery
3. Tried to integrate the device again
4. Moved it closer to the coordinator
5. Tried to add it via a router device

### Zigbee2MQTT version

1.33.0-1

### Adapter firmware version

 7.3.1.0 build 176

### Adapter

Sonoff Dongle Plus E

JustAnotherPU, 2023-09-06:
Exact same problem here, was able to pair only an aqara switch and and aqara water sensor. 
Can't pair aqara mini bouton or window sensor.

I am running the multi protocol firmware ezsp

catduckgnaf, 2023-09-08:
my WXKG11LM are having issues. I see the click events in z2m, but not showing available elsewhere.

JustAnotherPU, 2023-09-10:
same here look like there are no solution or workaround :/

notorioustubze, 2023-10-05:
I have the same issue for some reason all my Aqara buttons (WXKG11LM) and my door and windows sensors (MCCGQ11LM) keep disconnecting. This has only started happening within the last couple of months, they had all been rock solid prior. Using the Sonoff Dongle V2 and latest version of Z2Mqtt.

closed 2023-11-13 by github-actions[bot]