Koenkk/zigbee2mqtt #18884: [New device support]: supported SOMFY 1822647A

qlwz, 2023-09-06:
### Link

https://www.somfy.com.hk/cn-hk/products/1822647/zigbee-smart-plug

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0x4cc20601100002db","nwkAddr":18518,"manufId":4640,"manufName":"SOMFY","powerSource":"Mains (single phase)","modelId":"1822647","epList":[12,242],"endpoints":{"12":{"profId":260,"epId":12,"devId":81,"inClusterList":[0,3,1794,6,4],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"1822647","manufacturerName":"SOMFY","powerSource":1,"zclVersion":1,"appVersion":37,"stackVersion":3,"hwVersion":1,"dateCode":"2017-02-01"}},"seMetering":{"attributes":{"multiplier":1,"divisor":10000,"instantaneousDemand":657,"currentSummDelivered":[0,3715]}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b002590ded5","endpointID":242},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b002590ded5","endpointID":242},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002590ded5","endpointID":242},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b002590ded5","endpointID":242}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":1,"maxRepIntval":3600,"repChange":0},{"cluster":1794,"attrId":1024,"minRepIntval":5,"maxRepIntval":3600,"repChange":1},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":37,"stackVersion":3,"hwVersion":1,"dateCode":"2017-02-01","zclVersion":1,"interviewCompleted":true,"meta":{"configured":89021505},"lastSeen":1693933378020,"defaultSendRequestWhen":"immediate"}

### Comments

supported SOMFY 1822647A
[image]
[image]

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['1822647'],
    model: '1822647A',
    vendor: 'SOMFY',
    description: 'Zigbee Smart Plug',
	fromZigbee: [fz.on_off, fz.metering],
	toZigbee: [tz.on_off],
    exposes: [e.switch({disablePowerOnBehavior: true}), e.power(), e.energy()],
	configure: async (device, coordinatorEndpoint, logger) => {
		const ep = device.getEndpoint(12);
		await reporting.bind(ep, coordinatorEndpoint, ['genBasic', 'genIdentify', 'genOnOff', 'seMetering']);
		const payload = [{
			attribute: 'onOff',
			minimumReportInterval: 1,
			maximumReportInterval: 3600,
			reportableChange: 0,
		}];
		await ep.configureReporting('genOnOff', payload);
		await reporting.readMeteringMultiplierDivisor(ep);
		await reporting.instantaneousDemand(ep);
            await reporting.currentSummDelivered(ep);
            try {
                await reporting.currentSummReceived(ep);
            } catch (error) {
                /* fails for some: https://github.com/Koenkk/zigbee2mqtt/issues/13258 */
            }
            ep.saveClusterAttributeKeyValue('seMetering', {divisor: 10000, multiplier: 1});
		await ep.read('genOnOff', ['onOff']);
	},
};

module.exports = definition;
```

Koenkk, 2023-09-06:
Added, thanks!

closed 2023-10-14 by github-actions[bot]

jack338c, 2025-11-05:
Once restart zigbee2mqtt .the log shows a lot of these .Could someone fix this @qlwz @Koenkk 
I'm using Sonoff ZBDongle-E 7.4.4 with efr32mg21 CP2102 serial chip.and zigbee2mqtt version is 2.6.2
Sorry for troubles

```
`[2025-11-05 17:38:00] info:     z2m: Zigbee2MQTT started!
{"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '25581'.))
[2025-11-05 17:38:05] info:     z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/1221051039810110150109113116116_0xbc8d7efffee68ed3/connection_state/config', payload '{"default_entity_id":"binary_sensor.zigbee2mqtt_bridge_connection_state","device":{"hw_version":"EmberZNet 7.4.4 [GA]","identifiers":["zigbee2mqtt_bridge_0xbc8d7efffee68ed3"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"2.6.2"},"device_class":"connectivity","entity_category":"diagnostic","name":"Connection state","object_id":"zigbee2mqtt_bridge_connection_state","origin":{"name":"Zigbee2MQTT","sw":"2.6.2","url":"https://www.zigbee2mqtt.io"},"payload_off":"offline","payload_on":"online","state_topic":"zigbee2mqtt/bridge/state","unique_id":"bridge_0xbc8d7efffee68ed3_connection_state_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
[2025-11-05 17:38:08] info:     z2m: Configuring 'SOMFY'
[2025-11-05 17:38:08] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SOMFY', payload '{"energy":0.37,"linkquality":255,"power":229,"state":"ON"}'
[2025-11-05 17:38:09] error:    z2m: Failed to configure 'SOMFY', attempt 1 (Error: ZCL command 0x4cc2060110000423/12 seMetering.configReport([{"attribute":"currentSummReceived","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":257}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@6.1.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:385:28)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@6.1.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:1055:26)
    at Endpoint.configureReporting (/app/node_modules/.pnpm/zigbee-herdsman@6.1.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:768:9)
    at Object.currentSummReceived (/app/node_modules/.pnpm/zigbee-herdsman-converters@25.35.1/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:137:5)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@25.35.1/node_modules/zigbee-herdsman-converters/src/devices/somfy.ts:73:13)
    at Configure.configure (/app/lib/extension/configure.ts:123:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:252:17))
[2025-11-05 17:38:09] info:     z2m: Configuring 'SOMFY'
[2025-11-05 17:38:09] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SOMFY', payload '{"energy":0.37,"linkquality":255,"power":229,"state":"ON"}'
[2025-11-05 17:38:11] error:    z2m: Failed to configure 'SOMFY', attempt 2 (Error: ZCL command 0x4cc2060110000423/12 seMetering.configReport([{"attribute":"currentSummReceived","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":257}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@6.1.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:385:28)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@6.1.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:1055:26)
    at Endpoint.configureReporting (/app/node_modules/.pnpm/zigbee-herdsman@6.1.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:768:9)
    at Object.currentSummReceived (/app/node_modules/.pnpm/zigbee-herdsman-converters@25.35.1/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:137:5)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@25.35.1/node_modules/zigbee-herdsman-converters/src/devices/somfy.ts:73:13)
    at Configure.configure (/app/lib/extension/configure.ts:123:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:252:17))
[2025-11-05 17:38:11] info:     z2m: Configuring 'SOMFY'
[2025-11-05 17:38:11] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SOMFY', payload '{"energy":0.37,"linkquality":255,"power":229,"state":"ON"}'
[2025-11-05 17:38:12] error:    z2m: Failed to configure 'SOMFY', attempt 3 (Error: ZCL command 0x4cc2060110000423/12 seMetering.configReport([{"attribute":"currentSummReceived","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":257}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@6.1.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:385:28)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@6.1.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:1055:26)
[... cut ...]
[2025-11-05 17:38:57] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SOMFY', payload '{"energy":0.37,"linkquality":255,"power":228.7,"state":"ON"}'
[2025-11-05 17:39:02] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SOMFY', payload '{"energy":0.37,"linkquality":255,"power":230,"state":"ON"}'
[2025-11-05 17:39:07] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SOMFY', payload '{"energy":0.37,"linkquality":255,"power":228.7,"state":"ON"}'
[2025-11-05 17:39:12] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/SOMFY', payload '{"energy":0.37,"linkquality":255,"power":228.3,"state":"ON"}'`
```