Koenkk/zigbee2mqtt #18888: Missing devices and the UI can't connect to ws://

nicolaevladescu, 2023-09-06:
### What happened?

I recently found myself unable to use the UI, settings or devices never load and the below error is on repeat, looks like nothing that is async never loads, just the initial UI. Also while the initial UI is loading I see other interrupted requests for `.css` files for example.

```
Firefox can’t establish a connection to the server at ws://redacted.dns:8888/api. [index.c80ccfed.js:78:6663](http://redacted.dns:8888/assets/index.c80ccfed.js)
The connection to ws://redacted.dns:8888/api was interrupted while the page was loading.
```

I see no obvious errors in the debug log or in the zigbee-herdsman debug log.

I have only one IKEA device that is working, mqtt is working also, updated to latest version but got the same result.

P.S. I am using the official docker image

### What did you expect to happen?

See devices in the UI

### How to reproduce it (minimal and precise)

Not sure. I did not touch it or the devices for a couple of months then I logged in and I updated the firmware to my single IKEA device, I guess this is the only correlation, after that the above problem started.

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

Factory/Can't tell

### Adapter

SONOFF Dongle-E

### Debug log

```
Using '/app/data' as data directory
Zigbee2MQTT:debug 06-09-2023 22:44:44: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  06-09-2023 22:44:44: Logging to console and directory: '/app/data/log/2023-09-06.22-44-43' filename: log.txt
Zigbee2MQTT:debug 06-09-2023 22:44:44: Removing old log directory '/app/data/log/2023-07-15.16-17-01'
Zigbee2MQTT:info  06-09-2023 22:44:44: Starting Zigbee2MQTT version 1.33.0 (commit #7ee207f)
Zigbee2MQTT:info  06-09-2023 22:44:44: Starting zigbee-herdsman (0.18.5)
Zigbee2MQTT:debug 06-09-2023 22:44:44: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"ezsp","path":"/dev/ttyACM0"}}'
Zigbee2MQTT:info  06-09-2023 22:44:47: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  06-09-2023 22:44:47: Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"}'
Zigbee2MQTT:debug 06-09-2023 22:44:47: Zigbee network parameters: {"channel":11,"extendedPanID":221,"panID":6754}
Zigbee2MQTT:info  06-09-2023 22:44:47: Currently 1 devices are joined:
Zigbee2MQTT:info  06-09-2023 22:44:47: bathroom_lamp_switch (0xdc8e95fffec3949a): E1743 - IKEA TRADFRI ON/OFF switch (EndDevice)
Zigbee2MQTT:info  06-09-2023 22:44:47: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  06-09-2023 22:44:47: Connecting to MQTT server at mqtt://rpi.lan:1883
Zigbee2MQTT:debug 06-09-2023 22:44:47: Using MQTT login with username: mqtt_client
Zigbee2MQTT:info  06-09-2023 22:44:47: Connected to MQTT server
Zigbee2MQTT:info  06-09-2023 22:44:47: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:info  06-09-2023 22:44:47: MQTT publish: topic 'zigbee2mqtt/901/availability', payload '{"state":"online"}'
Zigbee2MQTT:info  06-09-2023 22:44:47: Started frontend on port 0.0.0.0:8080
Zigbee2MQTT:info  06-09-2023 22:44:47: MQTT publish: topic 'zigbee2mqtt/bathroom_lamp_switch', payload '{"battery":75,"linkquality":168,"update":{"installed_version":604241926,"latest_version":604241926,"state":"idle"}}'
Zigbee2MQTT:info  06-09-2023 22:44:47: Zigbee2MQTT started!
```

nicolaevladescu, 2023-09-18:
It started working as usual again, and being responsive. I can now see the device listed also, I have no clue to why and what's happening. The websocket now works and returns json. I have checked the logs for the past 3 days and nothing stands out.

closed 2023-10-27 by github-actions[bot]

gizmocuz, 2024-04-20:
I stumbled on this myself as well. Turned out I had a proxy enabled in the browser and this caused the issue.
Maybe it is of help for some others that land here