Koenkk/zigbee2mqtt #18892: [New device support]: Tuya TRV TRV601 (TRV602) _TZE200_rtrmfadk  (Converter & Datapoints provided)

Mikescotland, 2023-09-07:
### Link

https://www.aliexpress.com/item/1005004813488375.html?spm=a2g0o.productlist.main.37.130c23c8R8nr79&algo_pvid=27758fd9-0451-4f71-b0e8-f4f6a5ebe03a&algo_exp_id=27758fd9-0451-4f71-b0e8-f4f6a5ebe03a-18&pdp_npi=4%40dis%21GBP%2152.17%2122.43%21%21%2163.58%21%21%4021038edc16940448207022588e0728%2112000030589174994%21sea%21GB%210%21A&curPageLogUid=pqJjqAWtXHGe

### Database entry

{"id":36,"type":"EndDevice","ieeeAddr":"0xa4c13823ca26d31f","nwkAddr":8322,"manufId":4417,"manufName":"_TZE200_rtrmfadk","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"���,i���,i���,i���,i

### Comments

Yet another Tuya Smart TRV, the model I was working on is TRV601, it's probably exactly the same as TRV602 (with OLED screen). Disctinctive with these devices is the fact they take 3xAA not 2xAA batteries.
Device unsupported. 
Here are the datapoints from Tuya:
MODE 1 (actually it's Preset - auto, manual, off)
TARGET TEMP 2
CURRENT TEMP 3
WORKING STATUS 6
WINDOW STATUS 7
OPEN WINDOW 8
CHILD LOCK 12
BATTERY 13
FAULT ALARM 14
MIN LIMIT TEMP 15
MAX LIMIT TEMP 16
HEATING SCHEDULE (Actually it's MONDAY) 17
WEEK PROGRAM TUE 18
WEEK PROGRAM WED 19
WEEK PROGRAM THU 20
WEEK PROGRAM FRI 21
WEEK PROGRAM SAT 22
WEEK PROGRAM SUN 23
ROOM SENSOR CALIBRATION 101
VALVE 108 - Valve position, seems to be only either 0 or 100%
MODEL/TYPE (?) 109 - not sure what's that, it was in Chinese
MOTOR THRUST 110 - probably it is force valve to open/close
DISPLAY BRIGHTNESS 111 - not implemented in converter
SOFTWARE VERSION 112 - not implemented in converter
SCREEN ORIENTATION 113 - not implemented in the converter
SYSTEM MODE 114 (??) - not implemented in the converter

External converter attached, have it now working normally with Climate card and Google Home. I didn't bother with software version, screen orientation (this can be done via physical button so once set, forgotten) etc and seems this model DOES NOT have modes, just presets, auto - schedule, manual as set manually and off. Added modes just for Home Assistant so it doesn't throw out all HVAC options and it's Google Home friendly. 

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
           
            modelID: 'TS0601',
          
            manufacturerName: '_TZE200_rtrmfadk',
        },
    ],
    model: 'TS0601_new',
    vendor: 'TuYa',
    description: 'Fill in a description of the device here',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, 
    configure: tuya.configureMagicPacket,
    exposes: [
            e.battery(), e.child_lock(), e.max_temperature(), e.min_temperature(),
            e.position(), e.window_detection(),
            e.binary('window', ea.STATE, 'CLOSED', 'OPEN').withDescription('Window status closed or open '),
            e.binary('alarm_switch', ea.STATE, 'ON', 'OFF').withDescription('Thermostat in error state'),
            e.climate()
                .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
                .withLocalTemperature(ea.STATE)
                .withPreset(['auto', 'manual', 'off'], ea.STATE_SET)
                .withRunningState(['idle', 'heat'], ea.STATE)
                .withSystemMode(['off', 'heat'], ea.STATE_SET, 'Only for Homeassistant')
                .withLocalTemperatureCalibration(-3, 3, 0.5, ea.STATE_SET),
            ...tuya.exposes.scheduleAllDays(ea.STATE_SET, 'HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C'),
            
           
    ],
    meta: {
            tuyaDatapoints: [
                
                [1, 'preset', tuya.valueConverterBasic.lookup({'auto': tuya.enum(0),'manual': tuya.enum(1), 'off': tuya.enum(2)})],
                [1, 'running_state', tuya.valueConverterBasic.lookup({'heat': tuya.enum(0), 'idle': tuya.enum(1)})],
                [107, 'system_mode', tuya.valueConverter.TV02SystemMode],
                [2, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
                [3, 'local_temperature', tuya.valueConverter.divideBy10],
                
                [6, 'running_state', tuya.valueConverterBasic.lookup({'heat': 1, 'idle': 0})],
                [7, 'window', tuya.valueConverterBasic.lookup({'OPEN': 1, 'CLOSE': 0})],
                [8, 'window_detection', tuya.valueConverter.onOff],
                [12, 'child_lock', tuya.valueConverter.lockUnlock],
                [13, 'battery', tuya.valueConverter.raw],
                [14, 'alarm_switch', tuya.valueConverter.onOff],
                [15, 'min_temperature', tuya.valueConverter.divideBy10],
                [16, 'max_temperature', tuya.valueConverter.divideBy10],
                [17, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
                [18, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
                [19, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
                [20, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
                [21, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
                [22, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
                [23, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
                [101, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration1],
                [108, 'position', tuya.valueConverter.divideBy10],
            ],
    },
};

module.exports = definition;
```

Mikescotland, 2023-09-07:
I'm having some problems with adding a few things:
datapoint 110 - motor thrust, I can see from Tuya IoT, it's on settings "middle". Probably it can be set up to "low, middle, high", however when I try to set it up, no converters available. It can also be a value set up during the calibration of the valve, i.e. stroke, which then it's better not to be touched.
datapoint 113 - screen orientation. Possible values "up" and "down". Unable to resolve as "no converters available".
datapoint 115 - switch deviation. It's a slider in the tuya app to set up hysteresis for the "Eco" mode - by default is 0.5 degrees. Don't know how to incorporate it.

resettertje, 2023-09-25:
_datapoint 113 - screen orientation. Possible values "up" and "down". Unable to resolve as "no converters available"._

Pretty sure that is set on the device itself, mine came in the wrong orientation so i had to flip it.

Mikescotland, 2023-09-25:
Yes. It is on the device too, but also via software (tuya app allows it).

resettertje, 2023-09-25:
> Yes. It is on the device too, but also via software (tuya app allows it).

Aaaaah that I didn't know.

tomiss288, 2023-09-28:
Hi,
It's exactly the same device like TS0601_TZE200_z1tyspqw. I received new one on a warranty as a replacement for  this device. 
Maybe somebody who know can change script for TS0601_TZE200_z1tyspqw for support _TZE200_z1tyspqw ?

tomiss288, 2023-09-28:
I have one question, how to change temperature displayed from F to C ? Is it possible without Tuya Gateway ?

Mikescotland, 2023-09-28:
> Hi, It's exactly the same device like TS0601_TZE200_z1tyspqw. I received new one on a warranty as a replacement for this device. Maybe somebody who know can change script for TS0601_TZE200_z1tyspqw for support _TZE200_rtrmfadk ?

Try my converter, just replace manufacturer name with TS601_TZE200_rtrmfadk

tomiss288, 2023-09-28:
> Try my converter, just replace manufacturer name with TS601_TZE200_rtrmfadk

I did it, but it doesn't work in 100% good, e.g,  hy in your converter there is option for brightness ?  Also system mode don't work.

Mikescotland, 2023-09-28:
> > > Hi, It's exactly the same device like TS0601_TZE200_z1tyspqw. I received new one on a warranty as a replacement for this device. Maybe somebody who know can change script for TS0601_TZE200_z1tyspqw for support _TZE200_rtrmfadk ?
> > 
> > 
> > Try my converter, just replace manufacturer name with TS601_TZE200_rtrmfadk
> 
> I did it, but it doesn't work in 100% good, e.g, why in your converter there is option for brightness ? Also system mode don't work.

Then unfortunately you'd need to find data point and create the converter. As far as I know on mine it's not possible to switch between F and °C.

Tiggx, 2023-10-20:
in my case this worked: https://github.com/jacekk015/zha_quirks/blob/main/ts0601_trv_rtitek.py

Mikescotland, 2023-10-20:
Your sxript is for ZHA, this topic is for zigbee2mqtt.

Tiggx, 2023-10-20:
oh sorry. my mistake

Herrschacht, 2024-03-20:
Anyone got this working with SkyConnect v1.0 ?

closed 2024-10-18 by github-actions[bot]