Koenkk/zigbee2mqtt #18894: eglo-zm-rgb-tw Ceiling Light will not turn off stays at brightness 1

djalexnl, 2023-09-07:
### What happened?

I have a eglo-zm-rgb-tw Ceiling light connected succesfully with Z2MQTT in combination with HomeAssistant

I can control the light, no problem sofar, but when i turn the light off, Z2MQTT says the the light is off, but actualy it
 is on at brightness level 1.

I searched the internet and found that there was an issue with transitions and the lights, i know I am not the only one who crossed this issye,  But i still can NOT turn off the Ceiling light wich is pretty anoying..

Can someone explain this NOOB how to get the light to completely turn off.. not using Transition.. etc.
If i use the HomeControl app of the device itself, it turns OFF nicely, but i want to use Home Assistant (node red) to accomplish this automaticly.

Zigbee2MQTT
Current version: 1.33.0-1

Some more info about the device..:

Zigbee model
TLSR82xx
Zigbee Fabrikant
AwoX
Omschrijving
LED RGB & brightness
Ondersteuningsstatus
Ondersteund

Firmwareversie
0122052017
Fabrikant
[AwoX](https://www.zigbee2mqtt.io/supported-devices/#v=AwoX)
Model
[33943/33944/33946](https://www.zigbee2mqtt.io/devices/33943_33944_33946.html#awox-33943%252F33944%252F33946)

### What did you expect to happen?

Turn Light off (wich doesnt occur)

### How to reproduce it (minimal and precise)

Turn light On and Off

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20220219

### Adapter

usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

Diaoul, 2023-09-10:
Probably related https://github.com/Koenkk/zigbee2mqtt/discussions/18366#discussioncomment-6595176

djalexnl, 2023-09-18:
Thank you Diaoul... i found the solution. Under Specific Settings there is a Transition colum...
0 was set there.... i removed it ... empty that setting and problem is gone....
So 0 is not the same as NO parameter.... 

If transition is set to  a Number it wil work when changing color... but with ON or Off this transition must be empty otherwise you cannot turn the light off.

So for now i removed the settings... so no transition for now.

Thank you for pointing out the related problem.

closed 2023-10-27 by github-actions[bot]

zulufoxtrot, 2026-01-11:
> Thank you Diaoul... i found the solution. Under Specific Settings there is a Transition colum... 0 was set there.... i removed it ... empty that setting and problem is gone.... So 0 is not the same as NO parameter....

I can confirm that this bug is still present, and this workaround still works!