Koenkk/zigbee2mqtt #18897: Devices stopped reacting to most of the commands

kacperstopa, 2023-09-07:
### What happened?

After upgrade to 1.33.0 most of the commands sent to devices started to timeout. Regardless device type or manufacturer.
For example, when I turn on the light, it does not turn on. But when I send that command many times, ultimately it will turn on.
```
 2023-09-07 17:13:33Publish 'set' 'state' to 'switch4' failed: 'Error: Command 0x5c0272fffec4d4f9/1 genOnOff.off({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (118))'
```

### What did you expect to happen?

I expect all devices to work properly

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

0x26580700

### Adapter

ConBee2

### Debug log

```
Debug 2023-09-07 17:18:27Received MQTT message on 'zigbee2mqtt/switch4/set' with data '{"state":"ON"}'
Debug 2023-09-07 17:18:27Publishing 'set' 'state' to 'switch4'
Error 2023-09-07 17:18:37Publish 'set' 'state' to 'switch4' failed: 'Error: Command 0x5c0272fffec4d4f9/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (131))'
Info 2023-09-07 17:18:37MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'switch4' failed: 'Error: Command 0x5c0272fffec4d4f9/1 genOnOff.on({}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (131))'","meta":{"friendly_name":"switch4"},"type":"zigbee_publish_error"}'
```

jarilehtinen, 2023-09-07:
One of my Hue Dimmer Switches started acting up around that time. The "on" button works fine, but pressing the "off" button does nothing and the light on the switch starts flashing different colors.

Debug log says:

```
Debug 2023-09-07 19:22:43 Device 'Kylpyhuone Katkaisija' announced itself
Info 2023-09-07 19:22:43 MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Kylpyhuone Katkaisija","ieee_address":"0x0017880103caf532"},"type":"device_announce"}'
Debug 2023-09-07 19:22:45 Retrieving state of 'Kylpyhuone Katkaisija' after reconnect
Debug 2023-09-07 19:22:53 Received Zigbee message from 'Kylpyhuone Katkaisija', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 2 with groupID 0
Info 2023-09-07 19:22:53 MQTT publish: topic 'zigbee2mqtt/Kylpyhuone Katkaisija', payload '{"action":null,"action_duration":null,"battery":100,"brightness":255,"counter":1,"linkquality":76,"update":{"installed_version":1107324829,"latest_version":1107324829,"state":"idle"},"update_available":null}'
```

I have tried resetting, removing, resetting, adding, resetting... Stopping Zigbee2MTQQ, removing my dongle... pretty much everything.

Weird thing is, other switches for fine. So this might be related, or not...

olgru, 2023-09-07:
Assuming that's my Zigbee network with 115 devices (interferred by other radio) I didn't raise an error here. Reading that others do have the similiar problem - devices sometimes respond - I have to admit that I have the same issue.
That's wired,  for instance my shutter controls do allow to shut down, but Zigbee did not recognized the "I'm ready with shutdown" information. Lights switching on, but remains dark in HA Gui, Motion sensors not detecting motion but sometimes does etc... That all started maybe 2 weeks ago...

raedkit, 2023-09-07:
I have the same issue also. 
I have remarked that the devices took time to respond to my commands and some of my devices started showing as "offline".
I will try rollbacking to 1.32.2 and see if this fixes the issue.

mozdi, 2023-09-10:
I had the same issue.
After going back to Version 1.32.2 it works again.

raedkit, 2023-09-10:
> I have the same issue also. I have remarked that the devices took time to respond to my commands and some of my devices started showing as "offline". I will try rollbacking to 1.32.2 and see if this fixes the issue.

For me rollbacking have not fixed the issue.
The issue was resolved by upgrading the firmware of my coordinator to the latest version. Actually I have never flashed the coordinator since CC2652P since buying it. Flashing it seems to fix the issue. 2 days now without the issue.

FedericoKlappenbach, 2023-09-16:
Updating to 1.33.0 made most of my devices showing as offline. Reverted to 1.32.2 and all issues where resolved.

@raedkit 
Are you running 1.33.0 with latest coordinator firmware without issues?

raedkit, 2023-09-16:
> Updating to 1.33.0 made most of my devices showing as offline. Reverted to 1.32.2 and all issues where resolved.
> 
> @raedkit Are you running 1.33.0 with latest coordinator firmware without issues?

@FedericoKlappenbach yes I'm running a Sonoff CC2652P with the latest firmware and with the zigbee2mqtt [1.33.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.33.0) commit: [7ee207f](https://github.com/Koenkk/zigbee2mqtt/commit/7ee207f)
Now it's more stable. Also I increased the default ``transmit_power`` from 5 db to 10 db but I haven't seen an increase in LQI values.
What I found also is making sure that there's no near cables (power, usb or even RJ45) increase drastically the quality of the network and the availability of my devices more than anything else.
I will invest on a USB extension to make sure that the dongle is far from cables and everything behind my server.
I hope these details helps you.

closed 2023-10-25 by github-actions[bot]