Koenkk/zigbee2mqtt #18900: Bticino K4027C OTA Update

altrdev, 2023-09-07:
### What happened?

Hi, today I update a firmware from Ota Console. Always good but something strange happening now.
If I check new firmware, the red pop up appears with this message: `Firmware on
'O×000474000095133' is newer than latest firmware online.`

Checking logs, I can see this:
`{"installed_version":3556351,"latest_version":3555843`

Why installed version is bigger then latest version, if I do the update from z2m ota console?
Thanks

### Zigbee2MQTT version

1.33.0-1 

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8

developerGM, 2023-09-08:
I have the same problem.
How can we update the firmware?

closed 2023-10-16 by github-actions[bot]

donatigno, 2023-10-21:
I have same issue. Anyone can help to make ota update correctly?

blackwiz4rd, 2024-02-07:
This issue should still be open. I also have this issue :/. In addition, the docs mentions the calibration which does not work. Probably it does not work because I have an old firmware?
https://www.zigbee2mqtt.io/devices/K4027C_L4027C_N4027C_NT4027C.html

wiso, 2025-01-12:
Same problem in 2025
`zhc:ota:common: Firmware on '0x00047400015f04e2' ( Shutter SW with level control) is newer than latest firmware online.`