Koenkk/zigbee2mqtt #18902: Initial light turn_on ignores transition time, fades on, then immedty turns off

Megabytemb, 2023-09-08:
### What happened?

I have TS0505B Tuya Light.

When i set light.turn_on, it ignores the default set transition time, fades in over 1 second, then immedtly turns off.

If i then issue the exact same light.turn_on command again, it uses the correct transition time, and stays on.

if i do light.turn_on, then light.turn_off, then light.turn_on again, it exhibts the same issue.

### What did you expect to happen?

light.turn_on should turn the light on with the transition time set for the light, and stay on.

### Zigbee2MQTT version

 1.33.0

### Adapter firmware version

7.3.1.0 build 0

### Adapter

EZSP v12 > Silicon Labs Multiprotocol > QinHeng Electronics SONOFF Zigbee 3.0 USB Dongle Plus V2 

### Debug log

# Initial light turn on, where it fades in over 1 second, instead of 0.1 seconds, then turns off

Zigbee2MQTT:debug 2023-09-08 14:20:56: Received MQTT message on 'zigbee2mqtt/Bathroom Ceiling/set' with data '{"state":"ON"}'
Zigbee2MQTT:debug 2023-09-08 14:20:56: Publishing 'set' 'state' to 'Bathroom Ceiling'
2023-09-08T04:20:56.591Z zigbee-herdsman:controller:database:log Writing database to '/config/zigbee2mqtt/database.db'
2023-09-08T04:20:56.604Z zigbee-herdsman:controller:endpoint Command 0xa4c138ee650c3f3f/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":1}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2023-09-08T04:20:56.604Z zigbee-herdsman:adapter:ezsp:debg sendZclFrameToEndpointInternal 0xa4c138ee650c3f3f:37021/1 (0,0,0)
2023-09-08T04:20:56.606Z zigbee-herdsman:adapter:ezsp:ezsp ==> setExtendedTimeout: {"remoteEui64":{"_value":{"type":"Buffer","data":[164,193,56,238,101,12,63,63]}},"extendedTimeout":true}
2023-09-08T04:20:56.606Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"setExtendedTimeout","_id_":126,"_isRequest_":true,"remoteEui64":{"_value":{"type":"Buffer","data":[164,193,56,238,101,12,63,63]}},"extendedTimeout":true}
2023-09-08T04:20:56.606Z zigbee-herdsman:adapter:ezsp:uart --> DATA (7,7,0): 1700017e003f3f0c65ee38c1a401
2023-09-08T04:20:56.606Z zigbee-herdsman:adapter:ezsp:uart --> [775521a92a2a2a8d55f1a41d6bf193c84d7e]
2023-09-08T04:20:56.607Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (0)
2023-09-08T04:20:56.608Z zigbee-herdsman:adapter:ezsp:uart <-- [7055a1a92a2aae237e]
2023-09-08T04:20:56.608Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (7,0,0): 7055a1a92a2aae237e
2023-09-08T04:20:56.608Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (0)
2023-09-08T04:20:56.608Z zigbee-herdsman:adapter:ezsp:uart --> [8070787e]
2023-09-08T04:20:56.608Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 1780017e00
2023-09-08T04:20:56.608Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x7e: {"_cls_":"setExtendedTimeout","_id_":126,"_isRequest_":false}
2023-09-08T04:20:56.608Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (0) success
2023-09-08T04:20:56.608Z zigbee-herdsman:adapter:ezsp:ezsp ==> sendUnicast: {"type":0,"indexOrDestination":37021,"apsFrame":{"clusterId":8,"profileId":260,"sequence":158,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"messageTag":159,"message":{"type":"Buffer","data":[1,127,4,254,1,0]}}
2023-09-08T04:20:56.608Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"sendUnicast","_id_":52,"_isRequest_":true,"type":0,"indexOrDestination":37021,"apsFrame":{"clusterId":8,"profileId":260,"sequence":158,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"messageTag":159,"message":{"type":"Buffer","data":[1,127,4,254,1,0]}}
2023-09-08T04:20:56.608Z zigbee-herdsman:adapter:ezsp:uart --> DATA (0,0,0): 1800013400009d90040108000101000100009e9f06017f04fe0100
2023-09-08T04:20:56.609Z zigbee-herdsman:adapter:ezsp:uart --> [005a21a9602a152fc9904b2daa5493499d4e273572c866f4f9386289233f7e]
2023-09-08T04:20:56.609Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (1)
2023-09-08T04:20:56.609Z zigbee-herdsman:adapter:ezsp:uart <-- [015aa1a9]
2023-09-08T04:20:56.609Z zigbee-herdsman:adapter:ezsp:uart <-- [602a15913e6f7e]
2023-09-08T04:20:56.609Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,1,0): 015aa1a9602a15913e6f7e
2023-09-08T04:20:56.609Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2023-09-08T04:20:56.609Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2023-09-08T04:20:56.609Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 18800134000023
2023-09-08T04:20:56.609Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x34: {"_cls_":"sendUnicast","_id_":52,"_isRequest_":false,"status":0,"sequence":35}
2023-09-08T04:20:56.609Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (1) success
2023-09-08T04:20:56.618Z zigbee-herdsman:adapter:ezsp:uart <-- [7d315ab1a96b2a152fc9904b2daa549349]
2023-09-08T04:20:56.618Z zigbee-herdsman:adapter:ezsp:uart <-- [9d4e278872ce675c837e]
2023-09-08T04:20:56.618Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (1,1,0): 115ab1a96b2a152fc9904b2daa5493499d4e278872ce675c837e
2023-09-08T04:20:56.618Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (2)
2023-09-08T04:20:56.618Z zigbee-herdsman:adapter:ezsp:uart --> [82503a7e]
2023-09-08T04:20:56.618Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 1890013f00009d9004010800010100010000239f0000
2023-09-08T04:20:56.618Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x3f: {"_cls_":"messageSentHandler","_id_":63,"_isRequest_":false,"type":0,"indexOrDestination":37021,"apsFrame":{"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":35},"messageTag":159,"status":0,"message":{"type":"Buffer","data":[]}}
2023-09-08T04:20:56.653Z zigbee-herdsman:adapter:ezsp:uart <-- [215ab1a97d312a15b6589c4a24ab5593499c8dd812705e]
2023-09-08T04:20:56.653Z zigbee-herdsman:adapter:ezsp:uart <-- [9874f8de1c82f87d5e3b16757e]
2023-09-08T04:20:56.653Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (2,1,0): 215ab1a9112a15b6589c4a24ab5593499c8dd812705e9874f8de1c82f87e3b16757e
[... 33 lines cut ...]
2023-09-08T04:21:04.465Z zigbee-herdsman:adapter:ezsp:uart <-- [602a159615d07e]
2023-09-08T04:21:04.465Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (5,4,0): 5459a1a9602a159615d07e
2023-09-08T04:21:04.465Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (6)
2023-09-08T04:21:04.465Z zigbee-herdsman:adapter:ezsp:uart --> [8610be7e]
2023-09-08T04:21:04.466Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 1b800134000024
2023-09-08T04:21:04.466Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x34: {"_cls_":"sendUnicast","_id_":52,"_isRequest_":false,"status":0,"sequence":36}
2023-09-08T04:21:04.466Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (4) success
2023-09-08T04:21:04.474Z zigbee-herdsman:adapter:ezsp:uart <-- [6459b1a96b2a]
2023-09-08T04:21:04.474Z zigbee-herdsman:adapter:ezsp:uart <-- [152fc9904b2daa5493499d4e278f4dce6722167e]
2023-09-08T04:21:04.474Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (6,4,0): 6459b1a96b2a152fc9904b2daa5493499d4e278f4dce6722167e
2023-09-08T04:21:04.474Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (7)
2023-09-08T04:21:04.474Z zigbee-herdsman:adapter:ezsp:uart --> [87009f7e]
2023-09-08T04:21:04.474Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 1b90013f00009d900401080001010001000024a00000
2023-09-08T04:21:04.474Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x3f: {"_cls_":"messageSentHandler","_id_":63,"_isRequest_":false,"type":0,"indexOrDestination":37021,"apsFrame":{"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":36},"messageTag":160,"status":0,"message":{"type":"Buffer","data":[]}}
2023-09-08T04:21:04.509Z zigbee-herdsman:adapter:ezsp:uart <-- [7459b1a97d312a15b6589c4a24ab5593499c]
2023-09-08T04:21:04.509Z zigbee-herdsman:adapter:ezsp:uart <-- [8ad812705e9874f8dee382f87d5e3b4ceb7e]
2023-09-08T04:21:04.509Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (7,4,0): 7459b1a9112a15b6589c4a24ab5593499c8ad812705e9874f8dee382f87e3b4ceb7e
2023-09-08T04:21:04.509Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (0)
2023-09-08T04:21:04.509Z zigbee-herdsman:adapter:ezsp:uart --> [8070787e]
2023-09-08T04:21:04.510Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 1b900145000004010800010100010000c4ffb99d90ffff0518800b040004
2023-09-08T04:21:04.510Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":0,"apsFrame":{"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":1,"options":256,"groupId":0,"sequence":196},"lastHopLqi":255," [... cut ...]
Zigbee2MQTT:info  2023-09-08 14:21:04: MQTT publish: topic 'zigbee2mqtt/Bathroom Ceiling', payload '{"brightness":254,"color_mode":"color_temp","color_power_on_behavior":"previous","color_temp":500,"do_not_disturb":null,"last_seen":1694146864510,"linkquality":255,"state":"ON"}'
Zigbee2MQTT:info  2023-09-08 14:21:04: MQTT publish: topic 'zigbee2mqtt/Bathroom Ceiling', payload '{"brightness":254,"color_mode":"color_temp","color_power_on_behavior":"previous","color_temp":500,"do_not_disturb":null,"last_seen":1694146864510,"linkquality":255,"state":"ON"}'

kingii98, 2023-09-14:
I have same issue with same stick/version.

Megabytemb, 2023-10-15:
I'm still experinging this issue.

Does anyone have any ideas?

Megabytemb, 2023-11-22:
Hey, any thoughts on this? I'm still having this issue and its quite annoying......

whc2001, 2024-03-23:
Just realized this is the same problem I have experienced in #21625. From what I have tested, you can't even touch the transition setting otherwise this problem will appear. After that you need to pair it with HomeAssistant ZHA to temporarily bring it back.

closed 2024-10-20 by github-actions[bot]