Koenkk/zigbee2mqtt #18905: [Wrong device]:

Bidarin412, 2023-09-08:
### Link

https://www.alibaba.com/product-detail/Practical-Nice-black-color-90MM-big_1600733906704.html

### Model

ZTU-8136F

### Description

Smart switch (4 gang + 2 scene) with backlight and neutral wire

### Vendor

Tuya

### Picture (link)

https://s.alicdn.com/@sc04/kf/Hd7beecbe37b9406abeae53158cbde28d6.jpg_960x960.jpg

### Database entry

"type": router, "IEEEadd": 0xa4c1386c288daa58, "networkadd": 0xDE9B

### Notes

I have these switches, some are 1 gang, 4 gangs, one is 4 gangs + 2 scences (6 buttons). Their model all are ZTU, the 4 gang only is ZTU-4FN, 4 gangs +2 scences is ZTU-8136F. But after pairing via z2m, they are all labeled as TS0726 model which is 4 gang switch with neutral wire, even with the one 1 gang.
The manufacter doesn't have this own product link, so they just give me the link of a normal switch, the look of product is the same, they just add in the pcb board in zigbee.

Bidarin412, 2023-10-10:
hello? anyone? Please?

Killi77, 2023-10-31:
The same problem here... Have 4gang + 2scenes but integrated as "4 gang switch with neutral wire". 

TO: Bidarin412 - BTW I have opened few questions here in Z2M support for different topics but unfortunately got no reply to any of what I posted...  :(

Bidarin412, 2023-10-31:
> The same problem here... Have 4gang + 2scenes but integrated as "4 gang switch with neutral wire".
> 
> TO: Bidarin412 - BTW I have opened few questions here in Z2M support for different topics but unfortunately got no reply to any of what I posted... :(
Yea I posted this question in a facebook group, people told to post it here, the dev are actively supporting but still no sign of help :(

krylek, 2024-02-09:
Has anyone found a solution to this problem?

closed 2024-09-08 by github-actions[bot]

Koenkk, 2025-03-11:
Could you provide the data/database.db entry of these devices and indicated if they are the 4 gang, 1 gang or 4 gang + 2 scene button version?