Koenkk/zigbee2mqtt #18907: z2m exited with error code

tkarlsson, 2023-09-08:
### What happened?

z2m exited with error code:

Zigbee2MQTT:info  2023-09-07 21:53:45: MQTT publish: topic 'zigbee2mqtt/Luftvärmepump verkstad', payload '{"ac_frequency":50.1,"current_phase_a":0,"current_phase_b":null,"current_phase_c":null,"energy":562.303,"energy_phase_a":562.303,"energy_phase_b":null,"energy_phase_c":null,"last_seen":"2023-09-07T21:53:45+02:00","linkquality":0,"power":0,"power_apparent":0,"power_factor":-128,"power_phase_a":0,"power_phase_b":null,"power_phase_c":null,"voltage_phase_a":234,"voltage_phase_ab":null,"voltage_phase_b":null,"voltage_phase_bc":null,"voltage_phase_c":null,"voltage_phase_ca":null}'
Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)

### Zigbee2MQTT version

1.33.0-dev commit: ea8de2b

### Adapter firmware version

 20230507

### Adapter

Sonoff

### Debug log

Zigbee2MQTT:info  2023-09-07 21:53:45: MQTT publish: topic 'zigbee2mqtt/Luftvärmepump verkstad', payload '{"ac_frequency":50.1,"current_phase_a":0,"current_phase_b":null,"current_phase_c":null,"energy":562.303,"energy_phase_a":562.303,"energy_phase_b":null,"energy_phase_c":null,"last_seen":"2023-09-07T21:53:45+02:00","linkquality":0,"power":0,"power_apparent":0,"power_factor":-128,"power_phase_a":0,"power_phase_b":null,"power_phase_c":null,"voltage_phase_a":234,"voltage_phase_ab":null,"voltage_phase_b":null,"voltage_phase_bc":null,"voltage_phase_c":null,"voltage_phase_ca":null}'
Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)

Koenkk, 2023-09-08:
It looks like the coordinator crashed, this is very likely due to the 20230507 firmware, try the 20230901 firmware from [this thread](https://github.com/Koenkk/Z-Stack-firmware/discussions/474)

tkarlsson, 2023-09-10:
Tried 20230901 with latest dev release and got the same error after a while:

Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7)

Koenkk, 2023-09-10:
From when did this error start occuring?

tkarlsson, 2023-09-10:
This is the latest images I've downloaded:

zwavejs/zwavejs2mqtt   latest    fb28e455bd84   3 days ago      475MB
zwavejs/zwavejs2mqtt   <none>    c33b8f5d1acc   2 weeks ago     483MB
zwavejs/zwavejs2mqtt   <none>    a0d2f111f8eb   4 weeks ago     483MB
zwavejs/zwavejs2mqtt   <none>    388301f48367   8 weeks ago     477MB

I would say it started with image id c33b8f5d1acc and I didn't observe this specific error in a0d2f111f8eb

tkarlsson, 2023-09-11:
After the initial crash and restart it looks as the container has been running without restarting for 37h so far which is way better than before so hopefully it was just a initial flux and 20230901 are working much better.

closed 2023-10-20 by github-actions[bot]