Koenkk/zigbee2mqtt #18908: Upgraded to latest docker/coordinator firmware and after a while the coordinator is unresponsive

simowilso, 2023-09-08:
### What happened?

I updated to latest docker container on Rpi and found that after a while all zigbee devices appeared disconnected. So then I updated the coordinator (sonoff P) to the latest firmware but the problem remains. Although it was working bfore with the sonoff stick plugged in directly to the RPi - I added a usb extension cable - no difference.

Prior to the docker update this has been solid for months.

Zigbee2MQTT version
[1.32.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.32.2) commit: [1ec1e57](https://github.com/Koenkk/zigbee2mqtt/commit/1ec1e57)
Coordinator type
zStack3x0
Coordinator revision
20230507

The last message in the logs is always:

Error 2023-09-08 12:12:54Publish 'set' 'state' to 'LB Kitchen Lamp' failed: 'Error: Command 0x001788010245dca5/11 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'

Restarting the container (without rebooting or powering on/off the host) will fix the problem for maybe an hour or so.

### How to reproduce it (minimal and precise)

restart container.
wait a while (1 hour?)
try to turn lamp on and off

### Zigbee2MQTT version

1.32..2

### Adapter firmware version

20230507

### Adapter

sonoff usb stick P

simowilso, 2023-09-08:
partial debug log when the issue occurred:

https://paste.debian.net/1291345/

simowilso, 2023-09-08:
I'm totally baffled now. I think the image that I upgraded from was 1.30.2
I downgraded to that and it's behaving exactly the same. All good for around an hour then boom. :(

Koenkk, 2023-09-08:
It looks like the coordinator crashed, this is very likely due to the 20230507 firmware, try the 20230901 firmware from [this thread](https://github.com/Koenkk/Z-Stack-firmware/discussions/474)

simowilso, 2023-09-08:
> It looks like the coordinator crashed, this is very likely due to the 20230507 firmware, try the 20230901 firmware from [this thread](https://github.com/Koenkk/Z-Stack-firmware/discussions/474)

Thanks for the quick answer!
20230901 installed along with docker 1.32.2 I should know in an hour or two <crosses fingers>

simowilso, 2023-09-08:
No luck with 20230901

Debug 2023-09-08 19:44:22Received MQTT message on 'zigbee2mqtt/LB Kitchen Lamp/set' with data '{"state":"ON"}'
Debug 2023-09-08 19:44:22Publishing 'set' 'state' to 'LB Kitchen Lamp'
Error 2023-09-08 19:44:28Publish 'set' 'state' to 'LB Kitchen Lamp' failed: 'Error: Command 0x001788010245dca5/11 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
Debug 2023-09-08 19:44:28Error: Command 0x001788010245dca5/11 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7)

Koenkk, 2023-09-09:
Can you try with [`20221226`](https://github.com/Koenkk/Z-Stack-firmware/tree/Z-Stack_3.x.0_coordinator_20221226/coordinator/Z-Stack_3.x.0/bin). If not, provide me the herdsman debug log from starting z2m until is crashes.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

simowilso, 2023-09-09:
I tried 20230902 - same result.
Now installed 20221226, but network map shows only routers with everything else disconnected. Investigating.

simowilso, 2023-09-09:
Multiple "failed to execute LQI messages", network appears completely broken after downgrading to 20221226 firmware. What to do? Only routers are showing connected, but I cannot control them eg a Hue bulb router.

simowilso, 2023-09-09:
Panic over, network is becoming available again

simowilso, 2023-09-09:
Even with 20221216 I'm getting timeout. I will try to setup herdsman logging.

simowilso, 2023-09-09:
Ok. I *think* what happened is this. Some time ago I moved Ubuntu from SD card to usb SSD. Everything appeared to be fine, (but may not have been.)
When I updated zigbee2mqtt, it appeared to start misbehaving, hence me creating this issue.
After having tried several different Sonoff firmware versions with no success, I noticed that there were some strange usb errors in dmesg.
I eventually swapped the (3b) Pi out for a 4, and, hopefully that appears to have resolved the issue.

Thanks for the help, and sorry for the distraction.

closed 2023-09-09 by simowilso