Koenkk/zigbee2mqtt #18910: [New device support]: Giex TS0601 ist not supported (_TZE200_a7sghmms)

Blomus, 2023-09-08:
### Link

https://www.zigbee2mqtt.io/devices/QT06_2.html

### Database entry

Received Zigbee message from '0xa4c138b53665960a', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":107}],"seq":13824}' from endpoint 1 with groupID 0

### Comments

Device can not be added.
Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_a7sghmms
External converter is added (see below)

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_a7sghmms'},],
    zigbeeModel: ['TS0601'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'TS0601', // Vendor model number, look on the device for a model number
    vendor: 'giex', // Vendor of the device (only used for documentation and startup logging)
    description: 'Water irrigation valve', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.state(), e.mode(),e.cycle_irrigation_num_times(), e.irrigation_start_time(), e.irrigation_end_time(), le.ast_irrigation_duration(), e.water_consumed(), e.irrigation_target(), e.cycle_irrigation_interval(), e.linkquality()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

Koenkk, 2023-09-08:
Make sure you are running the latest z2m version.

cloudbr34k84, 2023-09-20:
There is an issue with this as its not reporting the State anymore

[image]

closed 2023-10-29 by github-actions[bot]