Koenkk/zigbee2mqtt #18912: BE468 smart lock - setting the pin code not working

RMNB, 2023-09-08:
### What happened?

When using the web ui to set the pin code for users, I get the error message below. This is when setting for users 0-29. Friendly name is "Front Door Lock"

`Publish 'set' 'pin_code' to 'Front Door Lock' failed: 'Error: user must be numbers'`

When try setting a pin code for users outside 0-29 I get an error that the user is out of bounds. 

### What did you expect to happen?

I was previously able to set pin codes using the web ui front end.

### Zigbee2MQTT version

1.33.0-dev commit: ea8de2b

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

alex-siqueira, 2023-09-17:
Same problem here: seems to be a recent bug.

maxi1134, 2023-09-26:
I also have this bug with my schlague lock. Since a few months actually.

RMNB, 2023-09-26:
I'm going to try setting the code manually using the number pad.

maxi1134, 2023-09-27:
> I'm going to try setting the code manually using the number pad.

This works for me; But then trying to add a code from Z2M breaks that initial code entered in the lock.

Montreal666, 2023-09-27:
Same here. following.

StephenJondrew, 2023-10-03:
I am encountering the same thing with the YRD256.

- When trying to enter a code in Z2M the lock failed: 'Error: user must be numbers'
- This error happens with either an empty or full slot
- When attempting to control a code slot that has a code already, the code is cleared from the lock for that slot

AliSuleiman01, 2023-10-18:
i have the same issue as well. trying to set one also deletes the one i set manually on the lock itself

zecoj, 2023-10-18:
I also have the exact same issue and behaviour with this lock [Yale YDD-D4F0-TSDB](https://www.zigbee2mqtt.io/devices/YDD-D4F0-TSDB.html). Trying to set code for an existing user (0-249) will result in that user's code being deleted, and won't work for new users.

Montreal666, 2023-10-18:
For people experiencing the issue, are you using the Home assistant add-on? Asking because I recently migrated to HA and the pin code feature was working prior. Maybe just a coincidence. Otherwise just wondering how it was broken; other than the Z2M update itself were there any changes to the Yale support code? Did you try rolling back to a prior version?

zecoj, 2023-10-18:
1. Yes, I'm using the Home Assistant add-on
2. It was working prior, I know this because I used to set up the lock previously. I noticed it stopped working a couple weeks ago but thought it was something wrong with my lock, which I don't think received any update since installed.
3. No I didn't try to roll back z2m, mainly because I don't know when it was broken.

RMNB, 2023-10-19:
I'm not using the addon. Z2m is installed using docker for me and I have the issue.

Montreal666, 2023-11-01:
Did anyone have a chance to test with the latest Z2M release ( https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.33.2) ?
I don't see anything Yale related in the fixes but just in case...
will test later has I temporarily switched these locks to a secondary ZHA controller...

zecoj, 2023-11-01:
I tried to set the pin to a couple of users and it seems to be working. No more error and code is set for the corresponding user ID correctly!

RMNB, 2023-11-02:
Working for me now as well! I'm going to close the issue.

closed 2023-11-02 by RMNB

maxi1134, 2023-11-02:
Works with the Schlage as well!