Koenkk/zigbee2mqtt #18928: Philips OTA broken since 1.29.0

siandco, 2023-09-10:
### What happened?

When checking a Philips device using the OTA 'check for new version' button, an error is received:
_Failed to check if update available for '' (unable to get local issuer certificate)_

I have captured logs and screenshots from the following versions:
- latest-dev (showing problem currently exists)
- 1.29.0 (first occurrence of this failure)
- 1.28.4 (last time OTA functioned as expected)

and have attached them as a zip file below.

ALSO: have notified that between **1.29.0** and **latest-dev** the two Philips motion detectors (Model 9290030675) stopped been displayed on the OTA page (even though they exist on my network).

[PhilipsOTAissue.zip](https://github.com/Koenkk/zigbee2mqtt/files/12568033/PhilipsOTAissue.zip)

### What did you expect to happen?

Response should be:
1. Failed to check if update available for '' (No image available for imageType '')
3. No update available for ''

### How to reproduce it (minimal and precise)

1. Discover Philips device
2. In OTA, click check for new updates"

### Zigbee2MQTT version

1.33.0-dev commit: 4daddfc

### Adapter firmware version

20230507

### Adapter

Zonoff ZBDongle-P

Hypfer, 2023-09-27:
Can't confirm with 1.33

> unable to get local issuer certificate

kinda sounds like an issue in your setup

siandco, 2023-09-27:
Thanks for trying.
However, if that was true, then it wouldn't work in 1.28.4 (which it does).
Something changed between 1.28.4 and 1.29.0 that causes the issue to happen.

Also, I have subsequently tested this on two different machines running Z2M (under Docker/Linux and Windows) and two different ISP's (one was a mobile hotspot).

siandco, 2023-11-01:
Still an issue

siandco, 2023-11-01:
> This issue is stale because it has been open 30 days with no activity. Remove stale label or comment or this will be closed in 7 days

Still happens

closed 2024-05-01 by siandco