Koenkk/zigbee2mqtt #18929: Impossibile to change TS0210 Sensitivity

tax987, 2023-09-10:
### What happened?

I try to change Sensitivity of TS0210 but always report an error.

### Zigbee2MQTT version

1.33.0-1 

### Adapter firmware version

20230507

### Adapter

CC2652P

### Debug log

Debug 2023-09-10 13:26:40Received MQTT message on 'zigbee2mqtt/sv_tenda/set' with data '{"sensitivity":10}'
Debug 2023-09-10 13:26:40Publishing 'set' 'sensitivity' to 'sv_tenda'
Error 2023-09-10 13:26:50Publish 'set' 'sensitivity' to 'sv_tenda' failed: 'Error: Write 0xa4c13878497ce809/1 ssIasZone({"currentZoneSensitivityLevel":10}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
Debug 2023-09-10 13:26:50Error: Write 0xa4c13878497ce809/1 ssIasZone({"currentZoneSensitivityLevel":10}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7)
Info 2023-09-10 13:26:50MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'sensitivity' to 'sv_tenda' failed: 'Error: Write 0xa4c13878497ce809/1 ssIasZone({\"currentZoneSensitivityLevel\":10}, {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SRSP - AF - dataRequest after 6000ms)'","meta":{"friendly_name":"sv_tenda"},"type":"zigbee_publish_error"}'

tax987, 2023-10-04:
HI!
Nobody can help me?

Andystew94, 2023-10-10:
Experiencing the same error.

Saesh123, 2023-10-31:
Same error

vindolin, 2023-10-31:
I have the same problem for two sensors (_TZ3000_xsjsnzhz, _TZ3000_bmfw9ykl).
Additionally there's no Label/Description for the property in the settings tab.
[image]

Ascariota, 2024-01-17:
Same error any Update ?

fsaris, 2024-01-19:
Similar issue here. I got 3 devices all versions of TS0202 and for 1 I can adjust the settings, the others give write error.

```
Publish 'set' 'sensitivity' to 'Bewegingsmelder hal' failed: 'Error: Write 0xa4c1382bd5334113/1 ssIasZone({"currentZoneSensitivityLevel":1}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (151))'
```

My devices:

- _works_ TS0202 _TZ3040_6ygjfyll https://www.zigbee2mqtt.io/devices/IH012-RT01.html#tuya-ih012-rt01
- _fails_ TS0202 _TZ3000_msl6wxk9 https://www.zigbee2mqtt.io/devices/ZMS-102.html#tuya-zms-102
- _fails_ TS0202 _TZ3000_6ygjfyll https://www.zigbee2mqtt.io/devices/IH012-RT01.html#tuya-ih012-rt01

Is there a starting place to debug this further?

Koenkk, 2024-01-20:
If this doesn't work even after waking the device (by pressing a button on it) right before sending the command, I expect those particular variants don't support it.

vindolin, 2024-01-20:
> If this doesn't work even after waking the device (by pressing a button on it) right before sending the command, I expect those particular variants don't support it.

Oh thanks for the tip, it actually worked after I pressed the button on the vibration sensor.
I think it would be helpful to add this to the device documentation.

Koenkk, 2024-01-21:
Done! changes will be reflected with the next z2m update

closed 2024-01-21 by Koenkk