Koenkk/zigbee2mqtt #18930: Updating to 1.33 changes Names to "MQTT Sensor"

PuckStar, 2023-09-10:
### What happened?

Updating to 1.33 changes Names to "MQTT Sensor" or "MQTT Binary Sensor".
I notice can change it manually again. When I edit an entity and click in the Name field it still shows the old friendly name in light grey. When I type that in and save it it's OK.
However I prefer not to have to do that for the many entities I have so I restored to 1.32.2-1

### Zigbee2MQTT version

1.32

### Adapter firmware version

n/a

### Adapter

n/a

timon, 2023-09-10:
Thanks for pin-pointing version. Pinning to `1.32.2` as per your suggestion fixed the issue for me as well

Koenkk, 2023-09-11:
You need to use at least HA 2023.8 for z2m 1.33.0 (https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.33.0)

closed 2023-09-11 by Koenkk