Koenkk/zigbee2mqtt #18932: Group State changed to "Off" when any member's state changed to "Off"

AlterSu24, 2023-09-10:
### What happened?

I created a group including a 2-gangs light switch.
Group members are each gang of the switch, and each gang controlled a light.
So this switch can control two lights.

Now, when I turn on one light in the group, the state of the group showed "ON"
Then I turn on another light in the group, of course, the state remains "ON"

BUT when I turn off any light in the group, the state of the group change to "OFF"!
I think it should remain "ON" until I turn another light off, right?

I tried to add "all_members_off" in configuration.yaml, but it didn't worked.
All my switch are exposed as light in home assistant, can it be the reason why this happened ?

### What did you expect to happen?

Group state remain "ON" until all members state is "OFF"

### How to reproduce it (minimal and precise)

Create a group with a 2-gang or 3-gang switch.

Turn on any light and check the group state is "ON"

Turn another light on and off.

Now check the group state, is it "ON"?

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

6.7.10.0 build 423

### Adapter

EZSP v8

AlterSu24, 2023-09-12:
Any tips that I can try to fix it?
@Koenkk ?
Thanks.

closed 2023-10-21 by github-actions[bot]

igorpuchkovcom, 2023-11-23:
I have the same issue.

wash02, 2024-04-28:
Same issue, if a solution could be provided this would be greatly appreciated

Koenkk, 2024-12-04:
Fixed in https://github.com/Koenkk/zigbee2mqtt/pull/25059, thanks @IIIEII!

PR #25059 (2024-12-04): Fixed calculation of group state when including non-default endpoints
Groups could contain endpoints with state stored in non default keys like state_l2. In this pull request I changed functions shouldPublishPayloadForGroup and areAllMembersOffOrClosed to consider actual state of every endpoint in group base on their names.
Fixes #18932