Koenkk/zigbee2mqtt #18934: [Feature request]: Increasing the permissible thermostat temperature for the boiler to 60 degrees

Malaysk, 2023-09-10:
### Is your feature request related to a problem? Please describe

Hello @Koenkk!
I have a thermostat with an external waterproof temperature sensor connected to the heating system. Thermostat for boiler - TS0601 (_TZE200_aoclfnxz).
The problem is that the permissible temperature in Settings can be set to a maximum temperature of 35 degrees, but the thermostat maintains a maximum temperature of 65. The water only heats up to 35 degrees, which is very little. Is it possible to increase the maximum permissible temperature in Settings to 60-70 degrees?
[image]
[image]

### Describe the solution you'd like

increase the maximum permissible temperature in Settings to 60-70 degrees in Thermostat

### Describe alternatives you've considered

You can manually set the temperature to 60 degrees, but if you touch the temperature slider, the temperature is again set to 35 degrees

### Additional context

[image]

malaakso, 2023-09-12:
The limitation is in the thermostat itself, datasheet lists the supported set point as 0...35 degrees C.

Malaysk, 2023-09-12:
> The limitation is in the thermostat itself, datasheet lists the supported set point as 0...35 degrees C.
In the thermostat settings, I can set the maximum temperature to 45 degrees. I also tried manually setting a higher temperature in the zigbee2mqtt settings - everything works and heats up.

Hypfer, 2023-09-27:
If I understand it correctly, the value you're looking for is here:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/cf0d07aeca1d869e6fab911c1eb0a877d26cd3b0/src/devices/moes.ts#L21
and here
https://github.com/Koenkk/zigbee-herdsman-converters/blob/cf0d07aeca1d869e6fab911c1eb0a877d26cd3b0/src/devices/moes.ts#L119C69-L119C71

So a PR might be possible, although I wonder if that might have side-effects for other devices using the same code?
It seems to be written for thermostats for regular room radiators. Those should not be hotter than 35°C

Still, maybe you'll get better feedback with a PR for that? Idk

Malaysk, 2023-10-02:
I added an external converter and now everything works great. The maximum temperature can be set to 60 degrees.
`
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = {...require('zigbee-herdsman-converters/converters/toZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').toZigbee};
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const exposesLocal = {
hour: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('h').withValueMin(0).withValueMax(23),
minute: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('m').withValueMin(0).withValueMax(59),
program_temperature: (name) => exposes.numeric(name, ea.STATE_SET).withUnit('°C')
.withValueMin(5).withValueMax(60).withValueStep(0.5),
};

const definition = {
zigbeeModel: ['TS0601'],
model: 'BHT-006-GCLZB',
fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_aoclfnxz'}],
vendor: 'BSED',
description: 'BSED thermostat',
fromZigbee: [fz.legacy.moes_thermostat],
toZigbee: [tz.legacy.moes_thermostat_child_lock, tz.legacy.moes_thermostat_current_heating_setpoint, tz.legacy.moes_thermostat_mode,
tz.legacy.moes_thermostat_standby, tz.legacy.moes_thermostat_sensor, tz.legacy.moes_thermostat_calibration,
tz.legacy.moes_thermostat_deadzone_temperature, tz.legacy.moes_thermostat_max_temperature_limit, tz.legacy.moes_thermostat_min_temperature_limit],
exposes: [e.child_lock(), e.deadzone_temperature(), e.max_temperature_limit(), e.min_temperature_limit(),
exposes.climate().withSetpoint('current_heating_setpoint', 5, 60, 1, ea.STATE_SET)
.withLocalTemperature(ea.STATE).withLocalTemperatureCalibration(-30, 30, 0.1, ea.STATE_SET)
.withSystemMode(['off', 'heat'], ea.STATE_SET).withRunningState(['idle', 'heat', 'cool'], ea.STATE)
.withPreset(['hold', 'program']),
e.temperature_sensor_select(['IN', 'AL', 'OU']),
exposes.composite('program', 'program', ea.STATE_SET).withDescription('Time of day and setpoint to use when in program mode')
.withFeature(exposesLocal.hour('weekdays_p1_hour'))
.withFeature(exposesLocal.minute('weekdays_p1_minute'))
.withFeature(exposesLocal.program_temperature('weekdays_p1_temperature'))
.withFeature(exposesLocal.hour('weekdays_p2_hour'))
.withFeature(exposesLocal.minute('weekdays_p2_minute'))
.withFeature(exposesLocal.program_temperature('weekdays_p2_temperature'))
.withFeature(exposesLocal.hour('weekdays_p3_hour'))
.withFeature(exposesLocal.minute('weekdays_p3_minute'))
.withFeature(exposesLocal.program_temperature('weekdays_p3_temperature'))
.withFeature(exposesLocal.hour('weekdays_p4_hour'))
.withFeature(exposesLocal.minute('weekdays_p4_minute'))
.withFeature(exposesLocal.program_temperature('weekdays_p4_temperature'))
.withFeature(exposesLocal.hour('saturday_p1_hour'))
.withFeature(exposesLocal.minute('saturday_p1_minute'))
.withFeature(exposesLocal.program_temperature('saturday_p1_temperature'))
.withFeature(exposesLocal.hour('saturday_p2_hour'))
.withFeature(exposesLocal.minute('saturday_p2_minute'))
.withFeature(exposesLocal.program_temperature('saturday_p2_temperature'))
.withFeature(exposesLocal.hour('saturday_p3_hour'))
.withFeature(exposesLocal.minute('saturday_p3_minute'))
.withFeature(exposesLocal.program_temperature('saturday_p3_temperature'))
.withFeature(exposesLocal.hour('saturday_p4_hour'))
.withFeature(exposesLocal.minute('saturday_p4_minute'))
.withFeature(exposesLocal.program_temperature('saturday_p4_temperature'))
.withFeature(exposesLocal.hour('sunday_p1_hour'))
.withFeature(exposesLocal.minute('sunday_p1_minute'))
.withFeature(exposesLocal.program_temperature('sunday_p1_temperature'))
.withFeature(exposesLocal.hour('sunday_p2_hour'))
.withFeature(exposesLocal.minute('sunday_p2_minute'))
.withFeature(exposesLocal.program_temperature('sunday_p2_temperature'))
.withFeature(exposesLocal.hour('sunday_p3_hour'))
.withFeature(exposesLocal.minute('sunday_p3_minute'))
.withFeature(exposesLocal.program_temperature('sunday_p3_temperature'))
.withFeature(exposesLocal.hour('sunday_p4_hour'))
.withFeature(exposesLocal.minute('sunday_p4_minute'))
.withFeature(exposesLocal.program_temperature('sunday_p4_temperature')),
],
onEvent: tuya.onEventSetLocalTime,
};

module.exports = definition;
`

closed 2023-11-10 by github-actions[bot]