Koenkk/zigbee2mqtt #18937: Wrong pictures

abecko, 2023-09-10:
### What happened?

Hello,

I found that the product image (TuYa ZY-M100-L and ZY-M100-S_2.html) is confused on the page. In the HA plugin is also displayed poorly.

https://www.zigbee2mqtt.io/devices/ZY-M100-L.html
https://www.zigbee2mqtt.io/devices/ZY-M100-S_2.html

### Zigbee2MQTT version

1.33.0-1

### Adapter firmware version

no meaning

### Adapter

ZZH!

### Debug log

[image]

abecko, 2023-09-11:
It would be great! Just replace or rename. 

Thanks
Edvard

Correct naming:
[image]

Koenkk, 2023-09-13:
Fixed!

abecko, 2023-09-14:
I confirm the pictures are correct now.

Thanks,
Edvard

DeKaN, 2024-02-03:
Hi@Koenkk,
It looks like `ZY-M100-S_2` can be in both implementations: wall mount and ceiling mount. My device `_TZE204_qasjif9e` looks like `ZG-205Z/A`, unlike the picture above. 
[Similar comment](https://github.com/Koenkk/zigbee2mqtt/issues/18450#issuecomment-1761203919)