Koenkk/zigbee2mqtt #18938: Cannot turn on with brightness (or transition)

Diaoul, 2023-09-10:
### What happened?

Using light model `NLG-TW light` from manufacturer `Paul Neuhaus`

When I try turning ON the light with a specified brightness, it does not actually turn ON and the lights reports that it is ON.

### What did you expect to happen?

I expected the light would actually be ON

### How to reproduce it (minimal and precise)

Requirement: Starting with the light OFF

Try to turn ON the light from Home Assistant using the brightness slider
OR
Send this payload `{"state": "ON", "brightness": 100}`
OR
Send this payload `{"state": "ON", "transition": 3}`

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20230507

### Adapter

tube_zb_gw_cc2652p2_poe

### Debug log

From OFF, using the brightness slider in HA. The light does not actually turn ON.

```
debug 2023-09-10 23:34:27Received MQTT message on 'zigbee2mqtt/Living Room Light/set' with data '{"state":"ON","brightness":97}'
debug 2023-09-10 23:34:27Publishing 'set' 'brightness' to 'Living Room Light'
info 2023-09-10 23:34:27MQTT publish: topic 'zigbee2mqtt/Living Room Light', payload '{"brightness":97,"color_mode":"color_temp","color_temp":370,"color_temp_startup":65535,"linkquality":25,"state":"ON"}'
```

Diaoul, 2023-09-10:
Note that when the light is ON:
- I can turn it OFF with a nice transition with a negative `brightness_move_onoff` (but the state is wrong)
- I can turn it back ON with a nice transition with a positive `brightness_move_onoff` (but the state is wrong)

Diaoul, 2023-09-11:
- The lights with the issue have firmware `1.1`
- I have another model with the same model `NLG-TW light` that reports firmware `V1.3` and works perfectly well

:point_right: I have contacted the help center of the manufacturer but I don't have high hopes that I will be able to do an upgrade...

Diaoul, 2023-09-16:
No answer from support besides asking me to have a replacement from my reseller :disappointed:
 
I have not yet bound the light to a switch, it might automatically force the light to do [reporting](https://www.zigbee2mqtt.io/guide/usage/binding.html#state-changes) and thus make this issue go away. I'll test as soon as I receive a new switch, hopefully during the week.

Diaoul, 2023-09-17:
> I have not yet bound the light to a switch, it might automatically force the light to do [reporting](https://www.zigbee2mqtt.io/guide/usage/binding.html#state-changes) and thus make this issue go away. I'll test as soon as I receive a new switch, hopefully during the week.

I've tried creating the reporting rules and a bind to the controller. However, it did not fix the issue: sometimes, turning on the light with either brightness or transitions sometimes does not work.

Wikibear, 2023-09-19:
Yeah, lights support isn't very good. There are more bugs on my side. Did you have installed the GUI of Zigbee2MQTT? Can you check if this happens also there? Are you monitoring everything viq e.g. MQTT Explorer? Did you execute a command via MQTT Explorer?

I have IKEA Ambient Bulb. If i switch off via GUI brightness stay at last value. Brightness must be 0 if off but stay on 1. Duno why. If i click on refresh button the brightness change to 1... not to 0.

And i have more bugs with color bulbs and HSB... I had open an issue many month ago without any response of anyone.

LookedPath, 2023-10-04:
Could this be related to #16388 ?
I've been having very similar issues to what's reported here.

Diaoul, 2023-10-04:
Here it seems really related to the transition and brightness. I have a remote with binds to the lights and I have to admit that it's a bit hit-or-miss as well: I sometimes have to press multiple times 🤷 
I am not sure what is the payload that is sent to the light or in what order but it seems to have _some_ effect on the light's behavior

LookedPath, 2023-10-05:
> Here it seems really related to the transition and brightness. I have a remote with binds to the lights and I have to admit that it's a bit hit-or-miss as well: I sometimes have to press multiple [...]

I suggested it might be related to the issue with IKEA lightbulbs because I'm experiencing very similar issues.
For instance with my [LED1925G6](https://www.zigbee2mqtt.io/devices/LED1925G6.html#ikea-led1925g6) (on fw 1.0.021) I turn them off being at max brightness, leave them off for some time and when I turn them back off they automatically reverted back to their minimum brightness. At least these turn on, even if at minimum brightness.
For [LED2003G10](https://www.zigbee2mqtt.io/devices/LED2003G10.html#ikea-led2003g10) (on fw 1.0.012) the issue is even worse as I turn them off while they are at max brightness, leave them off for a while, when I turn them back on they report being on but the light is actually off. 

Z2M is running as an addon on HASSIO and the only way I found to deal with both these issues is to turn them on specifying a brightness, either with Google Home or by using the brightness slider introduced in the Tiles cards in HA.

When I get home tonight I'll try to collect some debug logs.

closed 2023-11-13 by github-actions[bot]

slavirotte, 2025-12-14:
> * The lights with the issue have firmware `1.1`
> 
>     * I have another model with the same model `NLG-TW light` that reports firmware `V1.3` and works perfectly well
> 
> 
> 👉 I have contacted the help center of the manufacturer but I don't have high hopes that I will be able to do an upgrade...

Is it possible to update the firmware of a Paul Neuhaus device? I have an NLG-TW light device, but I cannot find it in the OTA list of the Zigbee2MQTT interface. I would like to update it to version 1.3.0 if there are no issues.

Diaoul, 2025-12-17:
> Is it possible to update the firmware of a Paul Neuhaus device?

Not to my knowledge, both Paul Neuhaus and the zigbee controller manufacturer Anway Elec did not reply to my many requests for that. Anway said that they don't provide OTA on devices that work with a remote as the two are mutually exclusive (for some reason 🤷).

More [here](https://github.com/Koenkk/zigbee-herdsman-converters/pull/10314#issuecomment-3645490988).