Koenkk/zigbee2mqtt #18939: [Feature request]: Force device to use different config

Ltek, 2023-09-11:
### Is your feature request related to a problem? Please describe

I'm not sure if there is a detection problem but my ask will solve the problem.

Real life example (happening to me now) of the problem...
I bought 6 Zemismart ZM25EL... they are identical and same model #s. But one detected as TS0601_cover_3 (https://www.zigbee2mqtt.io/devices/TS0601_cover_3.html#tuya-ts0601_cover_3 ... manufacture _TZE200_68nvbio9) and the others as TS0601_cover_1 (https://www.zigbee2mqtt.io/devices/TS0601_cover_1.html ... manufacture _TZE200_pw7mji0l)

TS0601_cover_3 has features not exposed (like battery level) that are needed. The seller, largest on Amazon, says they verified with tech support are the same motor.

### Describe the solution you'd like

Have a simple way (like in the UI) to tell Z2M to use the config from a different known device. In this case... simply tell Z2M "use config for TS0601_cover_3"

### Describe alternatives you've considered

I looked at the documentation for creating a converter... its confusing, and not user friendly. I don't want to "create" a device, just leverage an existing one.

### Additional context

With as many devices that are simply relabeled / white-labeled, I'd think this would solve a lot of issues and reduce dev time for merging in new code for just simply adding a new manufacturer number? 

Deconz has a similar method using "DDF" files... which are simply text config files for each model. Copy existing DDF, edit to change model/manufacture = done.

Koenkk, 2023-09-11:
@jracoma asking because you added the [TS0601_cover_3](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5563), does this device support battery?

jracoma, 2023-09-11:
> @jracoma asking because you added the [TS0601_cover_3](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5563), does this device support battery?

I have ZM16EL and it supports battery.

Ltek, 2023-09-11:
> @jracoma asking because you added the [TS0601_cover_3](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5563), does this device support battery?

yes it does... that's my point. I have 7 Zemismart model ZM25EL, oddly one of them detected as TS0601_cover_3, the rest as TS0601_cover_1 
so I want to force Z2M to "see" the other 6 motors as TS0601_cover_3 (not TS0601_cover_1).

I'd hope this would not be difficult to implement in Z2M's UI as I'd think it is just changing the pointer in the database for the paired devices -- and maybe storing those current values?  (I'm new to Z2M so don't know the format its in, where it is, etc).

Koenkk, 2023-09-17:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/26eb1f0bfaae51a8f5ab231b95e635d3

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Ltek, 2023-09-17:
Z2M will not start when I add that to the .yaml file, ... tried start and end

external_converters:
    - ext_converter.js

Koenkk, 2023-09-18:
can you provide the log with the error?

Ltek, 2023-09-18:
Where? The log folder has a folder with the date but no file. I tried
multiple times

Ltek, 2023-09-18:
error msg... "ERROR : POST to page returned data but errored unexpectedly while processing it. : Unexpected token 'E', "Error proc"... is not valid JSON"

I had to turn on Trace to get this error window to show. I still dont get an actual TXT log file in here...
\ZigbeePlus\zigbee2mqtt\data\log\2023-09-17.16-17-52

if I remove that line, and Z2M starts then I get log files as expected.

francisp2, 2023-09-19:
Check if the external convertor actually contains json, downloading from github can be tricky.

Ltek, 2023-09-19:
I created the JS file with the code provided. Several times.

This problem is  making my point that if we had a redirect (treat X like it
is Y) I'd have been done in 10 seconds wothout bithing anyone...so far I'm
2 hrs in and still not able to test the converter.

Koenkk, 2023-09-19:
- What z2m version are you using, is this 1.33.0?
- I don't know the Homseer plugin, but the z2m STDOUT must be available somewhere. This is where the error should be shown.

> I'd have been done in 10 seconds wothout bithing anyone...so far I'm 2 hrs in and still not able to test the converter.

My opinion: While it may have solved the issue in your case, it won't solve the issue for another user. I rather prefer 1 person having to spend some times on this and fix it for good, then all users having to spent a little time on it to figure out how to get it working.

Ltek, 2023-09-19:
Here.s what I found on the About page...

Zigbee2MQTT version
 [1.31.2] commit: [unknown]
Coordinator type
EZSP v8
Coordinator revision
6.10.3.0 build 297
Coordinator IEEE Address
0xe0798dfffef42cde
Frontend version
0.6.129

Ltek, 2023-09-20:
I just updated to 1.33.0 (really need an auto update feature) and that does not fix the problem

Ltek, 2023-09-20:
Update... 1.33.0 *does* work and now loads the YAML with the converter included.

Is there a way to 'force' a rescan/reinspection of the device to see if the device exposes the same parameters?

Ltek, 2023-09-21:
Yes, so far... I have not tested all the functions yet but all the features are now showing, including battery and reporting battery level.

Koenkk, 2023-09-21:
Great, integrated the fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-10-30 by github-actions[bot]