Koenkk/zigbee2mqtt #18941: After firmware update EGLO Connect.2 identified incorrectly and color changing stopped working

malaakso, 2023-09-11:
### What happened?

I updated [EGLO Connect.2 12253](https://www.eglo.com/en/lightbulb-bulb-12253.html) to firmware 2.7.0_1279 (using Bluetooth and Awox phone app). It used to be identified as AwoX 33943/33944/33946 (see issue #11915 ) but now identifies as SALITERAS-Z ceiling light. The biggest problem is that color changing with hue and saturation is no longer working, resulting in bad colors.

Database entry:
`{"id":49,"type":"Router","ieeeAddr":"0xa4c138713e047928","nwkAddr":10927,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"EGLO_ZM_RGB_TW","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64599],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"modelId":"EGLO_ZM_RGB_TW","manufacturerName":"AwoX","powerSource":1,"zclVersion":3,"appVersion":2,"stackVersion":1,"hwVersion":62,"swBuildId":"2.7.0_1279"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370,"colorMode":2,"currentX":28213,"currentY":21561,"enhancedCurrentHue":0,"currentSaturation":75,"colorTemperature":202,"currentHue":0,"options":1,"numPrimaries":0}},"genLevelCtrl":{"attributes":{"currentLevel":254}},"genOnOff":{"attributes":{"onOff":1,"startUpOnOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002298941a","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b002298941a","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":1}],"meta":{}},"3":{"profId":4751,"epId":3,"devId":269,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":62,"swBuildId":"2.7.0_1279","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1693475523614,"defaultSendRequestWhen":"immediate"}`

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20230517

### Adapter

Slaesh CC2652RB

Koenkk, 2023-09-11:
Could you check if the H/S color issue is fixed with the following external converter: https://gist.github.com/Koenkk/11527aad291cf5586d3a1039e721f487

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

malaakso, 2023-10-05:
@Koenkk sorry for the long delay, but I can confirm this converter works. However, I noticed that when using the Color (HS) control from UI, the color mode stays as "xy". Only by sending MQTT directly with hue and saturation payload does it change to color mode "hs".

Koenkk, 2023-10-05:
@malaakso found the issue, pushed a fix.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)