Koenkk/zigbee2mqtt #18948: Property 'coordinatorCheck' does not exist on type 'Controller'

segdy, 2023-09-11:
### What happened?

When upgrading to latest 1.33.0, z2m does not start. It hangs for hours on a process called `node /opt/zigbee2mqtt/node_modules/.bin/tsc` which uses 100% CPU:

```
# sudo -u zigbee2mqtt /usr/local/bin/npm start

> zigbee2mqtt@1.33.0 start  
> node index.js

Building Zigbee2MQTT... (initial build), failed
Error: Command failed: npm run build

    at ChildProcess.exithandler (node:child_process:419:12)
    at ChildProcess.emit (node:events:513:28)
    at maybeClose (node:internal/child_process:1091:16)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:302:5)
```

I manually tried to execute `npm run build` (I hope that's the right thing). Surprisingly the process exits much sooner and this is the error message I get:

```
# time sudo -u zigbee2mqtt /usr/local/bin/npm run build

> zigbee2mqtt@1.33.0 build
> tsc && node index.js writehash

lib/zigbee.ts:199:43 - error TS2339: Property 'coordinatorCheck' does not exist on type 'Controller'.

199         const check = await this.herdsman.coordinatorCheck();
                                              ~~~~~~~~~~~~~~~~

lib/zigbee.ts:200:59 - error TS7006: Parameter 'd' implicitly has an 'any' type.

200         return {missingRouters: check.missingRouters.map((d) => this.resolveDevice(d.ieeeAddr))};
                                                              ~

Found 2 errors in the same file, starting at: lib/zigbee.ts:199

real    5m21.861s
user    4m53.218s
sys     0m5.421s
```

Version info:

```
# lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye
# uname -a
Linux rfpi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux
# node -v
v18.16.0
# 
```

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

Currently not able to start z2m to check but issue not related to adapter

### Adapter

Sonoff Zigbee

Koenkk, 2023-09-11:
Did you execute `npm ci` after updating?

segdy, 2023-09-11:
How embarrassing I forgot this. Worked out. Sorry for bothering. But maybe it helps others when googling.

closed 2023-09-11 by segdy