Koenkk/zigbee2mqtt #18950: Tuya ZG-204ZL illuminance reported in lux while it's raw

dinde, 2023-09-11:
### What happened?

Just installed a ZG-204ZL motion sensor with luminance sensor.
Values reported for illuminance as lux is in fact raw (number can't be that high)
[image]

### What did you expect to happen?

illuminance is raw not in lux.
As a feature it could be nice to have both values (with a converter directly in z2m but it's manageable thru home assistant as well).

### How to reproduce it (minimal and precise)

Just install it

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20230507

### Adapter

Sonoff ZBDongle-P

### Debug log

Will upload if needed/requested

Crumar, 2023-12-25:
What is the actual calculation?

muktheo, 2024-01-13:
any update? ive face the same issue

KaelWD, 2024-06-08:
Looks like this one is using `illuminance()`: https://github.com/Koenkk/zigbee-herdsman-converters/blob/3ec951e4c16310be29cec0473030827fb9a5bc23/src/devices/tuya.ts#L6301
Others use `illuminance_lux()` https://github.com/Koenkk/zigbee-herdsman-converters/blob/3ec951e4c16310be29cec0473030827fb9a5bc23/src/devices/tuya.ts#L6391 which goes through this conversion: https://github.com/Koenkk/zigbee-herdsman-converters/blob/3ec951e4c16310be29cec0473030827fb9a5bc23/src/converters/fromZigbee.ts#L466

knuti1960, 2024-09-16:
I'm not sure if it's really the RAW value that the ZG-204ZL device delivers. I compared it with my HUE sensor (Philips 9290012607), which displays both the RAW value and the LUX value. I placed both devices right next to each other in the same room for the test. The result was:
[image]
[image]

rodrigojds, 2024-10-21:
Hi, anything we can do to show the correct value. 
it’s currently night time and values are incorrect. 

I have 2 of these devices..try and guess which ones from the screenshot? :D 

[image]

Koenkk, 2025-01-25:
You can apply a calibration via the device specific settings, if someone lets me know what calibration value works I can apply that by default.

closed 2025-04-04 by github-actions[bot]

engels0n, 2025-06-24:
Hi,
I just faced the same "problem".
I have now placed an Ikea [E2134 ](https://www.zigbee2mqtt.io/devices/E2134.html)directly next to the [ZG-204ZL](https://www.zigbee2mqtt.io/devices/ZG-204ZL.html) and determined the following values:

[image]

I don't know how the calibration has to be implemented, but maybe the values will give you a hint.

danielo515, 2025-08-09:
Why has this been closed? I'm having the same issue

JinRoh3, 2025-08-27:
Same issue here with all my devices. Did anyone fix it?

TrueBlue1000, 2026-09-21:
@m-sct 
>  I have recently bought a ZG-204ZL and will be doing a similar analysis to it. Will share my results once I have collected enough data. Anedoctal evidence from all the Tuya sensors I have at hand do seem to support my methodology above.
> 
> Hope that help to add more info to the odd behaviour of Tuya motion sensors that can report illuminance readings.
 
Hi, I just bought some of these. did you reach a conclusion?

djaytek, 2026-10-03:
I am also having the same issues with ZG-204ZL devices.  Any updates regarding the solution?