Koenkk/zigbee2mqtt #18952: Lost on/off state for TS0001 switches after 1.33.0-1 update

patrick-foubert-56770, 2023-09-12:
### What happened?

After udate all TS0001 switches have same issue:

Cette entité n'est plus fournie par l'intégration mqtt. Si l'entité n'est plus utilisée, supprimez-la depuis les paramètres.
This entity was not longer supported by mqtt...

If i look logs during on/off

Info 2023-09-12 09:48:31MQTT publish: topic 'zigbee2mqtt_2/Inter-Etage-Dressing', payload '{"linkquality":98,"power_on_behavior":"previous","state":"ON"}'
Debug 2023-09-12 09:48:32Received Zigbee message from 'Inter-Etage-Dressing', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2023-09-12 09:48:32MQTT publish: topic 'zigbee2mqtt_2/Inter-Etage-Dressing', payload '{"linkquality":94,"power_on_behavior":"previous","state":"OFF"}'

But not state change

### What did you expect to happen?

upgrade issue

### Zigbee2MQTT version

1.33.0-1

### Adapter firmware version

20220219

### Adapter

SLZB-06

Gollam, 2023-09-12:
Same happening with ikea tradfri

Paweu00, 2023-09-16:
Same problem....
[image]

francisp2, 2023-09-16:
Are you on HA 2023-9 or 2023-9 ?

Gollam, 2023-09-16:
@francisp2 You asked the same thing twice, but as for me, I am on 2023.9.2.

francisp2, 2023-09-17:
Mine work nice, but I only added them on 1.33

[image]

Gollam, 2023-10-18:
I still have this issue with 1.33.1.

riegelbrau, 2024-01-24:
I have this issue with a "MHCOZY 1 gang switch (TS0001)" and Zigbee2MQTT version 1.35.1 (commit #3c962042) tigether with the homeassistant integration in openHAB 4.0.4.
This is the config payload for the switch in `homeassistant/switch/0xa4c1380868f7bc9b/switch/config`:
`{
	"availability": [
		{
			"topic": "zb2mqtt/bridge/state"
		},
		{
			"topic": "zb2mqtt/0xa4c1380868f7bc9b/availability"
		}
	],
	"availability_mode": "all",
	"command_topic": "zb2mqtt/0xa4c1380868f7bc9b/set",
	"device": {
		"identifiers": [
			"zigbee2mqtt_0xa4c1380868f7bc9b"
		],
		"manufacturer": "TuYa",
		"model": "1 gang switch (TS0001)",
		"name": "0xa4c1380868f7bc9b",
		"via_device": "zigbee2mqtt_bridge_0x00212effff06c0d7"
	},
	"json_attributes_topic": "zb2mqtt/0xa4c1380868f7bc9b",
	"name": null,
	"object_id": "0xa4c1380868f7bc9b",
	"origin": {
		"name": "Zigbee2MQTT",
		"sw": "1.35.1",
		"url": "https://www.zigbee2mqtt.io"
	},
	"payload_off": "OFF",
	"payload_on": "ON",
	"state_topic": "zb2mqtt/0xa4c1380868f7bc9b",
	"unique_id": "0xa4c1380868f7bc9b_switch_zb2mqtt",
	"value_template": "{{ value_json.state }}"
}`
The Thing gets created with all expected config topics:
[image]
But only these advaced channels appear:
[image]

This is the message when I press the button that switches ON/OFF in `zb2mqtt/0xa4c1380868f7bc9b:`
`{"last_seen":"2024-01-24T17:22:35.483Z","linkquality":255,"power_on_behavior":"off","state":"ON"}`

closed 2024-08-23 by github-actions[bot]