Koenkk/zigbee2mqtt #18954: YS-MT750L curtain motor says not supported but its in the list of supported devices

dave21w, 2023-09-12:
### What happened?

When pairing says not supported

### What did you expect to happen?

Should pair as on supported list named YS-MT750-L shown here.
https://www.zigbee2mqtt.io/devices/YS-MT750L.html
or
https://zigbee.blakadder.com/Yushun_YS-MT750.html
however it also names it as this

Model YS-MT750 manufactured by Yushun
Zigbee ID: TS0601 | _TZE200_bqcqqjpb | _TZE200_fdtjuw7u 

But I get the below:

Zigbee Model              TS0601
Zigbee Manufacturer  _TZE200_gaj531w3
Support status           Not supported

Not sure why the lettets and numbers after the _TZE200_ are different on mine
Currently running z2m latest.

### How to reproduce it (minimal and precise)

Have tried pairing and deleting multiple times but no change

### Zigbee2MQTT version

1.33.0

### Adapter firmware version

20220219

### Adapter

Electrolama zig-a zig-ah zStack3x0

Koenkk, 2023-09-13:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-09-13 by Koenkk

dave21w, 2023-09-14:
Thanks for the super fast response and fix